lathoub/Arduino-AppleMIDI-Library #10: Not compatible with Arduino 1.6.4

Hackmancoltaire, 2015-07-02:
Tested against newest revision. No midi messages are received by host. Recompiled the same code under 1.0.6 and things work fine.

lathoub, 2015-07-02:
Thanks for signaling - I haven't tried with 1.6.4, but just tested it just now using IDE 1.6.5. It seems to work fine (see screenshot below). Can you check with 1.6.5?

[image]

lathoub, 2015-07-19:
I tested again under 1.6.5 and all works fine - have you tested under 1.6.5?

Hackmancoltaire, 2015-07-19:
I'm in the process of testing. Within the next few days I'll have an answer.

Hackmancoltaire, 2015-07-21:
Confirmed broken on Mac 1.6.5. Compiled master branch as of today under 1.0.6 and it works fine.

Hackmancoltaire, 2015-07-21:
There seems to be a problem with the session connection. I see packets in Wireshark as the host invites, the Arduino accepts, then it does it over again
[image]

Hackmancoltaire, 2015-07-21:
Ok, scratch that. Had a configuration issue. Seems to work.

closed 2015-07-21 by Hackmancoltaire