lathoub/Arduino-AppleMIDI-Library #11: Adapting Arduino-AppleMidi-Library to wifi for ESP8266

rubenbroman, 2015-08-05:
Hi,

I'm trying to adapt this lovely library to be used on the ESP8266 using wifi. So far I've changed all the EthernetUDP calls to WiFiUdp. It seemed to be working fine and when I connect to the ESP8266 from the OS X Audio Midi it seems like I get a connection. I won't get anything in the Midi monitor app and the ESP8266 starts spitting out gibberish over serial.

Do you have any idea where to start looking? I'm not really a programmer and I don't know much about the inner workings of AppleMidi so it would mean a lot to get some help.

This is what the serial output looks like. The example code follows.

``` 
Connected to rutgerfuchsgatan9.se
IP address: 10.0.1.170
MDNS responder started

IP address is 10.0.1.170.
OK, now make sure you an rtpMIDI session that is Enabled
Add device na< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
MacBookPro ,ssrc 0xEBD5716C ,initiatorToken = 0x109CF92E
� @`[�?`[�?���?p��?<��?test�?@��?{ @O_�
]�?�S�?���?�S�?Ȉ�?�S�?�� @�   4F5FDC11@0109CF92E
MacBookPro��?EBD5716C
test�?X��?{ @O_�
2�?`[�?�Z#@0A�?4F5FDC11
Connected to session MacBookPro
EBD5716C?�O�?`[�?�!@�"�?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
��?EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
[�?EBD5716C?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
[�?EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
[�?EBD5716C?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
[�?�3@(��?���?���,EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?�3@EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
�?p��?p��?p��?�� @�I @�Z�?�Z�?EBD5716C
Disconnected
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
��1, n=45, v=55)
MacBookPro:@`EBD5716CDED7263
Z�?�test�?@��?{ @O_�
2�?`[�?�Z#@0A�?(��?���?4F5FDC11@0A�?(��?���?DED7263
���?� @xc @0MacBookPro�?���?� @xc @02�?`[�?�Z#@0A�?(��?���?EBD5716C
test�?X��?{ @O_�
2�?`[�?�Z#@0A�?(��?���?4F5FDC11
Connected to session MacBookPro
EBD5716C?�O�?`[�?�!@�"�?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
]�?(!@EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?(!@EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
[�?(!@EBD5716C?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
[�?(!@EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?(!@EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
>@�;@EBD5716C?�O�?�"�?0
�?4F5FDC11?�O�?�"�?1
[�?�;@EBD5716C?�O�?�"�?2
�?4F5FDC11?�O�?�"�?0
[�?EBD5716C?�O�?�"�?1
�?4F5FDC11?�O�?�"�?2
1, n=45, v=55)
��1, n=45, v=55)
��1, n=45, v=55)
�?p��?p��?p��?�� @�I @�Z�?�Z�?EBD5716C
Disconnected
1, n=45, v=55)
��1, n=45, v=55)
1, n=45, v=55)
``` 

Here's the example code:

``` 
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include "AppleMidi.h"

MDNSResponder mdns;

const char* ssid = "rutgerfuchsgatan9.se";
const char* password = "freewifi";

unsigned long t0 = millis();

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void setup()
{
  // Serial communications and wait for port to open:
  Serial.begin(115200);

  Serial.print("Getting IP address...");

  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

   if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  
  // print your local IP address:
  Serial.println();
  Serial.print("IP address is ");
  for (byte thisByte = 0; thisByte < 4; thisByte++) {
    // print the value of each byte of the IP address:
    Serial.print(WiFi.localIP()[thisByte], DEC);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("OK, now make sure you an rtpMIDI session that is Enabled");
  Serial.print("Add device named Arduino with Host/Port ");
  Serial.print(WiFi.localIP()[0], DEC);
  Serial.print(".");
  Serial.print(WiFi.localIP()[1], DEC);
  Serial.print(".");
  Serial.print(WiFi.localIP()[2], DEC);
  Serial.print(".");
  Serial.print(WiFi.localIP()[3], DEC);
  Serial.println(":5004");
  Serial.println("Then press the Connect button");
  Serial.println("Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes");

  // Create a session and wait for a remote host to connect to us
  AppleMIDI.begin("test");

  AppleMIDI.OnConnected(OnAppleMidiConnected);
  AppleMIDI.OnDisconnected(OnAppleMidiDisconnected);

  AppleMIDI.OnReceiveNoteOn(OnAppleMidiNoteOn);
  AppleMIDI.OnReceiveNoteOff(OnAppleMidiNoteOff);

  Serial.println("Sending NoteOn/Off of note 45, every second");
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  // Listen to incoming notes
  AppleMIDI.run();

  yield();

  // send a note every second
  // (dont cáll delay(1000) as it will stall the pipeline)
  if ((millis() - t0) > 1000)
  {
    t0 = millis();
    //Serial.print(".");

    yield();
    int note = 45;
    int velocity = 55;
    int channel = 1;

    AppleMIDI.noteOn(note, velocity, channel);
    yield();
    AppleMIDI.noteOff(note, velocity, channel);
  }
}

// ====================================================================================
// Event handlers for incoming MIDI messages
// ====================================================================================

// -----------------------------------------------------------------------------
// rtpMIDI session. Device connected
// -----------------------------------------------------------------------------
void OnAppleMidiConnected(char* name) {
    Serial.print("Connected to session ");
    Serial.println(name);
}

// -----------------------------------------------------------------------------
// rtpMIDI session. Device disconnected
// -----------------------------------------------------------------------------
void OnAppleMidiDisconnected() {
    Serial.println("Disconnected");
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void OnAppleMidiNoteOn(byte channel, byte note, byte velocity) {
  Serial.print("Incoming NoteOn from channel:");
  Serial.print(channel);
  Serial.print(" note:");
  Serial.print(note);
  Serial.print(" velocity:");
  Serial.print(velocity);
  Serial.println();
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void OnAppleMidiNoteOff(byte channel, byte note, byte velocity) {
  Serial.print("Incoming NoteOff from channel:");
  Serial.print(channel);
  Serial.print(" note:");
  Serial.print(note);
  Serial.print(" velocity:");
  Serial.print(velocity);
  Serial.println();
}
```

rubenbroman, 2015-08-05:
I just read some of the closed issues and realised that the problem might be that I'm using IDE 1.6.5 and not IDE 1.0.*. Could that be it? I'm on OS X Yosemite by the way.

I'll do some poking around with Wireshark tomorrow and I'll post any findings.

lathoub, 2015-08-05:
Hi,
I recently tested the lib with 1.6.5 - so you should be fine.
Just reading through the log, it might be input buffer size related. I have no ESP8266 (maybe i should get one :-)) to test.
Please post your wireshark results, alternatively, you can compile using verbose debugging switch on (see APPLEMIDI_DEBUG_VERBOSE in setting header file)

rubenbroman, 2015-08-06:
Thanks for your quick answer.

I really think you should get a ESP8266, they're dirt cheap and there's plenty of things you can do with them. 

I switched on the verbose debugging and it looks like the protocolBuffer contains the bytes that are sent but that they're not correct RTP data, given that it's the "unsupported version 3". I uncommented the `return 0;` in the version check so that you have some more info if that helps.

I'm really grateful that you've made this library and that you're supporting it. It would be great if we could get it working on the ESP8266 (hoping that the latency will be good enough). Thanks!

Here's first the Serial output and underneath is the Wireshark data.

```
Getting IP address...
.........
Connected to rutgerfuchsgatan9.se
IP address: 10.0.1.170

IP address is 10.0.1.170.
OK, now make sure you an rtpMIDI session that is Enabled
Add device named Arduino with Host/Port 10.0.1.170:5004
Then press the Connect button
Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes
Starting Verbose logging
Sending NoteOn/Off of note 45, every second
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
Incoming buffer of 27 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 33 3A B1 5 51 73 11 8E 4D 61 63 42 6F 6F 6B 50 72 6F 0 
Protocol buffer contains 27 bytes. Content:
FF FF 49 4E 0 0 0 2 33 3A B1 5 51 73 11 8E 4D 61 63 42 6F 6F 6B 50 72 6F 0 
dissect_apple_midi 5004, packetSize is 27
> Control Invitation: peer = "MacBookPro"���?a� @���?\��?���?���?���?\��?���?���?���?���?\��?yV @5173118E�?\��?���?���?���?\��?���?���?���?���?\��?yV @333AB105
< Control InvitationAccepted: peer = "test"���?a� @���?4F5FDC11�?333AB105 ,in slot = 0
Purging left 27 bytes 
Incoming buffer of 27 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 33 3A B1 5 51 73 11 8E 4D 61 63 42 6F 6F 6B 50 72 6F 0 
Protocol buffer contains 27 bytes. Content:
FF FF 49 4E 0 0 0 2 33 3A B1 5 51 73 11 8E 4D 61 63 42 6F 6F 6B 50 72 6F 0 
5005, packetSize is 27
dissect_rtp
unsupported version 3
Seq Number: 494E
Timestamp : 2
Sync Src  : 333AB105
Have all the data
payload_type is 127
csrc_count = 15
extension hdr_extension_id = 0
hdr_extension = 0
padding = 
Nothing processed
Y�?i��?0
dissect_apple_midi 5005, packetSize is 27
> Content Invitation: peer = "MacBookPro"���?a� @g��?\��?���?$��?���?\��?���?$��?���?��?\��?�V @5173118E�?\��?���?$��?���?\��?���?$��?���?��?\��?�V @333AB105
< Content InvitationAccepted: peer = "test"���?a� @���?4F5FDC11�?333AB105 ,in slot = 0
Connected to session MacBookPro
Purging left 27 bytes 
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 51 73 11 8E 0 46 F0 BB 0 0 0 0 1 EF 58 A8 0 0 0 0 40 46 F0 BB D7 7F 0 0 0 0 F0 BB 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 51 73 11 8E 0 46 F0 BB 0 0 0 0 1 EF 58 A8 0 0 0 0 40 46 F0 BB D7 7F 0 0 0 0 F0 BB 
?�K@���?5005, packetSize is 36
dissect_rtp
[... 866 lines cut ...]
Ethernet II, Src: Espressi_fb:1b:73 (18:fe:34:fb:1b:73), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Internet Protocol Version 4, Src: 10.0.1.170 (10.0.1.170), Dst: 0.0.0.0 (0.0.0.0)
User Datagram Protocol, Src Port: 5005 (5005), Dst Port: 0 (0)
Data (16 bytes)

0000  ff ff ff ff ff ff 18 fe 34 fb 1b 73 08 00 45 00   ........4..s..E.
0010  00 2c 00 20 00 00 ff 11 af f7 0a 00 01 aa 00 00   .,. ............
0020  00 00 13 8d 00 00 00 18 03 eb 80 61 00 13 00 00   ...........a....
0030  00 00 4f 5f dc 11 03 80 2d 37                     ..O_....-7

No.     Time           Source                Destination           Protocol Length Info
    684 65.602120      10.0.1.18             10.0.1.170            AppleMIDI 58     End Session

Frame 684: 58 bytes on wire (464 bits), 58 bytes captured (464 bits) on interface 0
Ethernet II, Src: Apple_13:a7:ce (80:e6:50:13:a7:ce), Dst: Espressi_fb:1b:73 (18:fe:34:fb:1b:73)
Internet Protocol Version 4, Src: 10.0.1.18 (10.0.1.18), Dst: 10.0.1.170 (10.0.1.170)
User Datagram Protocol, Src Port: 5004 (5004), Dst Port: 5004 (5004)
Apple Network-MIDI Session Protocol

0000  18 fe 34 fb 1b 73 80 e6 50 13 a7 ce 08 00 45 00   ..4..s..P.....E.
0010  00 2c 3f dd 00 00 40 11 24 29 0a 00 01 12 0a 00   .,?...@.$)......
0020  01 aa 13 8c 13 8c 00 18 1c 8e ff ff 42 59 00 00   ............BY..
0030  00 02 00 00 00 00 51 73 11 8e                     ......Qs..

```

lenkost, 2015-08-11:
Hi,
I am working on a similar project using this good library with ESP8266.
Had the same problem. 

in WiFiUdp.h we have   #define UDP_TX_PACKET_MAX_SIZE 8192
in EthernetUdp.h we have   #define UDP_TX_PACKET_MAX_SIZE 24

Changing the UDP_TX_PACKET_MAX_SIZE to 24 in WiFiUdp.h fixes the problem with receiving midi.
 No gibberish anymore.

But I still cannot send midi out. 
Messages get all the way to: 
internalSend(Session_t* session, MidiType inType, DataByte inData1, DataByte inData2, Channel inChannel)

But nothing is received on a computer.
I would appreciate any help.

lathoub, 2015-08-11:
I have a ESP8266 now, but now I need a 3.3V Serial-USB programmer...

Good that got the buffer size figured out (I had a hunch it was that).

On Windows, I have rtpMIDI open so I can see the latency timer changing (sign that connection is alive); I also have a MIDI-OX session running, logging both MIDI in and out, and Wireshark, logging all rtpMIDI related traffic.

Do i get it right that you don't see any outgoing bytes on Wireshark?

lenkost, 2015-08-11:
For some reason WiFiUDP does not remember the last RemoteIP and Port. 
If I set it up manually, the sample sketch works OK. This is temporay, for testing only:

	void beginWrite(WiFiUDP* udp)
	{
		byte remIP[] = {0xc0, 0xa8, 0x0, 0x3};
		uint16_t remPort = 5005;
		udp->beginPacket(remIP, remPort);
//		udp->beginPacket(udp->remoteIP(), udp->remotePort());
		udp->write(&ddddd, sizeof(ddddd));

lathoub, 2015-08-11:
Good news! Once I have a 3.3V programmer I will try to get it working, based on the above comments (albeit I have to say that WIFI is not ideal to send MIDI notes - when testing on a portable computer I do notice that notes come in 'bursts' with a little delay. I do not exactly know your use case, but keep the latency in mind

I will try to pass the UDP class as a template argument, that has least impact and max flexibility (I think)

lenkost, 2015-08-13:
I finished my project and it works great. There is no latency at all when using with PC. Thank you for your great library. 
One issue remains. I understand that sysEx part is not finished yet. Is it possible to have it in simplified way? Just to get an array of bytes between F0 and F7 and number of bytes without any parsing of the content? For OnReceiveSystemExclusive event? And the same for sending?
Thank you again.

rubenbroman, 2015-08-13:
Wow!
I'm really happy to see that you've got it working. I haven't had time to do any testing but did a quick search to see if anyone else has had problems with remote ip on up and found this: https://communities.intel.com/message/240509

Is that maybe applicable for us?

Cheers

lenkost, 2015-08-13:
I understand that Remote IP is the IP of the last received packet. I save it in a session structure during sync and then use it to send  packets back. 

As I said it works perfect with Windows but I could not make it work with Mac or iPad. Any ideas?

lathoub, 2015-08-13:
Good to hear it works! Very nice.

I'm considering adding templates to AppleMIDI, so that you can pass the WiFiUDP as a template argument. This way I can support multiple UDP classes (assuming they use the same methods). Would this help you?
Also, what structure did you modify? I could add a void* userData attribute to the struct to support this.

lathoub, 2015-08-15:
Added template support for the AppleMIDI library.
This should make adding other ethernet libraries easier (?).
Can you have a look? See the 3 examples on how to use.
As I don't have a 3.3V programmer I can't test it with the Esp8266

I will close this issue, please open a new issue when you encounter problems

closed 2015-08-15 by lathoub