lathoub/Arduino-AppleMIDI-Library #12: ESP8266 - No valid rtp midi content

rubenbroman, 2015-08-17:
Hi,

I installed the new version of the library and tried the ESP8266 example sketch (there's a typo in the file name, which makes it not show up under examples in the IDE).

It seems there's something wrong with the rtp packets as we're still getting the ´unsupported version 3´ in the ´dissect_rtp´. Attaching the verbose log below.

I hope you'll get your 3.3V programmer soon. 

```
Getting IP address.....................
WiFi connected

IP address is 10.0.1.167
OK, now make sure you an rtpMIDI session that is Enabled
Add device named Arduino with Host/Port 10.0.1.167:5004
Then press the Connect button
Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes
Starting Verbose logging
Sending NoteOn/Off of note 45, every second
< Note On (c=1, n=45, v=55)
< Note Off (c=1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
1, n=45, v=55)
Incoming buffer of 27 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 5B D0 62 C2 96 F2 79 87 4D 61 63 42 6F 6F 6B 50 72 6F 0 
Protocol buffer contains 27 bytes. Content:
FF FF 49 4E 0 0 0 2 5B D0 62 C2 96 F2 79 87 4D 61 63 42 6F 6F 6B 50 72 6F 0 
dissect_apple_midi 5004, packetSize is 27
> Control Invitation: peer = "MacBookPro" ,ssrc 0x96F27987 ,initiatorToken = 0x5BD062C2
< Control InvitationAccepted: peer = "test" ,ssrc 0x4F5FDC11 ,initiatorToken = 0x5BD062C2 ,in slot = 0
Purging left 27 bytes 
Incoming buffer of 27 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 5B D0 62 C2 96 F2 79 87 4D 61 63 42 6F 6F 6B 50 72 6F 0 
Protocol buffer contains 27 bytes. Content:
FF FF 49 4E 0 0 0 2 5B D0 62 C2 96 F2 79 87 4D 61 63 42 6F 6F 6B 50 72 6F 0 
dissect_rtp_midi 5005, packetSize is 27
dissect_rtp
unsupported version 3
Unprocessed packet (No valid rtp midi content).
dissect_apple_midi 5005, packetSize is 27
> Content Invitation: peer = "MacBookPro" ,ssrc 0x96F27987 ,initiatorToken = 0x5BD062C2
< Content InvitationAccepted: peer = "test" ,ssrc 0x4F5FDC11 ,initiatorToken = 0x5BD062C2 ,in slot = 0
Connected to session MacBookPro
Purging left 27 bytes 
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 96 F2 79 87 0 C6 40 33 0 0 0 0 0 FA CF 3C 0 0 0 0 B0 C6 40 33 B4 7F 0 0 0 0 40 33 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 96 F2 79 87 0 C6 40 33 0 0 0 0 0 FA CF 3C 0 0 0 0 B0 C6 40 33 B4 7F 0 0 0 0 40 33 
dissect_rtp_midi 5005, packetSize is 36
dissect_rtp
unsupported version 3
Unprocessed packet (No valid rtp midi content).
dissect_apple_midi 5005, packetSize is 36
> Syncronization for ssrc 0x96F27987, count = 0
< Syncronization for ssrc 0x4F5FDC11, count = 1
Purging left 36 bytes 
kE#@0E�?���?�| @1, n=45, v=55)
 @1, n=45, v=55)
[... 332 lines cut ...]
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 96 F2 79 87 1 0 0 0 0 0 0 0 2 EB AE 40 0 0 0 0 1 0 F6 38 0 0 0 0 0 FF 82 38 
dissect_rtp_midi 5005, packetSize is 36
dissect_rtp
unsupported version 3
Unprocessed packet (No valid rtp midi content).
dissect_apple_midi 5005, packetSize is 36
> Syncronization for ssrc 0x96F27987, count = 1
< Syncronization for ssrc 0x4F5FDC11, count = 2
Purging left 36 bytes 
kE#@0E�?1, n=45, v=55)
1, n=45, v=55)
Incoming buffer of 16 bytes. These will be appended to the protocolBuffer
FF FF 42 59 0 0 0 2 0 0 0 0 96 F2 79 87 
Protocol buffer contains 16 bytes. Content:
FF FF 42 59 0 0 0 2 0 0 0 0 96 F2 79 87 
dissect_apple_midi 5004, packetSize is 16
> End Session for ssrc 0x96F27987, initiatorToken = 0x0
Disconnected
Purging left 16 bytes 
kE#@1, n=45, v=55)
1, n=45, v=55)
kE#@1, n=45, v=55)
1, n=45, v=55)
```

lathoub, 2015-08-17:
Thanks for testing!
The content socket has 2 parsers (1 for AppleMIDI, the other for RTP-MIDI); the unsupported version 3 is from one of the 2 parsers. From what I read from the log, it kinda looks at. Do you see incoming MIDI notes on your Mac? (Alternatively this might be linked to the previous errors mentioned in the related issue: buffer size and remote port and host attribute)

The test above on the ESP8266 means that the UDP socket are accepted by the compiler - that is good news.

I should have my 3.3V programmer this week, so i can test myself.

rubenbroman, 2015-08-17:
Thanks for giving me something to test! :)

I didn't see any incoming MIDI notes on the Mac, and didn't have time to try and send any (didn't have any music software installed). I'll see if I can test that tomorrow.

rubenbroman, 2015-08-18:
I tried sending MIDI to the ESP8266 and after receiving the packets it froze and restarted. 

´´´
Incoming buffer of 16 bytes. These will be appended to the protocolBuffer
80 61 6 A0 F FC A4 18 96 F2 79 87 3 90 2D 37 
Protocol buffer contains 16 bytes. Content:
80 61 6 A0 F FC A4 18 96 F2 79 87 3 90 2D 37 
dissect_rtp_mid
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
´´´

lathoub, 2015-08-18:
Got the programmer and started to test (I have the Sparkfun Thing, what a PITA to work with...). As Lenkost pointed out in the previous issue list, the remoteIP and remoteHost are nulled when no data is available; so the machine needs to explicitly remember the remote IP and port. It now does so in the session struct.

I tested the current code (sending Notes) on Mega and Ethernet Shield and ESP6288 Sparkfun Thing. I have not yet tested receiving notes.

Can you test on your side?

lathoub, 2015-08-19:
MIDI in is not working. The ESP8266 crashes whilst parsing, at various points during the parsing. (Arduino Ethernet has no problem). I will continue to investigate at a later moment in the week

rubenbroman, 2015-08-19:
I tried and came to the same result. Looks like the 12 byte packets are parsed properly by `dissect_rtp` but the 27 resp 36 byte packets gets parsed as the `unsupported version 3`.

```
Incoming buffer of 12 bytes. These will be appended to the protocolBuffer
FF FF 52 53 FE A2 B6 2F 0 20 60 3B 
Protocol buffer contains 12 bytes. Content:
FF FF 52 53 FE A2 B6 2F 0 20 60 3B 
dissect_apple_midi 5004, packetSize is 12
> Receiver Feedback: seq = 32
Purging left 12 bytes 
```

```
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B FE A2 B6 2F 1 0 0 0 0 0 0 0 2C 58 8A 0 0 0 0 0 0 C6 EC C4 0 0 0 0 0 C6 BD 90 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B FE A2 B6 2F 1 0 0 0 0 0 0 0 2C 58 8A 0 0 0 0 0 0 C6 EC C4 0 0 0 0 0 C6 BD 90 
dissect_rtp_midi 5005, packetSize is 36
dissect_rtp
unsupported version 3
Unprocessed packet (No valid rtp midi content).
dissect_apple_midi 5005, packetSize is 36
> Syncronization for ssrc 0xFEA2B62F, count = 1
< Syncronization for ssrc 0x4F5FDC11, count = 2
Purging left 36 bytes 
```

lathoub, 2015-08-19:
don't focus on the unsupprted 3 - the content socket has 2 parsers connected to it, the first parser reports the unsupported version 3, then the other parser tries and does know how to deal with the message (Syncronization in the case above, which is working fine). MIDI out is now working properly! (MIDI in is not for the moment)

rubenbroman, 2015-08-19:
Ok, I see.

I can't see any MIDI coming from the ESP8266 in MIDI Monitor though. Even though the session seems to be up and running as the ESP8266 is killed when I'm sending MIDI to it.

lathoub, 2015-08-19:
I got the MIDI in to work now as well, albeit limited.
In the code, a switch in packet-trp-midi.h (line 1625) crashes if code is included in the source, but not called during execution!! 
It feels like its the compiler going berzerk.

Also, logging in Release mode is different from Debug mode - in Release mode characters are garbled (but the process run correctly).

Not sure what do to here.... confused.

lathoub, 2015-08-19:
I'm looking at conditional compilation as a tmp solution

lathoub, 2015-08-20:
I tried all possible combinations - if I just include a call to a function to a switch, the ESP8266 crashes - even if the function is not called and just contains a return statement! Grrrrr. rubenbroman, can you have a look at the code? packet-trp-midi.h (switch at line 1625)

rubenbroman, 2015-08-25:
Hi,

Sorry for not replying earlier. I've been swamped with work and haven't had time to look at this.

I'm also sorry to say that this is above my skill set, I've been looking thru the code you mentioned and the only thing I notice is that the status bytes are defined different than they should be according to the comments after.

´#define RTP_MIDI_STATUS_CHANNEL_NOTE_OFF 0x08 /* 0x8n n->channel */´

My reading off the comment is that it should be ´0x80´ or ´0x88´ rather than ´0x08´.

But I've jumped in the deep end off the pool here.

Have you tried the http://www.esp8266.com forum? They might have some esp-related info that might give a clue.

seclorum, 2016-03-18:
Hi - great library!  I'm trying to use it with an ESP8266 as well (see http://magicshifter.net/) and running into similar problems.  I've loaded up the ESP MIDI send sample, and am experiencing intermittent crashes of this variety:

```

IP address is 10.0.0.62
OK, now make sure you an rtpMIDI session that is Enabled
Add device named Arduino with Host/Port 10.0.0.62:5004
Then press the Connect button
Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes
Sending NoteOn/Off of note 45, every second
..................................................................................................................................................................................................................................................................................................................................
Exception (9):
epc1=0x40223b1d epc2=0x00000000 epc3=0x00000000 excvaddr=0x80000077 depc=0x00000000

ctx: sys
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000000 00000000 4021e7e5 3ffee5cc
3fffff60:  3ffe9d34 3ffe9df2 40213bfd 3ffecf78
3fffff70:  3ffe9d10 000000f5 3ffe9528 3ffe9528
3fffff80:  00000080 3ffecf78 00000000 3ffef260
3fffff90:  402135a8 3fffdab0 00000000 3fffdcc0
3fffffa0:  3ffe9528 40000f49 3fffdab0 40000f49
<<<stack<<<
```

Note: I added a println(".") to the sample loop() function, so that I could see how often it gets called before the crash occurs - its not the same each time, but rather differs - so I have a 'feeling' this is related to handling of network traffic, perhaps an overflowed buffer or something similar deeper in the AppleMIDI stack ..

I'd really love to add AppleMIDI support to the MagicShifter, so I'm going to be spending some more time debugging this - setting up gd-bserver, and the esp-gdbstub, etc. - so if there is anything else I can do that you might be able to enlighten me about, please let me know and I'll dig deeper.  Would be amazing to have this working and onboard in the MagicShifter!  :)

seclorum, 2016-03-18:
> It feels like its the compiler going berzerk.

Just wanted to say: I 'feel' this way, too.  Perhaps something with the C++ runtime and dynamic functions, or something along those lines ..

closed 2016-03-20 by lathoub