lathoub/Arduino-AppleMIDI-Library #13: Arduino MIDI over Bluetooth LE IOS

jorgertrias, 2015-10-07:
Hello,

I'm doing a midi controller with Arduino and I'm implementing the Bluetooth 4.0 LE model HM-10 module.

Is it possible to convert the whole Arduino + BLE BLE module in a MIDI device to be recognized by an iPad or iPhone?

I've seen your library Arduino WiFi, but not if it can be adapted for this purpose ...

Thanks for your time

lathoub, 2015-10-10:
Hello

Thank you for your question. I do not know how the AppleMIDI protocol, that uses 2 sockets (port 5004 and 5005) can be send over 1 serial Bluetooth connection. Can you point me to documentation or other types of documentation?

lathoub, 2015-10-10:
I just did some searching myself, and it appears that MIDI over Bluetooth does not do any encoding transformations (like AppleMIDI does using RTP MIDI). So the recommended Arduino MIDI library (https://github.com/FortySevenEffects/arduino_midi_library) should work, if you map the bluetooth serial stream to a regular serial stream (never tried, just guessing)

closed 2015-11-29 by lathoub