lathoub/Arduino-AppleMIDI-Library #14: Arduino Micro ESP8266

Shuptuu, 2015-11-03:
Hi,
I would like to create a wireless rtp-midi link between an arduino and my computer, so I was really happy to discover your NoteOnOffEverySecEsp8266 example that I immediatly started to implement, so on an arduino micro with an ESP8266 attached to its serial port, but I can't get it working...
Communication between arduino and esp is ok, I tested others sketchs with success, connected to my network, so no hardware problem.
Analysing the code, my first worry is that there's no definition on how the esp is connected to the arduino, how they can communicate (using Serial1 in my case, with CH_PD connected to pin 2). So I'm not surprise WiFi.begin can't intialize the connection. I tried to add a line to do that using the command provided by the ESP8266 library (ESP8266wifi wifi(Serial1, Serial1, esp8266_reset_pin, Serial);) but that's the same...
Investigations in the libraries is not so easy.
Can you tell me how did you get this sketch working? How did you connect your esp8266 to your arduino? And what has to be adjust to make it working in my configuration?
Thanks for your help.
Olivier

lathoub, 2015-11-03:
Hi 

I tried to create a ESP8266 version, but gave up. The Arduino - ESP8266 combination (I'm using the Sparkfun Thing) is too new and has many teething problems (see the other open issue related to ESP8266).

Depending on your ESP8266 manufacturar/version, connecting is different. Try to google your ESP8266 manufacturar name and Arduino and make the basic samples to work. Good luck!

Shuptuu, 2015-11-03:
Hey, thanks for answering so quick!
Well, I made some of the basic samples to work perfectly! And I'm quite good now with all the AT+ commands ;-)
Is it mandatory to go through the WiFi library? Is there a list of all the functions, variables provided by this library used by AppleMidi?

MusicLeecher, 2015-11-06:
Hi shuptuu
I think you haven't realised yet how to use the NoteOnOffEverySecEsp8266 sketch.. You have to flash that sketch on the ESP8266 and not onto the arduino itself.

Shuptuu, 2015-11-19:
Thanks MusicLeecher,
You're right! I didn't understand this at all!!
I worked on my side, wrote a sketch for an arduino micro, with the ESP attached to its serial port. And it works! I'm able to connect to my PC running rtpMIDI, open rtpmidi link and send midi messages. The only thing is that the latency I get is quite big, changing all the time from 10ms (which is good) to 400ms... making unusable for realtime music playing.
I would say it's coming from the ESP, not the arduino. So I will investigate now on the ESP side, maybe writing my own sketch for the esp...
I can send my arduino sketch if somebody wants.

MusicLeecher, 2015-11-19:
Hi!
I'm glad I cloud help in some way! I'm currently developing a MIDI device too with the ESP and I'm  struggling too o get a good timing with it
I would definitely like to see what you've come up with!

Shuptuu, 2015-11-20:
Hi!
Great to find somebody with similar project!
Are you already working directly on the esp? Cause thanks to your message I was quite confident that moving all the rtp session management (invitation, synchronization) to the ESP will improve the peformance as long as the esp is running at 80MHz. Also that would reduce the exchanges on the serial port between the arduino and the esp which are quite slow. But well, now, I don't know...
Here is what I done: 
[PedalBoardWifiRtpMidi-v1.0.txt](https://github.com/lathoub/Arduino-AppleMidi-Library/files/39902/PedalBoardWifiRtpMidi-v1.0.txt)
Would like to see too what you've done!

Shuptuu, 2015-11-26:
That's it! I wrote a sketch for the esp, able to receive some midi messages on its serial port and to send them to a remote rtp-midi server through the wifi. The full rtp-midi protocol is not implemented and you can't send any kind of midi message, but I don't need more than this and it works perfectly! Between 1 and 5ms for the latency: perfect for realtime applications.
Here is the code:
[ESP01_rtpmidi_client.txt](https://github.com/lathoub/Arduino-AppleMidi-Library/files/45248/ESP01_rtpmidi_client.txt)

MusicLeecher, 2015-12-01:
I gave a quick look through your code and it looked to me to be very good for your what you are trying to accomplish!! With some modifications it may even better suit my application better than lathoub library, because unfortunately this library is quite unstable on the ESP, I got it to work quite fine, although not perfect.. I'll definitely try to modify with your code to see if it suits better my needs!
You should push it to github or something like that! ;)

closed 2016-03-20 by lathoub