lathoub/Arduino-AppleMIDI-Library #15: Bitrate Limit

Hackmancoltaire, 2016-01-14:
There seems to be hooks to take action when a bitrate limit command is received by the client. However, there doesn't seem to be anything that tells the initiator of a session to limit ITS bitrate when sending data to the client (in this case the Arduino). This can get the client into trouble if the initiator sends way more data than the buffer can handle, which happens when using a player program (such as Rondo on the Mac). I've seen packets get upwards of 700 bytes because it is sending so much data about all the simultaneous keys changes, pitch, and speed. The quick workaround was to increase the buffer sizes to prevent a crash, but automatically setting the bitrate limit to the size of the buffer would seem appropriate.

This problem wouldn't come up in cases where only a few note changes are sent per packet. Only in more extreme cases like using a program to send data to the client. I use this library to power a player piano conversion that I built. (https://vimeo.com/151462878)

Hackmancoltaire, 2016-01-16:
I think I've figured out how to set the bitrate limit, immediately after the invitation exchange. I got it to work a couple times and successfully send the limit command to the inviter, but for some reason I'm having issues with packets not terminating when the udp.endPacket() method is called. The results are that the first packet get's padded with a ton of extra packet data from subsequent invitation acknowledgments. Sometimes it doesn't send the invitation acceptation until the session has ended. I'm still trying to track this down.

Hackmancoltaire, 2016-01-16:
Here's an example of the wireshark data I'm seeing.

[image]

Hackmancoltaire, 2016-01-16:
Eventually I'll be submitting a pull request, but you can check out my changes here:
https://github.com/Hackmancoltaire/Arduino-AppleMidi-Library

Hackmancoltaire, 2016-01-19:
Strangely a full reset of the Arduino and Ethernet shield cleared up that weird behavior. Bitrate limit is being sent, but I'm not seeing it have an effect on the size of the packets that my host machine is sending. I'm wondering if the onus of adjusting the bitrate of the packets is the duty of the host program or the framework on the host machine. The packet is being successfully sent though.

lathoub, 2016-02-18:
With the issue resolved (by yourself :-)), I assume its OK to close this issue.

closed 2016-02-18 by lathoub