lathoub/Arduino-AppleMIDI-Library #18: Arduino rebooting after sending note on

admblake, 2016-03-05:
Greetings,

I am in the process of porting an existing project that was written for serial MIDI.  My program looks at incoming MIDI data from ProTools that is in the Mackie HUI format (which is MIDI data with lots of control change and sysex bulks).  

ProTools sends a HUI "are you there?" message on Channel 1, note 0, velocity 0.  The arduino answers back on the same channel and note with a velocity of 127.

When lines 98 and 99 of my program are commented out, the arduino functions as expected.  When lines 98 and 99 are compiled, it seems the arduino keeps rebooting itself or is jumping back to the beginning of setup().

I've tried running this program on both my mac pro and mac book pro and I am getting the same results.

Please let me know if there is any additional info I can provide.  Thank you.

[AppleMIDIrev02.ino.zip](https://github.com/lathoub/Arduino-AppleMidi-Library/files/159983/AppleMIDIrev02.ino.zip)
[image]
[image]
[HUI.pdf](https://github.com/lathoub/Arduino-AppleMidi-Library/files/159984/HUI.pdf)

admblake, 2016-03-15:
My code is attached in the previous e-mail as a zip file.  Is there any more information I can provide to help resolve this issue?

lathoub, 2016-03-19:
I tried the code from the zip file and it runs just fine (with line 98 non commented). I did however had to comment out the `if` statement on line 96 (i can't send notes at velocity 0 with my keyboard).

My environment: Windows 10 with rtpMIDI, Arduino Mega and Ethetnet shield on Arduino.cc IDE 1.6.8

lathoub, 2016-03-19:
I believe it would be good to move `AppleMIDI.begin("arduino");` after the code that prints the syncronisation source. That way you see the actual SSRC (now 0x0 is printed).
(I don't think this explains the above behavior)

Hackmancoltaire, 2016-04-15:
It's quite possible that the crash is being caused by something outside of your sketch. Protools may be sending a packet that isn't being processed properly and so it's crashing with nothing to do with your code. Try removing all of your note sending code and just do the invitation of your remote Protools machine to the session, and if possible use Wireshark to capture the traffic.

lathoub, 2016-05-22:
admblake, were you able to make progress on this issue?
Thanks

admblake, 2016-05-22:
Sadly my project has been on hold for a while.  Feel free to close this issue. 

best,

-Adam Michalak
http://www.adamichalak.com
310-261-9486

lathoub, 2016-05-22:
Thanks Adam. Lets hope you can co;e back to it. Good luck

closed 2016-05-22 by lathoub