lathoub/Arduino-AppleMIDI-Library #23: Midi clock support

rolimat, 2016-05-13:
Hi, 

first of all thanks for this amaizing library. It's working perfect on my NodeMCU dev kit from seeedstudio ;)

Just a question, is Midi clock supported? I don't receive any clock tick from my computer. Browsing your code I couldn't find where the callbacks OnReceiveClock or OnReceiveStart are called :(

Any thoughts? Thanks!

Hackmancoltaire, 2016-05-14:
The library is an rtpMidi implementation, which has its own clock synchronization calls. I think depending on the host product it may or may not send tick messages, but those tick messages aren't a standard part (at least to my knowledge) of the rtpMidi specification. What software are you using that generates these tick messages?

rolimat, 2016-05-15:
Hi Ramon! Im trying to sync my ESP8266 with Traktor (a Dj software un case
you dont know about it).

I just found the code I have to modify, somewhere in the disection  of the
RTP packet there is a check for 0xf8 byte which is exactly the midi clock
tick byte.

This is an standard, every midi software that is able to sync with others
send 0xf8 as midi clock sync.

Thanks!

Bluebie, 2016-07-22:
MIDI Clock support will be important to me in a project soon, so I'll probably work on bringing it about in a month or two if this isn't resolved by then.

@Hackmancoltaire The MIDI Beat Clock is not related to RTP timestamps or synchronisation. It occurs at a different layer and has entirely different purpose. MIDI Beat Clock syncs controllers, sequencers, and synths, to the same beat, so realtime performances and playback can be synchronised with a master clock's beat. It's useful for indicator lights to help musicians keep time.

rolimat, 2016-07-27:
Hi @Bluebie I found that MIDI Clock wasn't completelly implemented but with some minor changes I added the functionality to execute MIDI Clock related callbacks. It was quite inestable when I did changes, jitter made it to unsync but after your changes where you removed the buffering it works like a charm :) Now, as far as I've tested it remains stable and synced for at least 30 minutes more or less (I haven't been able to test it longer).

Anyway, MIDI clock continues having a problem which is the delay at start, I mean, the time since you start midi clock in your host (traktor in my case) and ESP receives the first midi clock tick. This delay is big and it's very difficult (I could say almost impossible) to receive the first tick on time.

Please, find here the changes I did: https://github.com/rolimat/Arduino-AppleMidi-Library

My fork isn't updated cause I saw that with latest changes this repo is unstable. If @lathoub and everybody agree I can do a pull request with my changes :)

Thanks!

PS: probably my code could be better but Im starting to seriously programm on C/Arduino, I use to work using Javascript and Java/Android

theudas, 2016-11-07:
Can you provide an example how to use your midi clock approach?

I cannot find a way to synchronize my microcontroller to ableton :/

lathoub, 2017-08-13:
I have added support for real-time messages, based on @rolimat 's code and description in [https://www.midi.org/specifications/item/table-1-summary-of-midi-message](url)

Tested using MIDI-OX MIDI Sync Transport window

sample code in examples folder [https://github.com/lathoub/Arduino-AppleMidi-Library/tree/master/examples/EthernetShield_RealTimeMessages](url)

Works?

closed 2017-08-17 by lathoub