lathoub/Arduino-AppleMIDI-Library #24: Class instantiation

DBrenchley, 2016-05-15:
Sir(s)
I'm sure this is an user issue vs code .. but any guidance you'd provide would be greatly appreciated.
Attempting to run the WiFi note on example .. target is Ardunio.cc Zero w/ WiFi 101 shield.
I've modified the example code to #include WiFi101.h vs WiFi.h
Use of the example creates a brick .. 
If I comment out 
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI); // see definition in AppleMidi_Defs.h
WiFi is established .. 
any recommendations as to how to debug this would be greatly appreciated !!

lathoub, 2016-05-15:
Hi
I have never tried with the new Wifi101 shield. Could it be memory (how much free space?) or pin related (other pins that you use in your sketch)

DBrenchley, 2016-05-15:
The only content of sketch is the example code. The binary w/o any optimization is 29K 11% of available memory. Pins required by the 101 are 5, 6, 7, 10.
I've attempted to debug but I can't get any breakpoints to function in any headers for some reason yet undetermined .. it's as though it calls the applemidi class instantiation and goes off to never never land .. I'm at a total loss .. and with my quite limited C++ experiance ... that's lost indeed ..
BTW .. not sure this is the appropriate forum ? I'm sure it's not an "issue" .. if you'd care to take the discussion else where david.s.brenchley@gmail.com .. THANK YOU so much

lathoub, 2016-05-15:
I don't have a Zero or the new Wifi101 shield, so i'm not going to be any help.
What IDE do you use? (you mention breakpoints)

lathoub, 2016-05-15:
(the title of the issue is misleading - "Example doesn't work with Zero/Wifi101 shield" is closer to reality

DBrenchley, 2016-05-15:
Using AtmelStudio w/ Visual Micro .. could you recommend a break point in one of your headers in which the code would be executed with that call  to instantiate the class ?

Hackmancoltaire, 2016-05-17:
Well this just means we should change how the random source ID is generated so it's not dependent on an analog source.

lathoub, 2016-05-22:
Indeed, `srand(analogRead(0)); ` is used to generate somewhat random SSRC's. Removing it should not affect the execution of the code (albeit SSRC's will be less random)

closed 2016-05-22 by lathoub