lathoub/Arduino-AppleMIDI-Library #25: AdafruitUDP

Krenuds, 2016-05-28:
Hey there! 
I'm starting today on porting your wonderful little package over to the WICED feather by adafruit. I'm hoping to get away with something as simple as this:
[image]

But of course, this is the result:
[image]

I've been pouring over your code for the last few days just trying to get everything straight in my head. I'm simply wondering if you could point me in the right direction. I'm seeing malformed packets over Wireshark and I'm assuming that this could be something as simple as a stray datatype somewhere. Is this the case, or should I be digging a little deeper?

DBrenchley, 2016-05-28:
I'm new here .. so I maybe talking out my arse ..

your ports don't match ..

I created a .local session at 5004 .. the remote session at 5004 .. added a
host in the RTP directory window remote IP:5004 .. works for me ..

Krenuds, 2016-05-28:
Hey there, thanks for the reply. Unfortunately the ports aren't the problem. Those screencaps were taken at the end of my testing. I've tried on the correct ports and here's the output from RTP-MIDI

`192.168.1.15:5004: Sending INVITE on control-port to resolved host: 192.168.1.15(5004)
TX-CTRL:Don:192.168.1.15:5004:Invite(Don): FF FF 49 4E 00 00 00 02 17 F5 1F B8 5C EB 22 4F 44 6F 6E 00 
RX-CTRL:Don:192.168.1.15:5004: 00 00 00 02 
Data received is too short for parsing!
RX-CTRL:Don:192.168.1.15:5004: 17 F5 1F B8 
Data received is too short for parsing!
RX-CTRL:Don:192.168.1.15:5004: 59 30 28 1F 
Data received is too short for parsing!
RX-CTRL:Don:192.168.1.15:5004:RTP-Data: 4D 79 20 53 65 73 73 69 6F 6E 00 
We don't understand this message-type: 4D79`

I'm looking into how the packets are being fired at the moment. It would seem that RTP-MIDI is looking for up to a 32 bit packet. I'm not quite sure yet, but it would seem that the feather is shooting many smaller packets rather than one larger one.

lathoub, 2016-05-28:
From the first screenshot, I noticed that the IP address is malformed as well (`3232235791 `in stead of a regular IP v4 `192.168.1.15`).
Could this be big-endian vs little-endian related?

It should not matter if the received packets are small, they are added to a buffer until sufficient data is received.

Also, in Wireshark, set your filter to `rtpmidi && applemidi`

Krenuds, 2016-05-28:
I'm wondering if this is the culprit

`[   175449.953] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don: RX-CTRL:Don:192.168.1.15:5004:RTP-Data: 4D 79 20 53 65 73 73 69 6F 6E 00 
[   175449.958] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don:Don: We don't understand this message-type: 4D79`

lathoub, 2016-05-28:
Is that a message from Wireshark?

Krenuds, 2016-05-28:
No, it's from RTP-MIDI logs

lathoub, 2016-05-28:
Can you point me to the `RTP-MIDI ` tool that does the logging?

lathoub, 2016-05-28:
or are those the RTP-MIDI logs from Wireshark? (I see both RTP-MIDI and AppleMIDI messages in Wireshark)

Krenuds, 2016-05-28:
No, in the advanced tab of RTP-MIDI you can specify where to drop the logs. ie. `c:\errlog.text`

lathoub, 2016-05-28:
Krenuds, I have no clue what `from RTP-MIDI `is (or where the advanced tab is). RTP-MIDI is a protocol, is it also a name of a tool? Help me out

Krenuds, 2016-05-28:
Oohh! Sorry for the confusion. I should have specified; rtpMIDI for windows
[image]

lathoub, 2016-05-28:
Ha, thanks! I know that one :-) - but I don't seem to have the Advanced tab. How did you get that?

Krenuds, 2016-05-28:
[http://www.tobias-erichsen.de/wp-content/uploads/2015/08/rtpMIDISetup_1_1_6_238.zip](url)

Latest release?

lathoub, 2016-05-28:
Indeed, the latest release has an "Advanced tab" - cool

Krenuds, 2016-05-28:
I just noticed this, maybe it's something? 
the SSRC  is different after the initiation. It starts out as 0xEBA54E05  but then in the next line it's transmitting 0x4F5FDC11. Could this be something? 

`> (OnControlInvitation) Control Invitation: peer = "test" ,ssrc 0xEBA54E05 ,initiatorToken = 0xD6E7C032
Session for host does not exist.
New Local Session in slot 0 with SSRC EBA54E05
< (OnControlInvitation) Control InvitationAccepted: peer = "test" ,ssrc 0x4F5FDC11 ,initiatorToken = 0xD6E7C032
`

Krenuds, 2016-05-29:
I've done a little more digging. Here's the chain of events so far.

I've started tracing the path between the feather and rtpMidi. Here's the connection log on the feathers side

`Getting IP address...
WiFi connected

IP address is Local IP    : 192.168.0.154

OK, now make sure you an rtpMIDI session that is Enabled
Add device named Arduino with Host/Port Local IP    : 192.168.0.154

:5004
Then press the Connect button
Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes
Starting Verbose logging
Sending NoteOn/Off of note 45, every second
Incoming buffer of 26 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 A9 E2 44 7F EB A5 4E 5 41 6C 69 65 6E 77 61 72 65 0 
Protocol buffer contains 26 bytes. Content:
FF FF 49 4E 0 0 0 2 A9 E2 44 7F EB A5 4E 5 41 6C 69 65 6E 77 61 72 65 0 
dissect_apple_midi 5004, packetSize is 26
21
Version
Initiator
SSRC
Name
Dissector
5004
> (OnControlInvitation) Control Invitation: peer = "Alienware" ,ssrc 0xEBA54E05 ,initiatorToken = 0xA9E2447F
Session for host does not exist.
New Local Session in slot 0 with SSRC EBA54E05
< (OnControlInvitation) Control InvitationAccepted: peer = "Feather Controller" ,ssrc 0x4F5FDC11 ,initiatorToken = 0xA9E2447F ,in slot = 0
OnInvitation
Purging left 26 bytes. New _protocolBufferIndex: 4
Incoming buffer of 26 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 A9 E2 44 7F EB A5 4E 5 41 6C 69 65 6E 77 61 72 65 0 
Protocol buffer contains 26 bytes. Content:
FF FF 49 4E 0 0 0 2 A9 E2 44 7F EB A5 4E 5 41 6C 69 65 6E 77 61 72 65 0 
dissect_apple_midi 5004, packetSize is 26
`
That section loops in sync with the output from rtpMIDI:
`[    12751.680] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: Don Session:192.168.0.154:5004: Trying to resolve host by DNS:192.168.0.154
[    12752.796] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: Don Session:192.168.0.154:5004:  Peer mac is: 44-39-C4-EB-BC-BF
[    12752.798] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: Don Session:192.168.0.154:5004: Sending INVITE on control-port to resolved host: 192.168.0.154(5004)
[    12752.800] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: TX-CTRL:Don Session:192.168.0.154:5004:Invite(Alienware): FF FF 49 4E 00 00 00 02 DE F5 8B B1 EB A5 4E 05 41 6C 69 65 6E 77 61 72 65 00 
[    12753.005] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: FF FF 
[    12753.009] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.013] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 4F 4B 
[    12753.017] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.021] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 00 00 00 02 
[    12753.025] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.029] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: DE F5 8B B1 
[    12753.033] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.037] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 4F 5F DC 11 
[    12753.041] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.045] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004:RTP-Data: 46 65 61 74 68 65 72 20 43 6F 6E 74 72 6F 6C 6C 65 72 00 
[    12753.048] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session:Don Session: We don't understand this message-type: 4665`

Interestingly, the misunderstood message is the session name coming back from the feather itself, namely: 
`(OnControlInvitation) Control InvitationAccepted: peer = "Feather Controller"`

in this case, `46 65 61 74 68 65 72 20 43 6F 6E 74 72 6F 6C 6C 65 72` == `"Feather Controller"`

lathoub, 2016-05-29:
What traffic does Wireshark report? (Set filter to `applemidi || rtpmidi  `) That gives a better overview of the actual bytes transmitted

Krenuds, 2016-05-29:
[image]

Krenuds, 2016-05-29:
But also: 
[image]

Krenuds, 2016-05-29:
So it would seem that even Wireshark isn't reporting proper `rtpmidi` data. Could this be where to start looking?

lathoub, 2016-05-29:
I have noted there is an issue with session names longer than 16 characters. Try reducing the session name on the Arduino to "Feather". 

So replace `AppleMIDI.begin("Feather Controller"); ` with `AppleMIDI.begin("Feather");
`

[image]

Krenuds, 2016-05-29:
Unfortunately the output was the same.

lathoub, 2016-05-29:
I have found a 16 char session name limitation in this library (in 3 files that deal with the invitation). I will investigate how to fix this issue (either by allowing longer session names or clipping the name to a certain size)

Krenuds, 2016-05-29:
Although I added some debug info here:

template<class UdpClass>
```
inline void AppleMidi_Class<UdpClass>::OnInvitation(void* sender, AppleMIDI_Invitation& invitation)
{
	Dissector* dissector = (Dissector*) sender;

	if (dissector->_identifier == Port)
		OnControlInvitation(sender, invitation);
		Serial.printf("Control Dissector ID: %d\n", dissector->_identifier);
	if (dissector->_identifier == Port + 1)
		OnContentInvitation(sender, invitation);
		Serial.printf("Command Dissector ID: %d\n", dissector->_identifier);
}
```

Here is the output:

```
< (OnControlInvitation) Control InvitationAccepted: peer = "Feather" ,ssrc 0x4F5FDC11 ,initiatorToken = 0xFB5CEF6B ,in slot = 0
Control Dissector ID: 5004
Command Dissector ID: 5004
```

Krenuds, 2016-05-29:
Attached debug info here:
```
template<class UdpClass>
inline void AppleMidi_Class<UdpClass>::OnInvitation(void* sender, AppleMIDI_Invitation& invitation)
{
	Dissector* dissector = (Dissector*) sender;

	if (dissector->_identifier == Port)
		OnControlInvitation(sender, invitation);
		Serial.printf("Control Dissector ID: %d\n", dissector->_identifier);
	if (dissector->_identifier == Port + 1)
		OnContentInvitation(sender, invitation);
		Serial.printf("Command Dissector ID: %d\n", dissector->_identifier);
}
```
Output:
```
< (OnControlInvitation) Control InvitationAccepted: peer = "Feather" ,ssrc 0x4F5FDC11 ,initiatorToken = 0xFB5CEF6B ,in slot = 0
Control Dissector ID: 5004
Command Dissector ID: 5004
```

lathoub, 2016-05-29:
I see from your screenshots that the session Invitation acceptance is never send from the Arduino to your PC.

Have you tried more basic Adafruit Feather WICED examples that use UDP? I feel the issue might be related to packet size (see above noted by yourself)

DBrenchley, 2016-05-29:
Just so stupid me is clear ..

On the Feather your session name is Don Session / it's IP is 192.168.0.154
using port 5004

The host 192.168.0.198 "Alienware" is that the platform host name or the
session name in RTPmidi?

Did you perform actions noted in attached screen shot.

My environment ..

Local (Host) session name Enterprise_1
Remote session Name Arduino_1

[image: Inline image 1]

Krenuds, 2016-05-29:
Here is my session setup:
[image]

Krenuds, 2016-05-29:
> I see from your screenshots that the session Invitation acceptance is never send from the Arduino to your PC.
> 
> Have you tried more basic Adafruit Feather WICED examples that use UDP? I feel the issue might be related to packet size (see above noted by yourself)

It looks as though the feather is sending SOME kind of information. Is this not the acceptance?

lathoub, 2016-05-29:
`??? `indicate that the session is not in a correct state. In this case the `Don Session `sends an invitation to the `Feather Controller `but never receive an `Invitation Accepted` (as per the above logs)

Krenuds, 2016-05-29:
What about this? I've changed the session name to Feather in this case, but this has been firing every time.
```
< (OnControlInvitation) Control InvitationAccepted: peer = "Feather" ,ssrc 0x4F5FDC11 ,initiatorToken = 0x477E3B20 ,in slot = 0
Control Dissector ID: 5004
Purging left 24 bytes. New _protocolBufferIndex: 4
```

lathoub, 2016-05-29:
Hmm, where does `Alienware `come from? Your rtpMIDI session name is `Don Session`. Your wireshark reported that `Alienware `started the session, but your session is named `Don Session` (so I'm expecting `Don Session `on Wireshark). And you also have `Enterprise_1` and `Arduino_1` (confused)

DBrenchley, 2016-05-29:
I'll defer to lathoub ..

but the way I got it to work was via the actions I indicated in the screen
shot ..
add the remote session name IP and port into the directory and press connect

BTW .. I was unable to read the text of your message the screen shot
overlayed it ..

one other thing whether it makes a difference or not .. my local session
name and bonjuor name are the same

lathoub, 2016-05-29:
Hold on, i see: `Alienware `is the bonjour name, your session name is `don session`. I'm used that both are the same (unconfused)

Krenuds, 2016-05-29:
Haha I'm so sorry man. I've been poking around like crazy and changing things here and there. Here's my current session details and the output:
**rtpMidi Setup:**
[image]

**rtpMidi Log:**
[image]
**Feather Serial Log:**
[image]
**Wireshark Sample**
[image]

lathoub, 2016-05-29:
The wireshark logging is interesting as it indicates malformed packets.

Could it be related to  `UDP_TX_PACKET_MAX_SIZE` defined in AppleMIDI_Defs.h? Maybe the feather wants more? (or less?)

Krenuds, 2016-05-29:
I'll look into it! Thanks for the tip!

DBrenchley, 2016-05-29:
Getting closer it seems ..

FeatherWifi is the session name in your code?

Krenuds, 2016-05-29:
No, FeatherWifi is the nickname for my feather in rtpMidi

DBrenchley, 2016-05-29:
Make it your session name in the code

Krenuds, 2016-05-29:
No dice... 

[image]

DBrenchley, 2016-05-29:
FeatherWifi is no longer in the Directory ???

It should be there .. hilight and press connect ..

Krenuds, 2016-05-29:
It moves from the directory to the connection panel. It's all set up correctly I believe

DBrenchley, 2016-05-29:
K .. attached is my RTPMidi log perhaps you folks can glean something from
it ..

I learned something new .. and something you should watch for .. my dhcp
lease for the Arduino expired and  it got a new IP ..

RTPMidi blew up when I attempted to attach .. had to reenter new host/IP

[   347569.006] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: RX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B E2 A6 20 E7 00 00 00 00 00 00 00 02 B7 CF 57 B5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   347569.006] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: TX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B 7F F7 FA D8 01 00 00 00 00 00 00 02 B7 CF 57 B5 00 00 00 00 CF 2A D2 0B 00 00 00 00 00 00 00 00 
[   347569.008] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: RX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B E2 A6 20 E7 02 00 00 00 00 00 00 02 B7 CF 57 B5 00 00 00 00 CF 2A D2 0B 00 00 00 02 B7 CF 57 EB 
[   347569.009] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:Homebase_2: Latency=24
[   347569.332] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 00 00 00 00 00 00 00 00 CF 2A DE C7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   347569.363] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 63 96 68 09 01 00 00 00 00 00 00 00 CF 2A DE C7 00 00 00 00 00 21 6A B0 00 00 00 00 00 00 00 00 
[   347569.364] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 02 00 00 00 00 00 00 00 CF 2A DE C7 00 00 00 00 00 21 6A B0 00 00 00 00 CF 2A E0 06 
[   347569.365] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Latency=319
[   347569.382] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 63 96 68 09 00 00 00 00 00 00 00 00 00 21 6A B0 00 00 00 00 00 21 6A B0 00 00 00 00 CF 2A E0 06 
[   347569.383] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 01 00 00 00 00 00 00 00 00 21 6A B0 00 00 00 00 CF 2A E0 C5 00 00 00 00 00 00 00 00 
[   347569.398] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 63 96 68 09 02 00 00 00 00 00 00 00 00 21 6A B0 00 00 00 00 CF 2A E0 C5 00 00 00 00 00 21 6A B0 
[   347569.398] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Latency=157
[   347575.085] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.15:5005:Synchronize: FF FF 43 4B 68 ED B2 5D 00 00 00 00 00 00 00 02 B7 D0 45 4D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   347575.086] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.15:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 01 00 00 00 00 00 00 02 B7 D0 45 4D 00 00 00 00 CF 2B BF 84 00 00 00 00 00 00 00 00 
[   347575.087] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.15:5005:Synchronize: FF FF 43 4B 68 ED B2 5D 02 00 00 00 00 00 00 02 B7 D0 45 4D 00 00 00 00 CF 2B BF 84 00 00 00 02 B7 D0 45 65 
[   347575.088] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:Homebase_1: Latency=21
[   347578.503] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:192.168.0.14:5004: Trying to resolve host by DNS:192.168.0.14
[   347578.516] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:192.168.0.14:5004:  Peer mac is: E0-B7-B1-11-CF-C2
[   347578.516] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:192.168.0.14:5004: Sending INVITE on control-port to resolved host: 192.168.0.14(5004)
[   347578.517] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: TX-CTRL:USS-Enterprise_2:192.168.0.14:5004:Invite(USS-Enterprise_2): FF FF 49 4E 00 00 00 02 4C 93 E0 DE 7F F7 FA D8 55 53 53 2D 45 6E 74 65 72 70 72 69 73 65 5F 32 00 
[   347579.008] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: RX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B E2 A6 20 E7 00 00 00 00 00 00 00 02 B7 D0 DE 8C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   347579.008] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: TX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B 7F F7 FA D8 01 00 00 00 00 00 00 02 B7 D0 DE 8C 00 00 00 00 CF 2C 58 C2 00 00 00 00 00 00 00 00 
[   347579.010] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: RX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B E2 A6 20 E7 02 00 00 00 00 00 00 02 B7 D0 DE 8C 00 00 00 00 CF 2C 58 C2 00 00 00 02 B7 D0 DE A5 
[   347579.011] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:Homebase_2: Latency=22
[   347579.328] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 00 00 00 00 00 00 00 00 CF 2C 65 3D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   347579.520] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_2:192.168.0.14:5004: Sending INVITE on control-port to resolved host: 192.168.0.14(5004)
[... cut ...]
[   347743.470] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-CTRL:USS-Enterprise_1:192.168.0.3:5004:Invite(USS_Enterprise_1): FF FF 49 4E 00 00 00 02 42 49 5E E5 87 F1 D6 75 55 53 53 5F 45 6E 74 65 72 70 72 69 73 65 5F 31 00 
[   347744.474] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Sending INVITE on control-port to resolved host: 192.168.0.3(5004)
[   347744.474] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-CTRL:USS-Enterprise_1:192.168.0.3:5004:Invite(USS_Enterprise_1): FF FF 49 4E 00 00 00 02 42 49 5E E5 87 F1 D6 75 55 53 53 5F 45 6E 74 65 72 70 72 69 73 65 5F 31 00 
[   347745.111] rtpMIDI_1_1

Krenuds, 2016-05-29:
So it looks to me as though you are sending an invitation and receiving a Synchronize statement:
`[ 347578.517] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: TX-CTRL:USS-Enterprise_2:192.168.0.14:5004:Invite(USS-Enterprise_2): FF FF 49 4E 00 00 00 02 4C 93 E0 DE 7F F7 FA D8 55 53 53 2D 45 6E 74 65 72 70 72 69 73 65 5F 32 00 [ 347579.008] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_2: RX-MIDI:USS-Enterprise_2:192.168.0.15:5007:Synchronize: FF FF 43 4B E2 A6 20 E7 00 00 00 00 00 00 00 02 B7 D0 DE 8C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

I get fragmented packets:
```
[    12752.796] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: Don Session:192.168.0.154:5004:  Peer mac is: 44-39-C4-EB-BC-BF
[    12752.798] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: Don Session:192.168.0.154:5004: Sending INVITE on control-port to resolved host: 192.168.0.154(5004)
[    12752.800] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: TX-CTRL:Don Session:192.168.0.154:5004:Invite(Alienware): FF FF 49 4E 00 00 00 02 DE F5 8B B1 EB A5 4E 05 41 6C 69 65 6E 77 61 72 65 00 
[    12753.005] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: FF FF 
[    12753.009] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.013] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 4F 4B 
[    12753.017] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.021] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 00 00 00 02 
[    12753.025] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.029] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: DE F5 8B B1 
[    12753.033] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.037] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004: 4F 5F DC 11 
[    12753.041] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session: Data received is too short for parsing!
[    12753.045] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: Don Session: RX-CTRL:Don Session:192.168.0.154:5004:RTP-Data: 46 65 61 74 68 65 72 20 43 6F 6E 74 72 6F 6C 6C 65 72 00 
[    12753.048] rtpMIDI_1_1_6_238 [ DIAG_ERROR   ]: Don Session:Don Session: We don't understand this message-type: 4665
```

DBrenchley, 2016-05-29:
K .. this is getting to deep for me to be of much effective use ..

The included libraries ?? Wifi.h Wifiudp,h ??

I'm using theArdunio Zero .. I had to ref its specific librayies ..

Also just a fyi .. you gave out your Wifi password in your code posting ..
asking if your a musician would be somewhat redundant duh ... MIDI ,,

DBrenchley, 2016-05-29:
I do believe as far as RTPMidi .. we got you setup correctly .. wouldn't
change anything there

DBrenchley, 2016-05-29:
BTW apologies to all .. didn't realize an email attachment would "flood"
the forum ..

I parsed all the other stuff out .. this seems to be the relevant exchange

1
[   348475.352] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Sending INVITE on control-port to resolved host: 192.168.0.3(5004)
[   348475.352] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-CTRL:USS-Enterprise_1:192.168.0.3:5004:Invite(USS_Enterprise_1): FF FF 49 4E 00 00 00 02 42 49 5E E5 87 F1 D6 75 55 53 53 5F 45 6E 74 65 72 70 72 69 73 65 5F 31 00 
[   348476.357] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Sending INVITE on control-port to resolved host: 192.168.0.3(5004)
[   348476.358] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-CTRL:USS-Enterprise_1:192.168.0.3:5004:Invite(USS_Enterprise_1): FF FF 49 4E 00 00 00 02 42 49 5E E5 87 F1 D6 75 55 53 53 5F 45 6E 74 65 72 70 72 69 73 65 5F 31 00 
[   348476.366] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-CTRL:USS-Enterprise_1:192.168.0.3:5004:Accepted(Arduino_1): FF FF 4F 4B 00 00 00 02 42 49 5E E5 63 96 68 09 41 72 64 75 69 6E 6F 5F 31 00 
[   348476.480] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:  peer mac is: F8-F0-05-F0-96-5D
[   348476.480] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Our connection on ctrl-port was accepted
[   348476.559] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Connection to control-port established: 192.168.0.3(5004)
[   348476.559] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Sending INVITE on midi-port to resolved host: 192.168.0.3(5005)
[   348476.560] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Invite(USS_Enterprise_1): FF FF 49 4E 00 00 00 02 42 49 5E E5 87 F1 D6 75 55 53 53 5F 45 6E 74 65 72 70 72 69 73 65 5F 31 00 
[   348476.567] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Accepted(Arduino_1): FF FF 4F 4B 00 00 00 02 42 49 5E E5 63 96 68 09 41 72 64 75 69 6E 6F 5F 31 00 
[   348476.568] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004:: Our connection on midi-port was accepted
[   348476.569] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-CTRL:USS-Enterprise_1:192.168.0.3:5004:unknown(524C): FF FF 52 4C 04 00 A8 C0 00 00 03 00 
[   348476.569] rtpMIDI_1_1_6_238 [ DIAG_WARNING ]: USS-Enterprise_1: Received Limit from unknown peer: 192.168.0.3(5004)
[   348476.661] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Connection to MIDI-port established: 192.168.0.3(5005)
[   348476.661] rtpMIDI_1_1_6_238 [ DIAG_INFO    ]: USS-Enterprise_1:192.168.0.3:5004: Initial-Sync to peer
[   348476.662] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 00 00 00 00 00 00 00 00 CF B5 51 54 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[   348476.669] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: RX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 63 96 68 09 01 00 00 00 00 00 00 00 CF B5 51 54 00 00 00 00 00 01 D4 C0 00 00 00 00 00 00 00 00 
[   348476.670] rtpMIDI_1_1_6_238 [ DIAG_VERBOSE ]: USS-Enterprise_1: TX-MIDI:USS-Enterprise_1:192.168.0.3:5005:Synchronize: FF FF 43 4B 87 F1 D6 75 02 00 00 00 00 00 00 00 CF B5 51 54 00 00 00 00 00 01 D4 C0 00 00 00 00 CF B5 51 A3

Krenuds, 2016-05-29:
Your init packet is much larger than mine

DBrenchley, 2016-05-29:
Yep got that .. as lathoub stated earlier .. I thinks something in the udp
library it trunking it ???

DBrenchley, 2016-05-29:
Download Atmel Studio 7 and Visual Micro .. it gives you live debugging ..

DBrenchley, 2016-05-29:
oops belay that .. likely not "live" on your platform

Krenuds, 2016-05-29:
It looks as though the packets from adafruit_UDP are getting set up by way of a struct:
```
size_t AdafruitUDP::write(const uint8_t* buffer, size_t size)
{
  if (_udp_handle == 0 || _sndIP == 0 || _sndPort == 0) return 0;

  sdep_cmd_para_t para_arr[] =
  {
      { .len = 4   , .p_value = &_udp_handle },
      { .len = 4   , .p_value = &_sndIP      },
      { .len = 2   , .p_value = &_sndPort    },
      { .len = size, .p_value = buffer       }
  };
  uint8_t para_count = sizeof(para_arr)/sizeof(sdep_cmd_para_t);

  VERIFY_RETURN(sdep_n(SDEP_CMD_UDP_WRITE, para_count, para_arr, NULL, NULL), 0);
  return size;
}
```
sdep_cmd_para_t:

```
typedef struct
{
  uint16_t    len;
  void const* p_value;
}sdep_cmd_para_t;
```
Here is the begin function:
```
uint8_t AdafruitUDP::begin(uint16_t port)
{
  if ( !Feather.connected() ) return 0;

  uint8_t interface = WIFI_INTERFACE_STATION;
  uint32_t this_value = (uint32_t) this;

  sdep_cmd_para_t para_arr[] =
  {
      { .len = 1, .p_value = &interface  },
      { .len = 2, .p_value = &port       },
      { .len = 4, .p_value = &this_value },
  };
  uint8_t para_count = sizeof(para_arr)/sizeof(sdep_cmd_para_t) - (rx_callback == NULL ? 1 : 0);

  return sdep_n(SDEP_CMD_UDP_CREATE, para_count, para_arr, NULL, &_udp_handle);
}
```
Compared to WifiUDP begin:

```
uint8_t WiFiUDP::begin(uint16_t port) {

    uint8_t sock = WiFiClass::getSocket();
    if (sock != NO_SOCKET_AVAIL)
    {
        ServerDrv::startServer(port, sock, UDP_MODE);
        WiFiClass::_server_port[sock] = port;
        _sock = sock;
        _port = port;
        return 1;
    }
    return 0;

}
```
I'm really starting to think that initial error from rtpMidi stating that the packets are too small to decipher is legitimate. I'm also thinking that the adafruit_udp is firing packets at certain invervals. ie... one piece of info at a time.

DBrenchley, 2016-05-29:
Likely a stupid question .. really show my lack of C++ experience ..

can you include wifiudp vs adafruitudp?

Krenuds, 2016-05-29:
No, unfortunately it's very different. Fortunately, lathoubs library accounts for this type of implementation.

Krenuds, 2016-05-29:
[https://forums.adafruit.com/viewtopic.php?f=57&t=97078&p=487418#p487418](Adafruit Forum)

lathoub, 2016-05-29:
Thanks for sharing the posting on adafruit  - fingers x-d.

(FYI, you might be confusing the adafruit folks by mentioning MIDI - the focus should be on how to send UDP packages on the WICED platform)

lathoub, 2016-05-29:
(and change the title to `Feather and UDP`) If not, people will believe the focus is on MIDI

lathoub, 2016-05-29:
pls upgrade to the latest, when in `DEBUG` mode (see in `AppleMidi_Settings.h`), UDP errors are reported (ignored in RELEASE mode (and do not use memory)). Do you see any?

(Checks based on the documentation from https://learn.adafruit.com/introducing-the-adafruit-wiced-feather-wifi/adafruitudp)

Krenuds, 2016-05-29:
Morning! Here are the settings I've been using. Correct, yes?:
```
#define DEBUG
//#define RELEASE

#if defined(RELEASE)
#define RELEASE_BUILD
#undef DEBUG_BUILD
#endif

#if defined(DEBUG)
#define DEBUG_BUILD
#undef RELEASE_BUILD
#endif

#if defined(RELEASE_BUILD)
#undef APPLEMIDI_DEBUG
#undef APPLEMIDI_DEBUG_VERBOSE
#endif

#if defined(DEBUG_BUILD)
#define APPLEMIDI_DEBUG			   1
#define APPLEMIDI_DEBUG_VERBOSE	   1
#endif

#define APPLEMIDI_BUILD_INPUT      1
#define APPLEMIDI_BUILD_OUTPUT     1

#define APPLEMIDI_USE_CALLBACKS    1
#define APPLEMIDI_USE_EVENTS       1
```

Still waking up here but I left myself lots of notes last night, I believe I'm on the precipice here!

[22 more replies]

closed 2016-05-31 by lathoub