lathoub/Arduino-AppleMIDI-Library #26: OSX TimeCodeQuarterFrame issue

Krenuds, 2016-06-04:
Hi again,
Been having smashing success with your framework. I can't thank you enough for writing this thing! One new issue however:  when connected to CoreMidi(mac) instead of rtpMidi(windows), mac outputs very large packets and the 8266 gets all bogged down. Eventually it just disconnects.

`ReportedBoundsError 3Purging left 34 bytes. New _protocolBufferIndex: 11`

I'm suspecting that this is a problem with Apples CoreMidi PoS but I need a workaround.

I've also noticed Wireshark reporting a "BitrateReceivedLimit" message. Could this have something to do with it?

Here are the mac logs from APPLEMIDI_DEBUG:

```
Getting IP address........
IP address is 192.168.1.15
Starting Session DonFeather on port 5004
Incoming buffer of 20 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 46 E8 7C CD 93 16 66 8E 6F 74 66 0 
Protocol buffer contains 20 bytes. Content:
FF FF 49 4E 0 0 0 2 46 E8 7C CD 93 16 66 8E 6F 74 66 0 
> (OnControlInvitation) Control Invitation: peer = "otf" ,ssrc 0x9316668E ,initiatorToken = 0x46E87CCD
Invition received for a new session.
New Local Session in slot 0 with SSRC 9316668E
< (OnControlInvitation) Control InvitationAccepted: peer = "DonFeather" ,ssrc 0xF0FC00B5 ,initiatorToken = 0x46E87CCD
...
...
> Content Invitation: peer = "otf" ,ssrc 0x9316668E
< Content InvitationAccepted: peer = "DonFeather" ,ssrc 0xF0FC00B5
Connected to session otf
Purging left 20 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 93 16 66 8E 0 0 0 0 0 0 0 0 1 1B 70 D9 95 7F 0 0 2 0 0 0 95 7F 0 0 0 20 5A 4 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 93 16 66 8E 0 0 0 0 0 0 0 0 1 1B 70 D9 95 7F 0 0 2 0 0 0 95 7F 0 0 0 20 5A 4 
> Syncronization for ssrc 0x9316668E, count = 0
< Syncronization for ssrc 0xF0FC00B5, count = 1
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 93 16 66 8E 2 4 FF 3F 5D 56 20 40 0 0 0 0 1 1B 70 D9 0 0 0 0 0 0 0 0 1 1B 72 B6 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 93 16 66 8E 2 4 FF 3F 5D 56 20 40 0 0 0 0 1 1B 70 D9 0 0 0 0 0 0 0 0 1 1B 72 B6 
> Syncronization for ssrc 0x9316668E, count = 2
< Syncronization for ssrc 0xF0FC00B5, count = 0
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 93 16 66 8E 1 4 FF 3F 5D 56 20 40 0 0 0 0 0 0 0 0 1 1B 74 7 0 0 0 0 0 0 0 0 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 93 16 66 8E 1 4 FF 3F 5D 56 20 40 0 0 0 0 0 0 0 0 1 1B 74 7 0 0 0 0 0 0 0 0 
> Syncronization for ssrc 0x9316668E, count = 1
< Syncronization for ssrc 0xF0FC00B5, count = 2
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
...
...
> TimeCodeQuarterFrame (c=48)
OnMidiReceiveTimeCodeQuarterFrame: 30
> TimeCodeQuarterFrame (c=64)
OnMidiReceiveTimeCodeQuarterFrame: 40
> TimeCodeQuarterFrame (c=80)
OnMidiReceiveTimeCodeQuarterFrame: 50
> TimeCodeQuarterFrame (c=96)
OnMidiReceiveTimeCodeQuarterFrame: 60
> TimeCodeQuarterFrame (c=118)
OnMidiReceiveTimeCodeQuarterFrame: 76
> TimeCodeQuarterFrame (c=8)
OnMidiReceiveTimeCodeQuarterFrame: 08

ReportedBoundsError 3Purging left 34 bytes. New _protocolBufferIndex: 11

Incoming buffer of 45 bytes. These will be appended to the protocolBuffer
80 61 6C 25 1 1B D6 44 93 16 66 8E C0 11 F1 10 54 F1 20 53 F1 30 53 F1 40 54 F1 50 53 F1 60 40 6C 23 8 B 60 60 0 0 6 80 0 0 0 
Protocol buffer contains 56 bytes. Content:
8 B 62 60 0 0 4 60 0 0 0 80 61 6C 25 1 1B D6 44 93 16 66 8E C0 11 F1 10 54 F1 20 53 F1 30 53 F1 40 54 F1 50 53 F1 60 40 6C 23 8 B 60 60 0 0 6 80 0 0 0 
Signature not supported.
Purging left 2 bytes. New _protocolBufferIndex: 54
Incoming buffer of 45 bytes. These will be appended to the protocolBuffer
80 61 6C 26 1 1B D8 38 93 16 66 8E C0 11 F1 76 53 F1 A 53 F1 10 54 F1 20 53 F1 30 53 F1 40 40 6C 23 8 B 66 60 0 0 6 80 0 0 0 
Protocol buffer contains 99 bytes. Content:
62 60 0 0 4 60 0 0 0 80 61 6C 25 1 1B D6 44 93 16 66 8E C0 11 F1 10 54 F1 20 53 F1 30 53 F1 40 54 F1 50 53 F1 60 40 6C 23 8 B 60 60 0 0 6 80 0 0 0 80 61 6C 26 1 1B D8 38 93 16 66 8E C0 11 F1 76 53 F1 A 53 F1 10 54 F1 20 53 F1 30 53 F1 40 40 6C 23 8 B 66 60 0 0 6 80 0 0 0 
```
And of course, windows rtpMidi purs like a kitten:

```
Getting IP address........
IP address is 192.168.1.15
Starting Session DonFeather on port 5004
pm open,type:2 0
Incoming buffer of 29 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 4D 6E 75 6E 3A D2 6C 81 44 6F 6E 74 65 68 66 72 69 6E 67 65 0 
Protocol buffer contains 29 bytes. Content:
FF FF 49 4E 0 0 0 2 4D 6E 75 6E 3A D2 6C 81 44 6F 6E 74 65 68 66 72 69 6E 67 65 0 
> (OnControlInvitation) Control Invitation: peer = "Dontehfringe" ,ssrc 0x3AD26C81 ,initiatorToken = 0x4D6E756E
Invition received for a new session.
New Local Session in slot 0 with SSRC 3AD26C81
< (OnControlInvitation) Control InvitationAccepted: peer = "DonFeather" ,ssrc 0xBD48542D ,initiatorToken = 0x4D6E756E
Purging left 29 bytes. New _protocolBufferIndex: 0
Incoming buffer of 29 bytes. These will be appended to the protocolBuffer
FF FF 49 4E 0 0 0 2 4D 6E 75 6E 3A D2 6C 81 44 6F 6E 74 65 68 66 72 69 6E 67 65 0 
Protocol buffer contains 29 bytes. Content:
FF FF 49 4E 0 0 0 2 4D 6E 75 6E 3A D2 6C 81 44 6F 6E 74 65 68 66 72 69 6E 67 65 0 
> Content Invitation: peer = "Dontehfringe" ,ssrc 0x3AD26C81
< Content InvitationAccepted: peer = "DonFeather" ,ssrc 0xBD48542D
Connected to session Dontehfringe
Purging left 29 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 3A D2 6C 81 0 0 0 0 0 0 0 0 0 2A CC 3D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 3A D2 6C 81 0 0 0 0 0 0 0 0 0 2A CC 3D 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
> Syncronization for ssrc 0x3AD26C81, count = 0
< Syncronization for ssrc 0xBD48542D, count = 1
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 3A D2 6C 81 2 0 0 0 5D 56 20 40 0 0 0 0 0 2A CC 3D 0 0 0 0 0 0 0 0 0 2A CD 79 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 3A D2 6C 81 2 0 0 0 5D 56 20 40 0 0 0 0 0 2A CC 3D 0 0 0 0 0 0 0 0 0 2A CD 79 
> Syncronization for ssrc 0x3AD26C81, count = 2
< Syncronization for ssrc 0xBD48542D, count = 0
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 3A D2 6C 81 1 0 0 0 5D 56 20 40 0 0 0 0 0 0 0 0 0 2A CE FE 0 0 0 0 0 0 0 0 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 3A D2 6C 81 1 0 0 0 5D 56 20 40 0 0 0 0 0 0 0 0 0 2A CE FE 0 0 0 0 0 0 0 0 
> Syncronization for ssrc 0x3AD26C81, count = 1
< Syncronization for ssrc 0xBD48542D, count = 2
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 3A D2 6C 81 0 0 0 0 0 0 0 0 0 2B 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 3A D2 6C 81 0 0 0 0 0 0 0 0 0 2B 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
> Syncronization for ssrc 0x3AD26C81, count = 0
< Syncronization for ssrc 0xBD48542D, count = 1
Purging left 36 bytes. New _protocolBufferIndex: 0
Incoming buffer of 36 bytes. These will be appended to the protocolBuffer
FF FF 43 4B 3A D2 6C 81 2 0 0 0 5D 56 20 40 0 0 0 0 0 2B 7 0 0 0 0 0 0 0 0 0 0 2B D 2B 
Protocol buffer contains 36 bytes. Content:
FF FF 43 4B 3A D2 6C 81 2 0 0 0 5D 56 20 40 0 0 0 0 0 2B 7 0 0 0 0 0 0 0 0 0 0 2B D 2B 
```

Hackmancoltaire, 2016-06-04:
I'm not sure OS X is actually respecting the bit rate limit. I added it but never saw a change from CoreMidi. You may want to up your MAX PACKET SIZE and see if that helps. Depending on the host program it might be sending large packets outside the limit.

Krenuds, 2016-06-04:
Interesting, I tried that but it made no difference. I'll start digging around tonight. Is there any way to truncate the incoming packet? It would seem that the protocol buffer is overflowing and causing the ardy to reset.

lathoub, 2016-06-04:
I have a Mac here that I can use for testing. What app do you use to send the TimeCodeQuarterFrame's?

lathoub, 2016-06-04:
Also, what hardware are you using? (Arduino ?, ESP8266, ...)

Krenuds, 2016-06-04:
I tested on Pro Tools 11 but I'm headed to a friends tonight to test it in Logic and Reaper. Also I've tried running it off an Uno with and ESP8266 and a Feather Huzzah ESP8266. Output is identical on both chipsets.

I feel bad posting having not gathered enough debug information. I'll try and have a full report done by the end of this evening. I wouldn't give it much though if I were you, until I've gone through everything on my end.

lathoub, 2016-06-04:
Don't worry Krenuds, thanks for reporting. If you can find a free program to reproduce the issue, that would be super (I don't have ProTools on my Mac)

Krenuds, 2016-06-04:
In the meantime, just out of curiosity - Are sysex callbacks in place? Just wondering, no intention of using them. I just tried one the other day I don't think it kicked anything back.

lathoub, 2016-06-04:
I don't think so - waiting for a good use case

Hackmancoltaire, 2016-06-04:
What did you set the packet size to? And if you look in wire shark what's the general size of the packets?

Krenuds, 2016-06-05:
Hey there, so [I tried the freeware program: Horae](https://sononum.net/horae?dm_campaign=horae-demo-landing) and this was the output when generatring MTC

Wireshark output
[image]

Arduino Serial Output
```
Purging left 2 bytes. New _protocolBufferIndex: 396
Incoming buffer of 47 bytes. These will be appended to the protocolBuffer
80 61 4 64 1 73 6C 81 E 8C 93 3 C0 17 F1 2 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 0 
Protocol buffer contains 443 bytes. Content:
40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 7 80 61 4 5C 1 73 53 81 E 8C 93 3 C0 17 F1 B 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 9 80 61 4 5D 1 73 56 A1 E 8C 93 3 C0 17 F1 D 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F B 80 61 4 5E 1 73 59 C1 E 8C 93 3 C0 17 F1 F 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F D 80 61 4 5F 1 73 5C E1 E 8C 93 3 C0 17 F1 1 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F F 80 61 4 60 1 73 60 1 E 8C 93 3 C0 17 F1 3 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 11 80 61 4 61 1 73 63 21 E 8C 93 3 C0 17 F1 5 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 13 80 61 4 62 1 73 66 41 E 8C 93 3 C0 17 F1 7 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 15 80 61 4 63 1 73 69 61 E 8C 93 3 C0 17 F1 0 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 17 80 61 4 64 1 73 6C 81 E 8C 93 3 C0 17 F1 2 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 0 
Signature not supported.
Purging left 2 bytes. New _protocolBufferIndex: 441
Incoming buffer of 47 bytes. These will be appended to the protocolBuffer
80 61 4 67 1 73 75 E1 E 8C 93 3 C0 17 F1 8 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 6 
Protocol buffer contains 488 bytes. Content:
F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 7 80 61 4 5C 1 73 53 81 E 8C 93 3 C0 17 F1 B 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 9 80 61 4 5D 1 73 56 A1 E 8C 93 3 C0 17 F1 D 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F B 80 61 4 5E 1 73 59 C1 E 8C 93 3 C0 17 F1 F 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F D 80 61 4 5F 1 73 5C E1 E 8C 93 3 C0 17 F1 1 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F F 80 61 4 60 1 73 60 1 E 8C 93 3 C0 17 F1 3 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 11 80 61 4 61 1 73 63 21 E 8C 93 3 C0 17 F1 5 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 13 80 61 4 62 1 73 66 41 E 8C 93 3 C0 17 F1 7 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 15 80 61 4 63 1 73 69 61 E 8C 93 3 C0 17 F1 0 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 17 80 61 4 64 1 73 6C 81 E 8C 93 3 C0 17 F1 2 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 0 80 61 4 67 1 73 75 E1 E 8C 93 3 C0 17 F1 8 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 6 
Signature not supported.
Purging left 2 bytes. New _protocolBufferIndex: 486
Incoming buffer of 47 bytes. These will be appended to the protocolBuffer
80 61 4 68 1 73 79 1 E 8C 93 3 C0 17 F1 A 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 8 
Protocol buffer contains 533 bytes. Content:
64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 7 80 61 4 5C 1 73 53 81 E 8C 93 3 C0 17 F1 B 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 9 80 61 4 5D 1 73 56 A1 E 8C 93 3 C0 17 F1 D 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F B 80 61 4 5E 1 73 59 C1 E 8C 93 3 C0 17 F1 F 64 F1 10 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F D 80 61 4 5F 1 73 5C E1 E 8C 93 3 C0 17 F1 1 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F F 80 61 4 60 1 73 60 1 E 8C 93 3 C0 17 F1 3 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 11 80 61 4 61 1 73 63 21 E 8C 93 3 C0 17 F1 5 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 13 80 61 4 62 1 73 66 41 E 8C 93 3 C0 17 F1 7 64 F1 11 64 F1 2F 64 F1 30 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 15 80 61 4 63 1 73 69 61 E 8C 93 3 C0 17 F1 0 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 F 17 80 61 4 64 1 73 6C 81 E 8C 93 3 C0 17 F1 2 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 0 80 61 4 67 1 73 75 E1 E 8C 93 3 C0 17 F1 8 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 6 80 61 4 68 1 73 79 1 E 8C 93 3 C0 17 F1 A 64 F1 10 64 F1 20 64 F1 31 64 F1 40 64 F1 50 64 F1 60 64 F1 72 40 4 58 8 7 47 20 0 10 8 
Signature not supported.
Purging left 2 bytes. New _protocolBufferIndex: 531
Incoming buffer of 47 bytes. These will be appended to the protocolBuffer
[... cut ...]  91 D 95 FF FF 43 4B E 8C 93 3 0 FF FF FF 0 0 0 0 1 7A 3E 6B A2 A7 3 0 0 AE D0 50 FF 7F 0 0 75 91 D 95 
Signature not supported.
Purging left 2 bytes. New _protocolBufferIndex: 734
Incoming buffer of 16 bytes. These will be appended to the protocolBuffer
FF FF 42 59 0 0 0 2 0 0 0 0 E 8C 93 3 
Protocol buffer contains 16 bytes. Content:
FF FF 42 59 0 0 0 2 0 0 0 0 E 8C 93 3 
> End Session for ssrc 0xE8C9303
Freeing Session slot 0
Disconnected
Purging left 16 bytes. New _protocolBufferIndex: 0
```

Hackmancoltaire, 2016-06-05:
Ok, yeah that's odd. It seems to parse some of the packets some of the time. And I see a bounds error in there. That might be throwing off the whole parse. Seems like legit bug to me. I'll look into it as well with that tool.

Krenuds, 2016-06-05:
You know I just noticed that on the PC, packets are interpereted as malformed in wireshark. could this point me in the right direction?

[image]

Krenuds, 2016-06-05:
Keep in mind, it works like a charm on PC

Hackmancoltaire, 2016-06-05:
Malformed means it probably didn't process it properly and ignored the info. There is a clear parse error I think.

Krenuds, 2016-06-05:
Yeah now, if I'm not mistaken parsePacket(); is outside the realm of this implementation. It's part of arduino Wifi, right?
Perhaps there's something in there that's not jiving? Or am I barking up the wrong tree?

Krenuds, 2016-06-09:
I started poking around today and I think I've found the culprit. There's this part of the journaling system that is called a "Sanity check". What does this mean? I've build in some debugging info in my version.

```
System Flags: 8
Sanity check not passed: 
Offset: 42
Start Offset: 40
Difference: 2
Journal Len: 8
```
Code: 
```
		if ( offset-start_offset != sysjourlen ) {
			return -1;
		}
```

I'll post a more detailed report soon, I'm just collecting my thoughts;

lathoub, 2016-06-09:
The sanity check is just a way to check that all parts are decoded/parsed correctly. And it looks like some of the decoding did not happen as planned.

It might be that AppleMIDI on OSX sends additional info that rtpMidi on Windows does not send.

Hackmancoltaire, 2016-06-09:
It's usually a mixture of the protocol layer controlled by the OS, and the audio program that's sending the notes. VMPK for instance doesn't seem to send journal information, while Rondo (a Mac midi player) sends all kinds of messages and journals per packet. The AppleMidi framework has to recognize and toss out most of the journal info cause as of yet we haven't written in how to process it. Similarly with sysex calls.

Krenuds, 2016-06-09:
Yes, it would seem that there is a journal attached tot he MTC on mac, but not on PC. I believe this is the culprit. I'm going to start working on those features this weekend. Also, I've the basic framework for Sysex callbacks set up on my branch. It's going to be a long weekend!

Krenuds, 2016-06-09:
Thanks for the input

lathoub, 2016-06-10:
I was starting to work on a "weight loss" program for the library, but will wait for your feedback on Sysex and the journaling issue. Thanks!

Bluebie, 2016-06-30:
Are there any workarounds for this overflow crasher on esp8266 as a stopgap?

Krenuds, 2016-06-30:
Yes there is one quick one, if you reconfigure the protocol buffer to be circular it will function without crashing.

Bluebie, 2016-07-22:
I think this issue is resolved now? I've not observed an ESP8266 crash since my changes eliminating the packetBuffer from the dissector.

lathoub, 2016-07-22:
@Krenuds @Hackmancoltaire can we mark this as resolved?

Hackmancoltaire, 2016-07-22:
Sure. I haven't had the chance to test my ESP with the updated code. But it didn't often crash for me. So I'd say we can mark it as resolved.

closed 2016-07-22 by lathoub