lathoub/Arduino-AppleMIDI-Library #28: SysEx

DBrenchley, 2016-06-16:
Good day gents ..
Do I assume correctly, implementation of support for SysEx is not yet complete?
I'm attempting a "dump" .. response "decode_sysex_start" ..
pouring though the code seems to reveal a work in progress ..

Hackmancoltaire, 2016-06-16:
It is a bit.

DBrenchley, 2016-06-16:
Thanks .. I'll await further updates!!

mathieu-b, 2017-07-10:
Any news?
Cheers.

lathoub, 2017-08-14:
I have **started** coding SysEx (first step).

Sending SysEx from the Arduino seems to work - can someone check (see sample code).
(Some code is still hardcoded and needs a beauty makeover.)

lathoub, 2017-08-15:
I modified the code so that it can now receive SysEx buffers.
Its up to the receiver to decode the buffer further (update SysEx sample code)

lathoub, 2017-08-15:
Added support for 1) Full SysEx messages and 2) SysEx Segments (when message is larger than 255 bytes). See updated sample

closed 2017-08-17 by lathoub