lathoub/Arduino-AppleMIDI-Library #29: Refactor incoming packetBuffer to be clearer

Bluebie, 2016-07-01:
The current design used in src/utility/dissector.h is intermixed with other code and is becoming complex and hiding bugs like this one: https://github.com/Bluebie/Arduino-AppleMidi-Library/commit/d061966879f869a20035f963d2810d83337a6bf7

Bluebie, 2016-07-01:
Can someone explain why the packet buffer exists at all? Intermediary network devices shouldn't ever be fragmenting these packets, and UDP doesn't deliver packets in the same order they're sent, so without a reordering mechanism this seems to be a Rube Goldberg machine for building packet buffers with random junk data appended to the beginning and end. This sort of buffering would make sense if this service ran over TCP, but it never does.

Bluebie, 2016-07-01:
I've created an experimental branch with a simplified dissector with no packet buffer. I'm not seeing any more stack crash types of instability with CoreAudio now, and the library connects and stays connected reliably, which it wouldn't do for me before. I've also simplified the AppleMidi.run() function in that branch to avoid watchdog resets that seemed to happen when the device was overwealmed by a big queue of rtp messages. My ESP8266 test board seems stable now. Yay! I'm sure I'll explore more of this library and try to clean things up as I go.

Bluebie, 2016-07-01:
Bufferless dissector should help with running this library on more memory constrained hardware too. When `#define PACKET_MAX_SIZE 128` I expect this frees at least 256 bytes on Harvard architecture machines like the classic arduino stuff (atmega).

DBrenchley, 2016-07-01:
Forgive my ignorance .. new to C++ and github postings ..
This would appear to be a significant change of some import ..
how is versioning controlled ? Are Jenna's changes reflected in lathoubs
lib?

Bluebie, 2016-07-01:
My changes are currently only in one branch of my own fork (copy) of this repository. Lathoub has the option to bring those changes back in to this repository if they like any of them.

lathoub, 2016-07-02:
A new pair of eyes looking at the code, good!
Just looked at the changes proposed by @Bluebie and they look very promising and will reduce the memory footprint. 

It seemslike @Bluebie used @Krenuds fork to do the changes, so I can't bring the changes into this one.

lathoub, 2016-07-02:
Just tested the changes and they work as advertised - nice!

@Bluebie : if you fork from this one, make the changes to dissect.h and create a pull request, I can pull them into the main branche. Also make sure you add your name in the list of authors

Bluebie, 2016-07-05:
How do you feel about the change to `AppleMIDI.run()` which changes it to only process one packet on control and data ports per execution of the `run()` function? I think it's a good change, but maybe some existing projects depended on the old behaviour?

Where is the list of authors?

Hackmancoltaire, 2016-07-05:
The 'Graphs' tab shows contributors to the project. Most of it is lathoub, with some minor bug fixes here and there by myself. I use the framework in a piano piece I built (mousepiano.com). I can check these changes against my piano tomorrow and see how they work in heavy duty real world conditions. The software I use (Rondo) can sometimes send packets of hundreds of bytes at once.

lathoub, 2016-07-06:
@Bluebie I'm not able to see the changes to `AppleMIDI.run()` in your branche of the code.

The list of authors is in the header of the `dissector.h` file (and implicitly in the logs in github, where I would accept your pull request)

Bluebie, 2016-07-06:
I've cherrypicked the changes and provided pull requests #31 and #32 so hopefully it's an easy merge. #32 requires consideration, as it changes the behaviour of the run function, but #31 shouldn't negatively affect any users as I don't think it has any adverse side effects.

lathoub, 2016-07-06:
#31 accepted. Will test #32

lathoub, 2016-07-07:
#32 tested, OK and merged! Thx Bluebie

closed 2016-07-07 by lathoub