lathoub/Arduino-AppleMIDI-Library #30: Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500 PRODUCT ID: 3010

trlafleur, 2016-07-02:
will we see support in this library with this board??

Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500
PRODUCT ID: 3010

lathoub, 2016-07-02:
Don't know - never tried (I don't have the hardware). 
If you have the hardware, try and come back with your feedback

DBrenchley, 2016-07-03:
I'm currently using it on a SAMD21 (Zero) + WIFI Shield 101.
Required changing some lib refs but seems to work fine.

trlafleur, 2016-07-06:
what changes did you make??

Thamnks

DBrenchley, 2016-07-06:
In the noteon/off ex. I changed the ref. to wifi.h to wifi101.h
not sure if that's relevant ..

trlafleur, 2016-07-07:
I get a crash during the the test to see if the wifi device is connected....
I made the changes for the .h files...

DBrenchley, 2016-07-07:
The libraries for your device are likely different. These may help ..
https://github.com/ARMmbed/sal-driver-atmel-winc1500
google
Software Programming Guide for ATWINC1500 Wi-Fi
using SAM D21 Xplained Pro

trlafleur, 2016-07-07:
Thanks, I do have a copy of this....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

trlafleur, 2016-07-12:
I've been trying to get AppleMidi working on a feather M0 with a WINC1500 wifi radio. 
(works fine on the feather M0 with an ESP8266)

I have tested with the Adafruit 1500 drivers and the WiFi101 driver with pull request 58 that exposes the GPIO pins for the WINC1500. I made the needed

 change to the .h files

In all cases the program compiles and loads fine, then on a reset I get the red LED on solid... ie: a crash, no output from the serial monitor.
IDE is: 1.6.9, SAMD is: 1.6.6

So I'm at a lost of where to start to fix this issue... Help

I have run both AdaFruit and the WiFi101 UdpNtpClient test code, works fine, so this tell me the basic function of the WINC1500 are working....

Thanks

~~~
// Hardware: Mega 2560 R2 + Wifi Shield

// These need to be included when using standard Ethernet
#include <SPI.h>
#include <WiFi101.h>
#include <WiFiUdp.h>

#include "AppleMidi.h"

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
  // Required for Serial on Zero based boards
  #define Serial SERIAL_PORT_USBVIRTUAL
#endif

int status = WL_IDLE_STATUS;
char ssid[] = "xxxxx"; //  your network SSID (name)
char pass[] = "xxxxxx";    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                 // your network key Index number (needed only for WEP)

unsigned long t0 = millis();
bool isConnected = false;

APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI); // see definition in AppleMidi_Defs.h

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void setup()
{
  // Serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  WiFi.setPins (8,7,4,2);

  Serial.print("Getting IP address...");

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if ( fv != "1.1.0" )
    Serial.println("Please upgrade the firmware");

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid);

    // wait 10 seconds for connection:
    delay(10000);
  }

  Serial.println();
  Serial.print("IP address is ");
  Serial.println(WiFi.localIP());

  Serial.println("OK, now make sure you an rtpMIDI session that is Enabled");
  Serial.print("Add device named Arduino with Host/Port ");
  Serial.print(WiFi.localIP());
  Serial.println(":5004");
  Serial.println("Then press the Connect button");
  Serial.println("Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes");

  // Create a session and wait for a remote host to connect to us
  AppleMIDI.begin("test");

  AppleMIDI.OnConnected(OnAppleMidiConnected);
  AppleMIDI.OnDisconnected(OnAppleMidiDisconnected);

  AppleMIDI.OnReceiveNoteOn(OnAppleMidiNoteOn);
  AppleMIDI.OnReceiveNoteOff(OnAppleMidiNoteOff);

  Serial.println("Sending NoteOn/Off of note 45, every second");
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  // Listen to incoming notes
  AppleMIDI.run();

  // send a note every second
  // (dont cáll delay(1000) as it will stall the pipeline)
  if (isConnected && (millis() - t0) > 1000)
  {
    t0 = millis();
    //   Serial.print(".");

    int note = 45;
    int velocity = 55;
    int channel = 1;

    AppleMIDI.noteOn(note, velocity, channel);
    AppleMIDI.noteOff(note, velocity, channel);
  }
}

// ====================================================================================
// Event handlers for incoming MIDI messages
// ====================================================================================

// -----------------------------------------------------------------------------
// rtpMIDI session. Device connected
// -----------------------------------------------------------------------------
void OnAppleMidiConnected(uint32_t ssrc, char* name) {
  isConnected = true;
  Serial.print("Connected to session ");
  Serial.println(name);
}

// -----------------------------------------------------------------------------
// rtpMIDI session. Device disconnected
// -----------------------------------------------------------------------------
void OnAppleMidiDisconnected(uint32_t ssrc) {
  isConnected = false;
  Serial.println("Disconnected");
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void OnAppleMidiNoteOn(byte channel, byte note, byte velocity) {
  Serial.print("Incoming NoteOn from channel:");
  Serial.print(channel);
  Serial.print(" note:");
  Serial.print(note);
  Serial.print(" velocity:");
  Serial.print(velocity);
  Serial.println();
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void OnAppleMidiNoteOff(byte channel, byte note, byte velocity) {
  Serial.print("Incoming NoteOff from channel:");
  Serial.print(channel);
  Serial.print(" note:");
  Serial.print(note);
  Serial.print(" velocity:");
  Serial.print(velocity);
  Serial.println();
}

~~~

DBrenchley, 2016-07-12:
I assume your testing of WiFi utilized similar serial I/O methods/settings to confirm connectivity? 
Is use of the software serial implementation absolutely necessary on your board? I had issues when I attempted on mine (arduino.cc Zero ) .. likely not related ..
Your lack of serial com makes me question this as a primary issue .. I'd start .. you likely already have, by commenting out the while statement that could stall the code .. it was unnecessary in my implementation .. 
Not that you've gotten here yet .. you do have RTPMIDI or an apple product that supports it running a current midi session?

trlafleur, 2016-07-12:
Yes, hardware serial I/O work fine to seril0, serialUSB and 1
WiFi stack works fine using NTP UDP test example with serial I/O

Yes, it looks like something primary... I do NOT have an ICE, so this is
hard to find..
Code work fine using the Feather M0 with and ESP8266

removing the serial I/O did NOT help, on the feather the Red led is use to
show status of uploads, it stays on after a load or reset...

I'm looking at the code to try and trace startup issue... but this is hard.

DBrenchley, 2016-07-12:
Have you considered Atmel VS7 IDE ? It might assist as it presents debugging tools ..

trlafleur, 2016-07-12:
not yet, I'm on a MAC... would need to set up a PC

trlafleur, 2016-07-14:
At this point I'm not sure what to do...
I have set GPIO pins in the startup code for WiFi101 and AppleMidi and they never get set..
I stripped the NoteOnOffEverySec.ino to remove just the midi code and it load and run the serial port and the wifi101 code just fine..

but when I add back just the: 
~~~
APPLEMIDI_CREATE_INSTANCE (WiFiUDP , AppleMIDI);
~~~
back to the code I get the RED led on the Feather M0 board.. 
It looks to me to be a loader issue , but without the right tools I'm at a lost...
I've tested with 1.6.4 and 1.6.9 with SAMD 1.6.6
 
anyone at AdaFruit who can give a clue???

DBrenchley, 2016-07-15:
lathoub:
Did you folks changed the portion of code where you read an analog input to capture a seed for the random number generation of the SSRC?

DBrenchley, 2016-07-15:
In AppleMIDI.hpp you use an analog input to provide a seed to generate a random SSRC ID
srand(analogRead(0)); 

I just changed the seed to a numeric value

trlafleur, 2016-07-15:
well, this got my test code running... How did you stumble on to this
issue???

I will try the full code later today

DBrenchley, 2016-07-15:
As soon as you said commenting the class instantiation .. I remembered .. 

just keep commenting out sections of code till the "hang" went away ..

trlafleur, 2016-07-15:
this is very interesting...

If I run this test code, the srand/rand functions look to works just fine
on the M0

/* srand test on M0
 *
 */

unsigned long t0 = millis();
unsigned long junk;

void setup() {
  // put your setup code here, to run once:

   // Serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  Serial.println("Starting srand test");
  srand(analogRead(0));

}

void loop() {
  // put your main code here, to run repeatedly:

 if ( (millis() - t0) > 1000)
  {
    t0 = millis();

   junk = rand();
   Serial.println(junk);
  }

}

DBrenchley, 2016-07-15:
It's not the C++ .. it's the reading of the analog input to get the seed ..

trlafleur, 2016-07-15:
My guest is that the analogRead(0) is not initialized when the
APPLEMIDI_CREATE_INSTANCE is created.

simonp54, 2016-07-15:
Is the pin initialised for input?

From: trlafleur 
Sent: Friday, July 15, 2016 3:18 PM
To: lathoub/Arduino-AppleMidi-Library 
Subject: Re: [lathoub/Arduino-AppleMidi-Library] Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500 PRODUCT ID: 3010 (#30)

My guest is that the analogRead(0) is not initialized when the
APPLEMIDI_CREATE_INSTANCE is created.

trlafleur, 2016-07-15:
> It's a unique way to ensure a truly rnd, not sure the application requires that "absoluteness" IMHO an unnecessary prerequisite on the HW. I'll defer to the author(s) ..

I understand what it is, still unclear as to why it crashes application.      Again, my thought is that analog read is not initialize correctly by this time

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

trlafleur, 2016-07-15:
Test program work fine without doing this.   And again we are just looking for noise on the line to give a seed to rand

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

DBrenchley, 2016-07-15:
Is the arduino "wiring" references proper for the SAMD ? It's newer device for them .. everything may not yet been shaken out .. on my brd pin (0) .. is serial I/O

DBrenchley, 2016-07-15:
scratch that .. even if it were Dig I/O it should return a value ?? Interesting ..

trlafleur, 2016-07-15:
Yes, we only need noise

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

trlafleur, 2016-07-15:
Ok, I now know the source of the problem...

In the analogRead function, this statement caused the issue:

ADC->CTRLA.bit.ENABLE = 0x01;

If you read the notes in the source code:

Before enabling the ADC, the asynchronous clock source must be selected and
enabled, and the ADC reference must be  configured.

Setting the ATD reference did not fix the issue, but setting the asynchronous
clock source did fix this..
This was done in the AppleMidi.hpp file. init() is a core routine needed to
setup the clocks.

I did it by doing an "init();" prior to the analogRead read call.. This is
a SAMD CORE routine that set up the clocks for the ATD and other devices.

init();
srand(analogRead(0)); // to generate our random ssrc (see in begin)

So this appeared to be an Arduino SAMD core issue... or ???

Now I'm not sure how to report this...

DBrenchley, 2016-07-15:
MHO .. this lib's author's issue. It's an unnecessary dependency on the HW platform .. my $.02

trlafleur, 2016-07-15:
Yes and no, you should be able to CREATE an INSTANCE and use the function
of the processor...

trlafleur, 2016-07-15:
The location of the analog-read for the random number, could be moved to
the AppleMIDI.begin or .run  function to solve this issue.

But again, you should be able to CREATE an INSTANCE and use the function of
the processor...

DBrenchley, 2016-07-15:
true .. but why a dependency is this lib? What if it was desired the pin be
used for another purpose aka as stated it's cfg'd by default as Ser I/O on
my brd .. here again I'll defer to the lib's author .. but IMHO doesn't
make sense ..

trlafleur, 2016-07-15:
It's used just once, so you can re assign the pin.   

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

trlafleur, 2016-07-16:
My suggestion that I have tested on a feather M0 board....

Move the srand(analogRead(0)); located in AppleMidi.hpp
from the:
inline AppleMidi_Class<UdpClass>::AppleMidi_Class()
to:
inline void AppleMidi_Class<UdpClass>::begin(const char* sessionName, uint16_t port)
 as listed below...  

This should work on all Arduino processor as AppleMidi_Class<UdpClass>::begin is used in the setup() section of the code that runs after all initialization are done.

```
/*! \brief Default constructor for MIDI_Class. */
template<class UdpClass>
inline AppleMidi_Class<UdpClass>::AppleMidi_Class()
{
	
#if APPLEMIDI_USE_CALLBACKS
	// Initialise callbacks to NULL pointer
	mConnectedCallback				= NULL;
	mDisconnectedCallback			= NULL;

	mNoteOffCallback				= NULL;
	mNoteOnCallback					= NULL;
	mAfterTouchPolyCallback			= NULL;
	mControlChangeCallback			= NULL;
	mProgramChangeCallback			= NULL;
	mAfterTouchChannelCallback		= NULL;
	mPitchBendCallback				= NULL;
	mSystemExclusiveCallback		= NULL;
	mTimeCodeQuarterFrameCallback	= NULL;
	mSongPositionCallback			= NULL;
	mSongSelectCallback				= NULL;
	mTuneRequestCallback			= NULL;
	mClockCallback					= NULL;
	mStartCallback					= NULL;
	mContinueCallback				= NULL;
	mStopCallback					= NULL;
	mActiveSensingCallback			= NULL;
	mSystemResetCallback			= NULL;
#endif

#if APPLEMIDI_USE_EVENTS
	mNoteOnSendingEvent				= NULL;
	mNoteOnSendEvent				= NULL;
	mNoteOffSendingEvent			= NULL;
	mNoteOffSendEvent				= NULL;
#endif
//	srand(analogRead(0)); // to generate our random ssrc (see in begin)   // <---------------

	uint32_t initialTimestamp_ = 0;
	_rtpMidiClock.Init(initialTimestamp_, MIDI_SAMPLING_RATE_DEFAULT);
}

/*! \brief Default destructor for MIDI_Class.

 This is not really useful for the Arduino, as it is never called...
 */
template<class UdpClass>
inline AppleMidi_Class<UdpClass>::~AppleMidi_Class()
{
}

//---------------------------------------------------------------------------------------

/*! \brief Call the begin method in the setup() function of the Arduino.

All parameters are set to their default values:
- Input channel set to 1 if no value is specified
- Full thru mirroring
*/
template<class UdpClass>
inline void AppleMidi_Class<UdpClass>::begin(const char* sessionName, uint16_t port)
{
	
#if (APPLEMIDI_DEBUG)
	if (strlen(sessionName) > SESSION_NAME_MAX_LEN)
	{
		Serial.print("SessionName exceeds ");
		Serial.print(sessionName);
		Serial.print(" exceeds ");
		Serial.print(SESSION_NAME_MAX_LEN);
		Serial.println(" chars. Name will be clipped.");
	}
#endif
	strncpy(_sessionName, sessionName, SESSION_NAME_MAX_LEN);

	Port = port;

	_inputChannel = MIDI_CHANNEL_OMNI;
	
	srand(analogRead(0)); // to generate our random ssrc (see in begin)   // <------------------ init(); was added

	// Generate Synchronization Source
	// Unique 32 bit number (see definition)
	byte buffer[4];
	for (int i = 0; i < 4; i++)
		buffer[i] = 17 + (rand() % 255);
		//buffer[i] = 17 + (random() % 255);
	_ssrc = *((uint32_t*) &buffer[0]);

	// Initialize Sessions
	DeleteSessions();

	// open UDP socket for control messages
	_controlUDP.begin(Port);
	// open UDP socket for rtp messages
	_contentUDP.begin(Port + 1);

	_controlDissector.init(Port, this);
	_controlDissector.addPacketDissector(&PacketAppleMidi::dissect_apple_midi);	// Add parser

	_contentDissector.init(Port + 1, this);
	_contentDissector.addPacketDissector(&PacketRtpMidi::dissect_rtp_midi);		// Add parser
	_contentDissector.addPacketDissector(&PacketAppleMidi::dissect_apple_midi);	// Add parser

	_rtpMidi.ssrc = _ssrc;
	_rtpMidi.sequenceNr = 1;

#if (APPLEMIDI_DEBUG)
	Serial.print("Starting Session ");
	Serial.print(_sessionName);
	Serial.print(" on port ");
	Serial.println(Port);
#endif
}
```
```

lathoub, 2016-07-16:
Hi, Thanks for surfacing this issue.

I agree, `srand(analogRead(0)); `should not be in the constructor, but closer to where the random number is generated. More than happy to accept a change request with that move.

An alternative to `srand(analogRead(0)); ` is to use the IP address (`Ethernet.localIP()`) as the unique SSRC (this removed the dependency on the hardware entropy generator - as @DBrenchley righteously suggests).

lathoub, 2016-07-16:
OK, `_ssrc` now generate from `localIP `and should fix the issue of `srand(analogRead(0)); `
Changes committed, thanks for reporting and finding the issue

lathoub, 2016-07-16:
(please test and report)

trlafleur, 2016-07-17:
Yes, this fixes the issue...

Thanks

lathoub, 2016-07-17:
Yes sir. You can create the AppleMIDI instance without a connection (invalid IP address), but once you call `begin()`, the board needs to be connected (valid IP address). I could add an additional test in `begin() `to check that first

lathoub, 2016-07-17:
a valid network connection (valid IP address) **is** needed to start the UDP sockets in begin()

lathoub, 2016-07-17:
To check if `Ethernet.begin` has been called prior to `AppleMIDI.begin`, is to check `Ethernet.dnsServerIP()` - if its 0 (zero), then `Ethernet.begin` was not called. 

This works on both static and dynamic IP addresses (obtained via `Ethernet.begin(mac, ip)` or `Ethernet.begin(mac)` respectively - or even `Ethernet.begin(mac, ip, dns, gateway)`)

Is there another method to check if `Ethernet.begin` was called?

lathoub, 2016-07-17:
I have added the above check to the master branch (commit ebbff49)

lathoub, 2016-07-17:
If the changes fix the issue with the Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500 , then close this issue and the device to the list of supported device in README.md and add an example. (yes, I'm optimistic :-) )

trlafleur, 2016-07-18:
closing  issue

closed 2016-07-18 by trlafleur

Bluebie, 2016-07-22:
This library no longer builds on ESP8266. Dependancy on Ethernet makes the library incompatible with any devices which use an alternate network interface.