lathoub/Arduino-AppleMIDI-Library #34: Fails to build on ESP8266 platform, due to new dependancy on Ethernet library

Bluebie, 2016-07-22:
Ethernet library is not compatible with ESP8266 Arduino core. A more general way to calculate ssrc is required.

lathoub, 2016-07-22:
Indeed :-( I was to trigger happy to include the `<Ethernet.h>` in `AppleMIDI.h`

In order to reduce the issue to just` _ssrc = Ethernet.localIP();` I suggest to remove the test to see if Ethernet.begin was called. (line 90)

A quick fix is to use `_ssrc = random(UINT32_MAX);`, knowing this might not work in environments with multiple devices that use this libs (where random(UINT32_MAX); could generate the same number).

Another fix is to generate SSRC outside of the lib, and pass it as a parameter to the `begin `function (not as nice, as SSRC is a pure internal number)

Bluebie, 2016-07-22:
I'm not familiar with the workings of the AppleMIDI session management protocol, but would it be possible to calculate _ssrc lazily, the first time a network device requires it? Then maybe the time between device power on and first network traffic requiring ssrc could create some decent randomness in micros() that we could use to generate a more reliably random ssrc.

lathoub, 2016-07-22:
Removed dependency on Ethernet library to calculate `_ssrc`. `_ssrc` is now lazily initialized and accessed using `getSynchronizationSource`

lathoub, 2016-07-22:
I just tried the latest version of the library on my Adafruit Huzzah ESP8266 (https://www.adafruit.com/product/2471) - and into issues running the NoteOnOff example: connecting goes OK, the first note on/off are send, then no more notes are received and timing on rtpMIDI goes into high numbers (3878784ms!). Everything seems to work fine on my Arduino Mega with the Ethernet shield.

Does the ESP8266 noteOnOff example work for you?

alan-dodd, 2016-10-26:
See the issue I posted, there are some hints about just this problem (had'nt noticed your issue, I would have replied here) - looking at the captured data with wireshark (i'm using windows) I notice timestamp data from esp8266 not consistent, and with some bytes more than needed (8 bytes too much) - look at the issue. Don't know solution. Maybe a different way that processor is producing long integers??
a.

closed 2016-12-25 by lathoub