lathoub/Arduino-AppleMIDI-Library #35: Sending noteOn/noteOff to macOS does not work

Bluebie, 2016-07-22:
The example projects don't work when connected to a macOS computer. Audio MIDI Setup can connect to the device, and even route it's midi commands to other ports or software, but timestamps on messages appear frozen in the past and both GarageBand and Ableton Live ignore the notes.

Software MIDI controllers generate MIDI notes with an up to date timestamp in CoreMIDI as observed with [MIDI Monitor](https://www.snoize.com/MIDIMonitor/), and when those notes are sent out the Network session the RTP header includes an accurate timestamp. MIDI notes sent from an AppleMIDI-Library device enter CoreMIDI, can be observed with MIDI Monitor, but have past tense unchanging timestamps and are ignored by music production apps.

I've had a look at network traffic with Wireshark, both looking at noteOn traffic generated by the Arduino-AppleMIDI-Library examples, and by [vmpk](http://vmpk.sourceforge.net) virtual software midi keyboard routing it's output to the network session, and the only major differences I can see are:

* CoreMIDI includes journals with it's messages
* NoteOn messages from AppleMIDI-Library have the P-flag set, indicating the "Status Byte was NOT present in the original MIDI-command". CoreMIDI does not set this bit.
* Timestamps from CoreMIDI advance sensibly. Timestamps in the RTP header from AppleMIDI-Library are always set to 0.

Because MIDI Monitor only shows the timestamps being different, I doubt apps like Ableton are aware of the journal or P-flag. Can we repair the timestamps to be compatible with macOS?

lathoub, 2016-07-22:
I have a Mac, but use it for biz apps only- so yes, please repair and I can test on Windows.

Hackmancoltaire, 2016-07-22:
I tried all these variations as suggested in the past, but found them to make the AppleMidi-Library completely broken. I can't explain it, as I would have expected the timestamp to advance as well, however some of the docs indicate that the timestamp should be used when the notes are in the future, and that to have the notes played immediately by the other listeners in the session that the timestamp had to be 0. 

https://developer.apple.com/library/ios/documentation/CoreMidi/Reference/MIDIServices_Reference/#//apple_ref/doc/uid/TP40010316-CHMIDIServiceshFunctions-SW30
The time at which the events occurred, if receiving MIDI, or, if sending MIDI,
the time at which the events are to be played. Zero means "now." The time stamp
applies to the first MIDI byte in the packet.

I can't tell you how many hours I tried getting this to fit what I understood as the protocol. But results from the Mac just completely didn't work. As of this commit I was receiving notes on my Mac from an AppleMidi device. I've been using it for months and it still works.

https://github.com/lathoub/Arduino-AppleMidi-Library/commit/cfe8745f60b0ce36ddf3e7a7d9b86c591b2fddf6

My setup is:

AppleMidi-Library device -> Mac -> Processing

trlafleur, 2016-07-22:
I'm using a Mac, runs just fine

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

Bluebie, 2016-07-24:
@lathoub we should talk about noteon/off here. Re your #34 comment:
> I just tried the latest version of the library on my Adafruit Huzzah ESP8266 (https://www.adafruit.com/product/2471) - and into issues running the NoteOnOff example: connecting goes OK, the first note on/off are send, then no more notes are received and timing on rtpMIDI goes into high numbers (3878784ms!). Everything seems to work fine on my Arduino Mega with the Ethernet shield.
> 
> Does the ESP8266 noteOnOff example work for you?

I tried the noteOnOff example initially when trying to get the library working, back when the buffer problems still existed, and when I acted very quickly I managed to get a note to map in Ableton and trigger a clip a few times, but i've not been able to get it to work since that very first time I used the library. I've rebooted my computer and reopened software several times since then. I'm not sure what conditions changed since then, but I was definitely using an older release of the library (installed via platformio's library manager), and judging by the earlier commits talking about timestamp changes, I'm suspecting the version I used then may have been attempting to send nonzero timestamps.

I've just tested the ESP8266_NoteOnOffEverySec example again with the current git version and the results are consistent with my recent experience. Ableton is unresponsive, seeming to not see any notes from the device. MIDI Mapper can detect the notes, but only if I set the Live Routing in Audio MIDI Setup to go in to some other port, or straight back to the network session & esp8266 device. The ESP8266 device properly responds to it's own notes being mirrored back. The log in MIDI Mapper looks like this:

```
Time:           Source:         Message:    Channel:  Data:
03:49:43.870    To Session 1    Note On     1         45  55
03:49:43.870    To Session 1    Note Off    1         45  55
03:49:43.870    To Session 1    Note On     1         45  55
03:49:43.870    To Session 1    Note Off    1         45  55
03:49:43.870    To Session 1    Note On     1         45  55
03:49:43.870    To Session 1    Note Off    1         45  55
03:49:43.870    To Session 1    Note On     1         45  55
03:49:43.870    To Session 1    Note Off    1         45  55
```

Continuing like that, with the timestamp static, at about the time the session was established. Notably MIDI Monitor is not a network sniffer - it reads the system midi APIs, so these timestamps are some high level information being communicated to midi software like Ableton and GarageBand. My guess is there is some protection in either the DAWs or in CoreMIDI which is considering these packets to be very old and discarding them. Maybe it's to do with the rtpMIDI spec's suggestions that rtpMIDI could be used for streaming music in non-realtime applications? Could they be interpreting the device as non-realtime and trying to schedule the notes to play at the moment of the timestamp, but since it's in the past, it never actually executes?

I've not observed the latency thing you talk of. the Latency Adj value in Audio MIDI Setup app stays at exactly 0ms for me and I've not seen it change at all, except for the first time I used the library and managed to get a few notes out of it, but I don't remember what the latency measurement did precisely.

Bluebie, 2016-07-24:
I've connected my iPhone to CoreMIDI via the same AppleMIDI/rtpMIDI networking system and I'm transmitting realtime instrument controls from the phone in to ableton fine. MIDI Monitor shows the timestamps on those packets increasing in a realtime fashion as I would expect, not stuck in the past. Going to break out wireshark and see what that looks like on the network.

Bluebie, 2016-07-24:
rtpMIDI packets from the iOS device look very similar to those from the macOS CoreMIDI system. The packets have an incrementing timestamp, indicate the "status byte was present in the original midi command", and have a journal. Whatever information indicated iOS devices don't use incrementing timestamps seems to be wrong. Maybe iOS tolerates a constantly 0 timestamp more than macOS (I haven't tested this), but it is certainly not how the device transmits midi packets itself.

Wireshark example of a noteOff from iOS 9:

```
Frame 111: 68 bytes on wire (544 bits), 68 bytes captured (544 bits) on interface 0
Ethernet II, Src: Apple_01:9f:b7 (a8:66:7f:01:9f:b7), Dst: Apple_c5:6d:8f (90:b2:1f:c5:6d:8f)
Internet Protocol Version 4, Src: 192.168.1.10, Dst: 192.168.1.17
User Datagram Protocol, Src Port: 5005 (5005), Dst Port: 5005 (5005)
Real-Time Transport Protocol
    [Stream setup by AppleMI (frame 8)]
    10.. .... = Version: RFC 1889 Version (2)
    ..0. .... = Padding: False
    ...0 .... = Extension: False
    .... 0000 = Contributing source identifiers count: 0
    0... .... = Marker: False
    Payload type: rtp-midi (97)
    Sequence number: 39820
    [Extended sequence number: 105356]
    Timestamp: 8566212
    Synchronization Source identifier: 0xd8238c63 (3626208355)
RFC 4695/6295 RTP-MIDI
    0... .... = B-Flag: Short Header: 1 octet
    .1.. .... = J-Flag: Journal present
    ..0. .... = Z-Flag: No Delta-Time for first MIDI-command
    ...0 .... = P-Flag: Status byte was present in original MIDI-command
    .... 0011 = Command length (short): 3
    Command Section
        Note Off (c=1, n=G4, v=0)
            1000 .... = Channel Status: Note Off (0x08)
            .... 0000 = Channel: Channel 1 (0x00)
            .100 0011 = Note: G4 (67)
            .000 0000 = Velocity: 0
    Journal Section
```

Bluebie, 2016-07-24:
Also @lathoub you say everything works fine on the Ethernet shield. Are you testing the noteOnOff example? I know one of the example uses of the library was to drive an electrified piano/organ type of machine, but that would mainly be receiving notes, which works fine, even with macOS/iOS devices sending them.

lathoub, 2016-08-06:
(Sorry for my late response, just back from vacation)
I tested the noteOnOff on Ardiuno and Ethernet shield on Win10/rtpMidi. I will test the ESP8266 tomorrow and report back (I will try to include a test using MacOS).

Bluebie, 2016-08-07:
I finally got around to looking in to the docs @Hackmancoltaire linked, and I don't see any relevance. They are not documenting the rtpMIDI wire protocol, they're documenting the IPC API Apple uses when the CoreMIDI bus communicates abstract midi packets between apps running on an iOS device. Those packets can come from USB, Bluetooth, rtpMIDI devices, or even other apps. When the behaviour was tested on macOS, sending rtpMIDI packets with a zero timestamp is not interpreted as a 'now' midi message, it's interpreted as a midi message that is timed in the past at the moment the session was connected.

It's also important to note @Hackmancoltaire is feeding midi to processing, and may not be paying attention to timestamps on those messages. The packets do enter the CoreMIDI bus, and apps can read them. They're just too far in to the past to be useful in DAW software.

chris-zen, 2016-12-18:
I am having the same problem reported by @Bluebie 
I installed the library from the master branch with 
```platformio lib install https://github.com/lathoub/Arduino-AppleMidi-Library.git```
When looking at the packets in Wireshark I see the timestamp being 0 always.
I can "Live Route" the events back to the network and the ESP8266 prints them appropriately.
GarageBand doesn't do anything with the NoteOn/Off events.
MIDI monitor doesn't show the events.
@lathoub any plan for fixing this ? Or else, any suggestion on where to look at for fixing it ?

Hackmancoltaire, 2016-12-18:
@bluebie I tried several pieces of software including Processing, GarageBand, VMPK, and Rondo. As I said above, I tried many variations that matched the protocol spec but had little luck getting consistent operation. I have no explanation unfortunately. :(

chris-zen, 2016-12-18:
I have been diving through the code and I have found some bugs with timestamp calculation in ```RtpMidi_Clock```. Now I am working on synchronization protocol, which I suspect has also some bugs (but not sure yet). I'll be working on trying to fix them and see if I get it running on Mac OSX during the next weeks (I don't have much free time). I'll fork and commit my current work as soon as I can, in case someone wants to follow up with my investigations.

chris-zen, 2016-12-18:
@lathoub, @Bluebie, @Hackmancoltaire, I found a solution sooner than I expected, please see PR #42. It allows to use AppleMIDI between an ESP8266 and a Mac OSX, but I suspect that it will require some extra work to ensure that it will work with all the devices (basically their respective compilers).

Hackmancoltaire, 2016-12-19:
This is interesting because I use a practically unmodified version of AppleMidi with an ESP. An ArduinoMega running AppleMidi invites a Mac to a session, then the ESP connects to the Mac's session. Sometimes the Arduino requires a few restarts but it eventually works.

chris-zen, 2016-12-19:
@Hackmancoltaire the problem won't appear on non 8-bit architectures such as the atmel devices because the compiler doesn't introduce any padding in the structs. But in the case of the ESP which is a 32 bit one, the compiler introduces padding by default that adds to the one defined explicitly in the struct typedef, messing with the packet content.
In my case with the ESP, the problem only appeared when sending events, not receiving (which worked fine because your lib is not as strict with timestamps as it is CoreMIDI). If in your setup you only use the ESP for receiving from the Arduino then the problem won't show up and everything will work fine. If you try to send events from the ESP into a Mac OSX you will see with Wireshark that the packets are received (malformed) but CoreMIDI doesn't do anything with them (they are discarded because the timestamps are corrupted basically).

Hackmancoltaire, 2016-12-19:
Fascinating. Great find.

chris-zen, 2016-12-20:
@Hackmancoltaire I updated the PR with a better strategy to serialise the synchronization packet that doesn't require packed structs (and so it is not dependant on the compiler). Please consider using the same strategy for the other packet types too.

Looking forward for comments from the maintainers ;-)

lathoub, 2016-12-20:
Great find chris-zen, thanks so much!
I suggest that I merge the pull-request and then look into the other packet types as well (over the weekend next week). Woaw, this bug was well hidden

mathieu-b, 2017-07-10:
Hi

As far as I can see from:
*  pull request #42 
*  a [search in the current sources](https://github.com/lathoub/Arduino-AppleMidi-Library/search?utf8=%E2%9C%93&q=__attribute__%28%28packed%29%29&type=)

this issue has been solved by using the GCC / Clang-compatible "__attribute__((packed))" directive.

Is this correct @lathoub ?

Regards.

lathoub, 2017-08-13:
AFAIK this issue has been resolved, but I'd like to get confirmation before I close it.
Can you test it @mathieu-b ? (I need to get someone else's Mac in order to test)

lathoub, 2017-08-13:
Tested and worked alright on Mac OS 10.12.6 (using Arduino Ethernet).

OK to close this issue?

[image]

closed 2017-08-13 by lathoub

mathieu-b, 2017-08-14:
Hi  @lathoub 
I don't have a Mac to test this now. Maybe @Bluebie, @chris-zen, or @Hackmancoltaire can confirm if it now also works for them.
Regards.

Hackmancoltaire, 2017-08-14:
It's been a while since I've updated but I can test again soon.

sjc-or, 2019-11-02:
Hmm I see it has been a few years since this problem. But I'm seeing it on my Mega 2560. 
I'm using WiFiEsp package to communicate with may ESP8266 shield. I can receive MIDI on my Arduino but not send to either PC or Mac.  My PC rtpMIDI log shows "Data received is too short for parsing!"

Any ideas. I'm on Mojave on my Mac and Window 10 (of course) on PC.

Should I post my sketch?

lathoub, 2019-11-02:
A similar topic was reported earlier #58 - using also the ESP8266 as a wifi shield. Not sure if it is related.
Alternatively, you can test the new branch, that has a totally new parser: https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0

sjc-or, 2019-11-02:
Thanks, yes this may indeed be the problem. The new parser doesn't seem to fix it so it is probably the WiFiESP library that is broken here for small UDP packet sizes.