lathoub/Arduino-AppleMIDI-Library #36: Set callback from a class function

rolimat, 2016-07-27:
I want to use some functions defined at a class as callback for Midi events. For example, lets say that instead of having:

`void onAppleMidiConnected(uint32_t ssrc, char* name) {
  Serial.println("Connected to session ");
}
AppleMIDI.OnConnected(onAppleMidiConnected);`

What I want to do is define a class AppleMIDIWrapper where I define callbacks:

`class AppleMIDIWrapper
{
public:
    AppleMIDIWrapper();
    void onAppleMidiConnected(uint32_t ssrc, char* name);

private:
}

AppleMIDIWrapper myMIDIWrapper;
AppleMIDI.OnConnected(myMIDIWrapper::onAppleMidiConnected);`
 
@lathoub @Bluebie @Hackmancoltaire any one of you knows how to do that?? I've been searching for a solution on Internet but most of the people say that is not possible :(  I would like to have my main .ino file with only setup and loop functions as well as object implementations.

Thank you all

lathoub, 2016-08-06:
What is the added value of this approach over what is in the lib right now?

I've been looking at subscribing to events using the `+=` addition assignment operator syntax, but this would be a significant change and doesn't add any additional functionality.

rolimat, 2016-08-06:
Well, I don't like to write functions in the .ino main file. In my opinion,
that file should include only setup and loop functions as well as object
creations. Thats how I've learnt oop in java ir javascript that are the
languages I use to work with.

Thats not about adding value or functionality but about maintaining an
order. If you need to write a complex software and use some other libraries
with callbacks you will end up with an ino file with thousands of lines.

Anyway, I dont know if this way of programming is even possible in arduino
C. I was just asking cause I dont have good skills at arduino ;)

closed 2016-12-25 by lathoub