lathoub/Arduino-AppleMIDI-Library #38: Not functional with Arduino Ethernet

DatanoiseTV, 2016-08-21:
[image]

Unfortunately it seems no longer work / or has issues with Arduino Ethernet.
The connection from OSX times out, but firewall etc. is off and the Arduino Ethernet is
on the same network.

DatanoiseTV, 2016-08-21:
My current guess is that the Arduino Ethernet might be running out of memory? It has only 2KB or KB RAM as far as I remember.

lathoub, 2016-12-25:
Correct, the current version is too large for Arduino Ethernet.

lathoub, 2017-08-13:
I've tested the `NoteOnOffEverySec ` on an `Arduino Ethernet `today and mem usage was:

```
Sketch uses 24002 bytes (74%) of program storage space. Maximum is 32256 bytes.
Global variables use 969 bytes (47%) of dynamic memory, leaving 1079 bytes for local variables. Maximum is 2048 bytes.
```

closed 2017-08-13 by lathoub