lathoub/Arduino-AppleMIDI-Library #39: very good on mega, not so much on arduino due and esp-12

alan-dodd, 2016-10-26:
Hello to everyone.
Excellent lib, works fine on mega 2560 + ENC28J60.
But not on 'arduino due' or 'esp-12'. (btw on 'uno' it just does not work for scarce memory)
On these two boards I managed to get ipv4 working (i can ping the boards, and also send and receive udp
 packets), but applemidi has sync problems.
I captured packets on the net with wireshark, seems to me 'due' and 'esp-12' are sending spourious data in between of the correct packets.
Please find attached 2 files with the captured data and comments, and a file with the 2 sketches I used for testing.

[debug mixed mega2560.txt](https://github.com/lathoub/Arduino-AppleMidi-Library/files/553514/debug.mixed.mega2560.txt)
[debug mixed nodemcu.txt](https://github.com/lathoub/Arduino-AppleMidi-Library/files/553513/debug.mixed.nodemcu.txt)
[sketches.txt](https://github.com/lathoub/Arduino-AppleMidi-Library/files/553515/sketches.txt)

Alan

chris-zen, 2016-12-19:
Hi @alan-didd I think the problem you are reporting is related to my fix here #42, could you test it to see if it fixes your problem too ?
Thanks

lathoub, 2016-12-25:
possible fix, see #42

lathoub, 2017-08-13:
Tested on ESP8266 (ESP-12) (Adafruit Huzzah). All OK!

closed 2017-08-13 by lathoub