lathoub/Arduino-AppleMIDI-Library #41: footswitch for Amplitube

ShockLife, 2016-11-07:
Hi man, Im in a project, that connect an arduino to ipad via MIDI, to use with some simulation amps; I was looking for use with bluetooth LE 4.1 or rpt-midi, but I need to connect in a direct form, without a computer, this library do that? thanks !

lathoub, 2016-12-24:
Hi, this lib uses the AppleMIDI protocol over Ethernet, not sure how that works over bluetooth LE. Sorry

Bluebie, 2016-12-24:
iPads cannot make the outgoing connection to other RTP-Midi devices, last time I read about it. Other devices can connect to iPads tho. You could in theory make a wifi hotspot using something like an esp8266 arduino, and then have the arduino thing connect to the ipad automatically when you press a button or automatically keep trying until it connects. That seems possible.

This library can't work with Bluetooth. Bluetooth would be a better way to do it, but i don't know of any open source libraries for doing midi over bluetooth with apple devices.

closed 2016-12-25 by lathoub