lathoub/Arduino-AppleMIDI-Library #45: AfterTouch sends extra data

ftrias, 2017-04-24:
The apple-midi AfterTouch and ProgramChange messages only include one data byte, so data length should be 2. However, data length is set to 3. This causes my Mac (OSX Sierra) to misinterpret the data and create superfluous AfterTouch messages. I suggest a change like the following in AppleMidi.hpp:1630:
```
		// uint8_t length = 16 | 3; // Adds the P-Flag to the length octet
		uint8_t length = 16; // Adds the P-Flag to the length octet
		if (inType == ProgramChange || inType == AfterTouchChannel) {
			length |= 2;
		}
		else {
			length |= 3;
		}
		_contentUDP.write(&length, 1);
```

lathoub, 2017-05-09:
Thanks for spotting that ftrias!

In both cases inData2 is zero. How about a test for (0 == inData2) to void having to name the types specifically> (This also needs to be done on line (1652). However, I'm not sure if other Types set inData2 deliberately to zero to be send. (the function header does mention that when inData2 is zero, it does not need to be send)

<uint8_t length = 16 | (inData2 != 0) ? 3 : 2; // Adds the P-Flag to the length octet>

I prefer that you wrap the changes in a pull request, easier to test and integrate (also make sure you add your name to the contributors on top)

lathoub, 2017-05-14:
Fixed by ftrias (thanks!)

closed 2017-05-14 by lathoub