lathoub/Arduino-AppleMIDI-Library #49: «Help wanted with Time Code»

JLightMedia, 2017-08-26:
I try to receive and send  MIDI Time Code via AppleMIDI. I use Arduino Leonardo ETH in my experiments. But I can't fully understand how work with MTC using your Library. Can you explain or give a basic example of working time code please? And also an additional question. When I create rtpMIDI session in my MacBook and try to connect my arduino I need to manually create new client device(set up IP, Port, Name) and after that I can connect a new client to session. But Apple device self advertise yourself and I just select device and connect to my session. Is it possible to do with AppleMIDI Library?

lathoub, 2017-08-26:
@JLightMedia I have added an example showing how to receive RTC. To send RTC from the Leonardo, use `timeCodeQuarterFrame`

As for your second question: Apple uses Bonjour (ZeroConf) to announce devices on the network. So you will have to add a Bonjour stack on your Arduino Leonardo.

JLightMedia, 2017-08-27:
Thank you for answer! 
Can I ask additional question? 
 Quarter Frame contains two byte the first of that always "F1" and the second data byte. But "AppleMIDI.OnReceiveTimeCodeQuarterFrame" returns only one byte. Is it ok?

lathoub, 2017-08-27:
That is perfectly OK.

0xF1 is the identifier for Quarter-frame messages. Look at the logging in MIDI-OX: status is 0xF1 and DATA1 is the byte of content.

See also here: [https://en.wikipedia.org/wiki/MIDI_timecode](url)

> A quarter-frame messages consists of a status byte of 0xF1, followed by a single 7-bit data value: 3 bits to identify the piece, and 4 bits of partial time code. When time is running forward, the piece numbers increment from 0–7; with the time that piece 0 is transmitted is the coded instant, and the remaining pieces are transmitted later.

JLightMedia, 2017-08-28:
Thank you! Got it!

closed 2017-08-28 by lathoub