lathoub/Arduino-AppleMIDI-Library #53: multiple sessions with one arduino?

RasRomanoff, 2018-01-03:
Hello 

First i want to thank you for this great library!
Im new to all his, so i hope it is ok to write here...

Im running a arduino with esp8622 and rtp-midi (win) - works fine!
The Question: is it possible to set up two simultaneous sessions with one arduino and one windows-pc?
(two mackie conrtolls, more than 16 midi channels...)
cheers, roman

lathoub, 2018-01-03:
Hello

This should be possible, but I have not tried.

replace `APPLEMIDI_CREATE_DEFAULT_INSTANCE();` with:
```
appleMidi::AppleMidi_Class<EthernetUDP> AppleMIDI;
appleMidi::AppleMidi_Class<EthernetUDP> AppleMIDI2;
```
then in setup, begin both instances with:
```
  AppleMIDI.begin("test", CONTROL_PORT);
  AppleMIDI2.begin("test2", CONTROL_PORT + 2);
```

This trick is in `CONTROL_PORT + 2`. Each instance uses 2 `UDP `connections, normally op port `5004 `and `5005`. The 2nd instance will be active on the next available ports: port `5006 `and `5007`

Don't forget in Loop:
```
  AppleMIDI.run();
  AppleMIDI2.run();
```

On Windows (assuming rtpMidi) connect to both instances using 5004 and 5006 reprectively.

Let me know if this works!

RasRomanoff, 2018-01-06:
It works great! 
with low latency also (ESP8266WiFi)... very nice, thanks

lathoub, 2018-01-07:
Cool, glad it worked.
I have created a 2 session example and added it to the repo.
Thx for the suggestion.

If you are happy with the result, please close this issue

RasRomanoff, 2018-01-07:
thanks again ;)

closed 2018-01-07 by RasRomanoff