lathoub/Arduino-AppleMIDI-Library #54: Control Change

arcturus2k, 2018-02-27:
Hi. Sorry for a seemingly simple question.....

I'm using this library with great results on an Adafruit Huzzah. My question is:

How do I format a MIDI controller message? I mostly need to send controller data. 

Thanks in advance. 

(sorry if this is not the correct place to post this question. I could not find another way)

lathoub, 2018-02-27:
In this library you do not format your own messages, all MIDI messages come pre-formatted in the API. All MIDI messages should be supported.

Sending a Control Change message require 3 parameters, 2 related to Control Change (controlNumber and controlValue), the 3 parameters the midi channel. (similar if you receive them)

More information on Control Change here: https://www.midi.org/specifications/item/table-3-control-change-messages-data-bytes-2

arcturus2k, 2018-03-04:
Maybe I didn’t ask the correct question. I understand the construction of a MIDI controller message, but how do I call the actual function in your code? I’m looking for the actual function as I know what to pass to the function.

lathoub, 2018-03-04:
OK. Load the `Esp8266_NoteOnOffEverySec` example, and replace `AppleMIDI.noteOn(note, velocity, channel);` with `AppleMIDI.controlChange(1,2);` replace 1,2 with the controlNumber and controlValue of your choice

In general: look in `src/AppleMidi.h`, at lineNr 183 for the MIDI commands that are available.

arcturus2k, 2018-03-05:
Thanks. That is exactly what I need. I looked through several .h files but did not find the section you referenced - I need to look harder. Thanks again for the help and the pointer to the code. I really appreciate it! 

Steve

lathoub, 2018-03-10:
Closing - question answered

arcturus2k, 2018-03-10:
Thanks. One last question. I can only get a single controller to send at a time. The first cc I address inline in code. Am I missing something? Is there a way to send multiple CCs using this library? 

Thanks again. 

Steve J

lathoub, 2018-03-10:
Hi Steve J

I read your question multiple times, but I'm not sure if I understand (I do not know what your knowledge level of MIDI, programming or Arduino is).

Is this what you want to do? 
```
AppleMIDI.controlChange(1,2);
AppleMIDI.controlChange(1,2);
```

lathoub, 2018-03-25:
@arcturus2k are you happy with answer? then please close this issue

lathoub, 2018-04-10:
going twice...

closed 2018-04-14 by lathoub

arcturus2k, 2018-04-14:
sorry.....I'm good for now. Thank so much for your help!!