lathoub/Arduino-AppleMIDI-Library #55: Pitch Bend Reception/Decoding

massimofasciano, 2018-04-08:
Hi,

It looks like there are 2 issues with the decoding of incoming pitch bend events.

In packet-rtp-midi.h, the following line
int pitch = ( octet1 << 7 ) | octet2;
should be replaced with
int pitch = (( octet2 << 7 ) | octet1) - 8192;

This should correct both issues:
1) the 2 octets (high and low) were reversed (low appears first in the midi stream)
2) the 8192 negative offset was not applied to get a signed int (midi encodes from 0 to 16383 but we expect -8192 to 8191)

With this short fix, it works as expected.
Sending pitch bend was already working using a signed int.

Regards,
Massimo

lathoub, 2018-04-09:
Hi Massimo

Thank you for spotting the pitch bend byte order and offset.
Would you mind creating a Pull request, so that the fix is attributed to you (make sure to add your name at the top of the file)

Thanks!

massimofasciano, 2018-04-09:
Ok thanks. Will prepare a pull request later today.

massimofasciano, 2018-04-09:
Pull request sent... thanks.

closed 2018-04-09 by lathoub