lathoub/Arduino-AppleMIDI-Library #59: [question] is there any example to send and receive MIDI CC?

rinaldohack, 2018-10-03:


rinaldohack, 2018-10-03:
I still confused tbh :(

is there any way to receive everything from rtpmidi, and forward it as MIDI standard to serial?

lathoub, 2018-10-03:
You confuse me even more - is this a question on CC or on how to combine rtpMIDI and serial MIDI?

See here how to use serial MIDI:
https://github.com/FortySevenEffects/arduino_midi_library

So when you receive a note using rtpMIDI:

```
...
MIDI_CREATE_DEFAULT_INSTANCE();
APPLEMIDI_CREATE_DEFAULT_INSTANCE(); 
...
static void OnAppleMidiNoteOn(byte channel, byte note, byte velocity) {
   MIDI.sendNoteOn(note, velocity, channel);
}
```

rinaldohack, 2018-10-03:
basically, how to combine everything to rtpMIDI.
I've successfully combined MIDI note to rtpMIDI, now, MIDI CC, then everything else ... 
sorry man.

lathoub, 2018-10-03:
Feel free to provide an example for everything else

closed 2018-10-03 by lathoub

lathoub, 2018-10-03:
take the NoteOnOffEverySec example, remove the line of `noteOn`, replace `noteOff` with `controlChange`

So:
```
byte controlNumber = 54;
byte controlValue = 12;
byte channel = 1;

controlChange(controlNumber, controlValue, channel);
```

For other commands, see in `AppleMIDI.h`, line 186.

Similar for receiving messages, replace the call to `OnReceiveNoteOn` with `OnControlChange`, see `AppleMIDI.h`, line 268