lathoub/Arduino-AppleMIDI-Library #63: Error: class appleMidi::AppleMidiInterface<WiFiUDP>' has no member named 'read'

oschade, 2019-05-12:
Hello,

this library does not work for me. I get the following errors, no matter, which board I used:

`sketch_may12a:69:13: error: 'class appleMidi::AppleMidiInterface<WiFiUDP>' has no member named 'read'

   AppleMIDI.read();
`

For this, I tried the esp32 example. But others examples doesnt work either.

Do you have some advices?

Thank you!

lathoub, 2019-05-12:
Hello @oschade

I just cloned a new version of the library, and all examples (IDE 1.8.9 - Esp32, WIFI, ESP8266 & Uno) compile without a problem (I test ran on an Arduino Ethernet - all OK)

Please check your environment - what version of the IDE do you run?

oschade, 2019-05-13:
I using IDE 1.8.9.
Thank you for your advice! If I download the lib manually, it runs fine! If I install it over the IDE, I got the error. But the versions of AppleMidi (1.6.0) are the same for both installations.

lathoub, 2019-05-17:
I delete the lib from my libraries and installed it from the IDE (1.6.0), and compiled just fine.

lathoub, 2019-07-04:
were you able to make progress @oschade ?

oschade, 2019-07-04:
> 
> 
> were you able to make progress @oschade ?

I just deleted the library, installed it over the arduino IDE and tried to compile it:
and it worked.

Strange. But thank you!

closed 2019-07-04 by oschade

rinaldohack, 2019-07-11:
I also got this problem recently. Downloading the github zip and extracting it manually solves the issue. 

Probably the one on library manager was broken?