lathoub/Arduino-AppleMIDI-Library #64: Teensy 3.6 + wiz850 io (question)

Sonopopee, 2019-05-15:
I'm not able to get library worked with teensy 3.6 and wiz850io with the "ethernetshield_initiatesession" sketch. The Shield can't obtain IP address via dhcp...(I tried on windows 10 with rtp midi ) I suppose it's normal since wiz850io seems to be not supported by rtp libray (even if ethernet.h does it ?).
Is there any hope it can work ? I missed something ?

If I change for wiz820IO it should work ?

Thanks

lorenzofattori, 2019-05-16:
I used the same components and it worked, give it a try with fixed IP instead of DHCP, that was my only difference

Sonopopee, 2019-05-16:
Thanks for your reply ! 
I already tried with fixed IP, with no more chances, but I probably did something wrong.
I forgot to say I don't use adaptor to connect WIZ850 IO, I just connect it directly to the teensy , with a custom reset code in the setup. It could be the problem ?
Ah, and I tried this config with OSC lib, it works...

lathoub, 2019-05-17:
Hmm, getting an IP address happens outside of the library, and is a managed by the built in Ethernet class. Can you run the UdpNtpClient example? (under Examples > Ethernet)

lathoub, 2019-07-04:
Do you have an update? If this issue can be closed, please do

closed 2019-08-11 by lathoub