lathoub/Arduino-AppleMIDI-Library #68: (not an issue) I made rtpMIDI to MIDI converter, but it is partially done.

rinaldohack, 2019-07-11:
so far, only the NoteOn and NoteOff were tested. but to this part, this code compiles. 
the problem is, I can't finish it, because I'm not that pro at Arduino and stuff. While it is sufficient for my final project, probably you guys want to finish it? 
It is based on ESP8266 (Wemos D1).
 
code : https://pastebin.com/Virjuz1x
this issue also open on FortySevenEffects/arduino_midi_library

lathoub, 2019-07-12:
Hi @rinaldohack
Looking at your code, you are doing it right. Read from rtpMIDI and send it out via (serial) MIDI.
Why don't go the additional step and finish what you started, looks good so far - invest a bit more time in getting to know the Arduino IDE and you have a project you can post on GitHub.
It is beyond the scope of this library to provide the code all the possible implementation of rtpMIDI.
Lathoub

rinaldohack, 2019-07-12:
I thought you want to use it as an example, but okay. I'll try to finish this one.

closed 2020-02-11 by lathoub