lathoub/Arduino-AppleMIDI-Library #70: [no issue] Connect to a remote session

emu1701, 2019-08-28:
Hi, 

I am trying to establish a two way communication between 2 ESPs. One initiates a session. 
The other one should join (connect to) this session to exchange events. 
The communication should be wireless. 

Both ESPs connect to each other and can talk to each other (I tested this via a Webserver/HTTPClient). The "Server" ESP instances a session. The problem is I don't know how to tell the second ESP to connect to this session. 

is there any example that shows how to join a remote session? Or can you provide sample code?

Thanks

luyfigue, 2019-08-30:
You can use the `invite` method:
https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/master/src/AppleMidi.h#L89
e.g:
`AppleMIDI.invite(IPAddress(192,168,0,2), 5004);`

luyfigue, 2019-08-30:
Actually there is an example here:
https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/master/examples/EthernetShield_InitiateSession/EthernetShield_InitiateSession.ino#L56
The other end just needs to do the `AppleMIDI.begin("xyz");`

emu1701, 2019-09-03:
Sorry for the late answer. I wasn't able to access my computer the last days. 

Your solution works perfectly. 

Thanks a lot

closed 2019-09-03 by emu1701