lathoub/Arduino-AppleMIDI-Library #71: When  receive too many MIDI data, it will crash

ZZZ8, 2019-09-08:
I used rtpMIDI V1.1.11  , esp32 KIT V1   , 

When ESP32 receive too much MIDI data,(Receive about 10 midiNotes or midiController at the same time) 
the esp32  will crash.
What is the reason?

thx

lathoub, 2019-09-10:
Not enough information, you will have to expand. Provide code, method to reproduce, etc

ZZZ8, 2019-09-10:
> Not enough information, you will have to expand. Provide code, method to reproduce, etc

It has been added.

ZZZ8, 2019-09-11:
single note     40notes/s   Perfect

lathoub, 2019-10-06:
I have partially rewritten the library, in the [2.0.0 branch](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0). Can you test it?

ZZZ8, 2019-10-15:
`C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/endian.h:61:53: error: macro "ntohl" passed 4 arguments, but takes just 1

 static uint32_t ntohl(byte a, byte b, byte c, byte d)

                                                     ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/endian.h:75:37: error: macro "ntohs" passed 2 arguments, but takes just 1

 static uint16_t ntohs(byte a, byte b)

                                     ^

In file included from C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi.h:20:0,

                 from C:\Arduino\esp32MidiTest\esp32MidiTest.ino:10:

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:105:60: error: macro "ntohl" passed 4 arguments, but takes just 1

    invitation.initiatorToken = ntohl(a[0], a[1], a[2], a[3]);

                                                            ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:112:50: error: macro "ntohl" passed 4 arguments, but takes just 1

    invitation.ssrc = ntohl(a[0], a[1], a[2], a[3]);

                                                  ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:177:60: error: macro "ntohl" passed 4 arguments, but takes just 1

    endSession.initiatorToken = ntohl(a[0], a[1], a[2], a[3]);

                                                            ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:183:50: error: macro "ntohl" passed 4 arguments, but takes just 1

    endSession.ssrc = ntohl(a[0], a[1], a[2], a[3]);

                                                  ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:211:55: error: macro "ntohl" passed 4 arguments, but takes just 1

    synchronization.ssrc = ntohl(a[0], a[1], a[2], a[3]);

                                                       ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:271:56: error: macro "ntohl" passed 4 arguments, but takes just 1

    receiverFeedback.ssrc = ntohl(a[0], a[1], a[2], a[3]);

                                                        ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:275:50: error: macro "ntohs" passed 2 arguments, but takes just 1

    receiverFeedback.sequenceNr = ntohs(a[0], a[1]);

                                                  ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi_Parser.h:278:45: error: macro "ntohs" passed 2 arguments, but takes just 1

    receiverFeedback.dummy = ntohs(a[0], a[1]);

                                             ^

In file included from C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/AppleMidi.h:21:0,

                 from C:\Arduino\esp32MidiTest\esp32MidiTest.ino:10:

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:58:36: error: macro "ntohs" passed 2 arguments, but takes just 1

   rtp.sequenceNr = ntohs(a[0], a[1]);

                                    ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:63:47: error: macro "ntohl" passed 4 arguments, but takes just 1

   rtp.timestamp = ntohl(a[0], a[1], a[2], a[3]);

                                               ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:68:42: error: macro "ntohl" passed 4 arguments, but takes just 1

   rtp.ssrc = ntohl(a[0], a[1], a[2], a[3]);

                                          ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:161:42: error: macro "ntohs" passed 2 arguments, but takes just 1

    uint16_t checkPoint = ntohs(a[0], a[1]);

                                          ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:187:55: error: macro "ntohl" passed 4 arguments, but takes just 1

      uint32_t chanflags = ntohl(0x00, a[0], a[1], a[2]);

                                                       ^

C:\Arduino\libraries\Arduino-AppleMIDI-Library-feat-2.0.0\src/rtpMidi_Parser.h:227:47: error: macro "ntohs" passed 2 arguments, but takes just 1

       const uint16_t header = ntohs(a[0], a[1]);

                                               ^`

lathoub, 2019-10-20:
I added more ESP32 code ([especially on the endianness](https://github.com/lathoub/Arduino_Endian)) and I can get the ESP32 to participate in a session :-). Hopefully you get better results now (not all MIDI messages are parsed in the AppleMIDI parser, still some way to go -looking for volunteers!)

designerfuzzi, 2019-12-27:
a NoteOff is actually a NoteOn Message with Velocity lower then 64 or more often (usually) with velocity (zero) 0. 
Meaning the Callbacks for both can be handled at once and your amazing lib could follow the standard and handle NoteOff as NoteOn with 0 velocity. This way you can also handle stuck notes that have been never closed and implementation of active sensing makes sense then. 

Why? Because as it is now the lib or programmer using it have to compare NoteOn and NoteOff Messages to know which notes are open and largely depend on AllNoteOff messages to close all open while AllNoteOff messages are not send by all AppleMidiNetwork emitters.

Receiving Opening Note Messages on already open Notes would re-trigger the corresponding callback

lathoub, 2020-02-11:
Branch 2.0.0 (heavily modified the parser for combined messages, incl large SysEx)) has been stress tested (Active Sensing at 270ms, Timing Clock at 250 BPM, large repeated Sysex and continuous NoteOn/Off). Can you please test branch 2.0.0 @ZZZ8 ?

ZZZ8, 2020-02-11:
OK .  tomorrow ,i test it.

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

lathoub, 2020-02-11:
Thanks - make sure you test the latest 2.0.0 branch

ZZZ8, 2020-02-13:
[applemidiTest.xlsx](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/4198244/applemidiTest.xlsx)

Test results in the Excel document

ZZZ8, 2020-02-13:
More stable  than last time , No crashes

lathoub, 2020-02-15:
Thanks @ZZZ8 
Interesting results!

Do I understand it correctly that notes are dropped? (miss).
Missing notes without seeing errors in the log is strange - a note in rtpMIDI is surrounded by protocol bytes and counters. So missing a note without causing other protocol error is strange.

Are you able to include Wireshark in your tests?

lathoub, 2020-02-15:
@ZZZ8 Are you sure you are using the latest branch v.2.0.0?

I installed Reaper v6.03 earlier and that revealed some Journal section errors in the code - fixed and uploaded. (Journal section was much larger than what I have seen in the past)

ZZZ8, 2020-02-15:
666, You modify very fast .Thanks .

   I haven’t had time to test the latest version,today. I will test it,these two days

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

lathoub, 2020-02-23:
Let me know how the tests go - I tried with REAPER as well (v6.03/64) on MacOS with no drops
([branch 2.0.2 rc1](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/v2.0.2-rc.1))

ZZZ8, 2020-02-26:
[cid:image001.png@01D5ECC9.863E2920]
1\   Input notes  in the DAW,8notes,9notes,10notes.   and  play it .
 2\  look at the  arduino Serialprot   ,  Number of note.On

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

lathoub, 2020-02-27:
added Session Initiator in branch 2.0.2 (and example)

lathoub, 2020-02-28:
more Initiator testing (see Initiator example), can you test the latest branch feat/2.0.0 (I have not made a release - fetch the latest from feat/2.0.0)
 @ZZZ8

lathoub, 2020-04-21:
Fixed in the latest release 2.0.3 (now in the master branch)

closed 2020-05-01 by lathoub