lathoub/Arduino-AppleMIDI-Library #72: the Lib does not handle RPN or NRPN messages

designerfuzzi, 2019-12-27:
as midi via network is usually used to get midi feedback from studio mixers with lots of parameters wondering why NRPN && RPN is not prepared to be recognized from this library. 
In other words, any networked midi stream containing RPN or NRPN messages will mess up the buffer and treated as normal CC and vice versa. 
You will have to implement LSB && MSB encoding and decoding (bit shifting) and distinguish Control Change Messages from RPN or NRPN with its corresponding LSB/MSB data pairs almost same as you would do for pitch-wheel data.

another thing.. 
Active-sensing should have a timeout to shut down the connection if not received in between 300ms after the last signal.  If it is implemented without timer it is not needed at all, because a stream would stay valid until manually shut down.

designerfuzzi, 2019-12-27:
same for bank-select, one of the most common messages apart from notes and CC.

lathoub, 2019-12-30:
Thanks for the comments, branch [feat/2.0.0](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0) uses the MIDI parser from [47effects](https://github.com/FortySevenEffects/arduino_midi_library) that supports RPN and NRPN. All

feat/2.0.0 isn't fully ready, would be great if you could test

lathoub, 2020-02-07:
@designerfuzzi active-Sensing added to MIDI parser
More features added that need testing. Please check [https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0](url)

designerfuzzi, 2020-02-08:
amazing.. nice done!

closed 2020-02-11 by lathoub