lathoub/Arduino-AppleMIDI-Library #73: [Question]Communication between Arduinos

bauerbyter, 2020-01-07:
Hey,
sorry, I was looking around but I dont find an answer.
Is it possible to communicate with your library beetween multiple Arduinos (or at least two) without any PC, or something similar?

I want a (or more) Footcontroller with  ESP32 and a Box (also ESP32) next to my amp which is translating the apple midi to standard midi.

Thanks in advance

Greetings

alf45tar, 2020-01-07:
The feature you are requesting is already implemented into PedalinoMini.

bauerbyter, 2020-01-07:
Okay thats amazing. But how do I do this? (should I ask this in a PedalinoMini Issue?)

alf45tar, 2020-01-07:
It is better to move the discussion into PedalinoMini because is not AppleMidi specific.

AppleMidi library is a session listener over WiFi. It does not support session initiation functionalities, which requires the use of an external session initiator on the network to open a RTP-MIDI session with the libray. This session initiator can be a macOS computer (Audio MIDI Setup->MIDI Studio->MIDI Network Setup) or a Windows computer with the RTP-MIDI driver, an app on your iOS device (for example MIDI Network) or an embedded RTP-MIDI device.

PedalinoMini implement the feature you requested via ipMidi because it do not request a computer.

bauerbyter, 2020-01-07:
okay thanks again!

closed 2020-01-07 by bauerbyter