lathoub/Arduino-AppleMIDI-Library #74: Incompatible with FortySevenEffects MIDI library

fastfourier666, 2020-02-24:
Hi, thanks for the 2.0.0 branch - I was getting disconnects with the default branch when sending lots of notes.

I am making a DIN MIDI to RTP MIDI converter which connects to a remote session. Previously I was using AppleMIDI.invite (ip, port) for this, but now it doesn't exist. I did find sendInvite but I get errors compiling when using that.

lathoub, 2020-03-01:
I updated 2.0.0 branch - can you test again @fastfourier666 
(now also includes the AppleMIDI Initiator part - see example Initiator)

lathoub, 2020-04-21:
This should be fixed in the new v2.0.3 release (see [EthernetShield_Initiator](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/examples/EthernetShield_Initiator) example)

closed 2020-05-01 by lathoub