lathoub/Arduino-AppleMIDI-Library #75: Is feat/2.0.0 compatible with FortySevenEffects/arduino_midi_library?

alf45tar, 2020-03-01:
Both create a MIDI object. Can they coexist?

Thanks
al45tar

lathoub, 2020-03-01:
feat/2.0.0 **uses** 47Effects/arduino_midi_lib (branch feat/4.4.0see src/utility/midi_feat4_4_0) thru static polymorphism.

So the `MIDI` is an abstract MIDI object (the transport Layer has been removed) and AppleMIDI fills in the transport layer (ipMIDI is another transport layer, or Serial, or BLE,... - all using the same abstract MIDI core).

(in src/utility/midi_feat4_4_0 `MIDI.h` search for `Encoder` - this will be filled in by the second object: AppleMIDI, ipMIDI, ...)

Makes sense?

alf45tar, 2020-03-01:
Can you provide a small example how to use a couple of transport layers?

lathoub, 2020-03-01:
Not sure what you are asking for.

Examples like USB, BLE, ipMIDI - or actual implementations? (or are you asking how static polymorphism works in this case?)

Here is an easy one to learn a bit of all: [ipMIDI](https://github.com/lathoub/Arduino-ipMIDI)

alf45tar, 2020-03-01:
How can I send a MIDI message (like a NoteOn) via AppleMIDI and via Serial in the same program?
The command is always the same MIDI.sendNoteOn(note, velocity, channel). How can I change the name of MIDI object to differentiate different trasport layer?

lathoub, 2020-03-02:
```
#include <AppleMidi.h>
#include <MIDI.h>
...
typedef APPLEMIDI_NAMESPACE::AppleMidiSession<EthernetUDP> AppleMidiSession_t;
AppleMidiSession_t Session1("Session 1", 5004);
MIDI_NAMESPACE::MidiInterface<AppleMidiSession_t> AppleMidiCore((AppleMidiSession_t &)Session1);

MIDI_CREATE_DEFAULT_INSTANCE();
...
AppleMidiCore.sendNoteOn(27, 127, 1);
MIDI.sendNoteOn(27, 127, 1);
...
```

closed 2020-04-20 by alf45tar