lathoub/Arduino-AppleMIDI-Library #76: Can not wrap instance of the AppleMidi library in an own class

luc-debug, 2020-03-24:
Hello guys,
as described i would like to wrap the use of the library in a class.
-----------------------------------------------------------------------
class WirelessMidi
{
public:
    WirelessMidi();
    ~WirelessMidi();

    appleMidi::AppleMidiInterface<WiFiUDP> *appleMidiProperty;
    static void OnAppleMidiConnected(uint32_t ssrc, char *name);  
};
-----------------------------------------
#include <WirelessMidi.h>
#include <WiFiUdp.h>

WirelessMidi::WirelessMidi()
{
    APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI);
    this->appleMidiProperty=&AppleMIDI;
}

WirelessMidi::~WirelessMidi()
{
}

void WirelessMidi:: OnAppleMidiConnected(uint32_t ssrc, char *name)
{
  isConnected = true;
  Serial.print(F("Connected to session "));
  Serial.println(name);
}
--------------------------------------
WirelessMidi midi;
 midi.appleMidiProperty->begin("test");
------------------------------------------

This is the error I get:
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\az-delivery-devkit-v4\firmware.elf] Error 1

Any help is highly appreciated. Thank you in advance.

lathoub, 2020-03-24:
@luc-debug this is not so much of a AppleMIDI library question, but rather a C++ question.
(I will not ask what you are trying to accomplish - out of scope of the issues section of the library)

However:
```
WirelessMidi::WirelessMidi()
{
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI);
this->appleMidiProperty=&AppleMIDI;
}
```
would create an instance of `AppleMIDISession` in the constructor - and bring it back down when the constructor is out of scope, with a stale reference in this->appleMidiProperty.

See what is deflated and instantiated in the `APPLEMIDI_CREATE_INSTANCE` macro (AppleMIDI.h, towards the end).

closed 2020-03-24 by lathoub

luc-debug, 2020-03-24:
Awesome thank you so much for your fast response.
Sorry for the dump question I thought it would be related to the library, I am pretty new to C++.
Anyway thank you for your C++ hint.

lathoub, 2020-03-24:
no worries @luc-debug 
there is actually no need to wrap the instance - `APPLEMIDI_CREATE_INSTANCE` creates the instance you can use right away

luc-debug, 2020-03-24:
I want to clean up my main file and give my code a object-oriented design. That is why I want to wrap all functionalities (like the instance of the library, event handlers, IsConnected bool variable) of your (very awsome!) library in my own class.
Thanks to your advice I know now that I have to improve my C++ skills to solve it.

luc-debug, 2020-03-25:
@lathoub I am right in assuming that I can only declare the AppleMidi instance globally?

lathoub, 2020-03-25:
the `AppleMIDI` instance has the same lifetime/scope of any other c++ instance (incl: `int`) (it's just a bit more complex using templates - do not be blinded by it. Start with simple things, then work your way up). Good luck.