lathoub/Arduino-AppleMIDI-Library #78: Question: Possible to join a RTP session?

microfx, 2020-04-10:
Hey!

The example shows how to initiate a RTP session. Is it also possible to join a MIDI session? I got a MIDI router which always has a few open which can be joined.

Stay safe! 

Cheers

lathoub, 2020-04-11:
Have a look at the [InitiateSession ](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/examples/EthernetShield_InitiateSession)example

microfx, 2020-04-11:
Reading the example... I can see no difference. Initiating is not joining an existing session, is it?

lathoub, 2020-04-11:
Any other example then: https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/examples/EthernetShield_NoteOnOffEverySec

microfx, 2020-04-11:
So what‘s different here?

  // Create a session and wait for a remote host to connect to us

where is this another example? This is still the other way around?

lathoub, 2020-04-13:
> I got a MIDI router which always has a few open which can be joined

Not sure how to answer, are you familiar with the RtpMidi (AppleMIDI) protocol?
Most examples wait for the **session initiator** to connect to the Arduino (Is your` MIDI router` a session initiator?)

microfx, 2020-04-13:
Yes, my MIDI router is a session initiator. iConnectivity mio10. I use rtp MIDI all the time (connecting macOS/iOS to the sessions of the router)

lathoub, 2020-04-13:
Then start with the [NoteOn/Off](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/examples/EthernetShield_NoteOnOffEverySec) example

rockomatthews, 2020-04-13:
It is a slightly different experience and would seem like the other way around because you have to manually create the Arduino session.  Pretty sure what he is saying is he is expecting this connection step to look different.  My iPhone in the picture is "Projector Bach" since I use the app for MIDI Designer Pro 2 App on my iPhone.  With that, your device automatically appears in the "dictionary" after initiating a session on 5004.  Only difference is that there are steps described on how to add the Arduino in the comments.  But other things just pop up if that helps...

[image]

lathoub, 2020-04-21:
Did you get it to work?

closed 2020-05-09 by lathoub