lathoub/Arduino-AppleMIDI-Library #81: update

rockomatthews, 2020-04-22:
What are some reasons I should or shouldn't update to this build?  Sounds awesome. I'm using an esp-32 though and Mojave... I've stuck with the build right before the alpha because I had trouble figuring out how to make it work with an esp-32.  Now even, it looks like esp-32's require extra helpers?  Is that a good sign or bad?  I have no idea. I am not the sharpest coder.

lathoub, 2020-04-22:
Hi @rockomatthews, I would update right away, here is why:

- faster and more stable
- no dynamic memory allocation
- stress tested
- cleaner design (separation of transport layer from MIDI layer) (2 objects: AppleMIDI & MIDI; the MIDI object is the pure FortySevenEffects MIDI library, AppleMIDI to pick on specific events).

Migration is easy, take any of the examples and see what changed (also, I have a [wiki page](https://github.com/lathoub/Arduino-AppleMIDI-Library/wiki/Migration-from-v1.*-to-v2.*) on it)

As for the ESP32, excellent choice! The ESP32 examples I have is for the Ethernet wired hardware (eg wESP32), that used a Helper. Your code of the ESP32 is just fine.

Looking at your code, you should consider using a Library for the buttons, I like this one: [Bounce2](https://github.com/thomasfredericks/Bounce2) - your code has 4 repeating sections (making to code harder to read, error prone and use more memory).

Also, add a separate helper file to hide your ssid and  password :-)

lathoub, 2020-05-01:
Pls close this issue if you are happy with the answer

closed 2020-05-09 by lathoub