lathoub/Arduino-AppleMIDI-Library #85: AppleMIDI on ESP8266 NodeMCU

ciapamusche, 2020-05-11:
Hi . I'm tryng to use the library with a NodeMCU board but i dont find examples.

Is still the library compatible?

lathoub, 2020-05-15:
Hi @ciapamusche thanks for letting me know! I don't have any 8266 hardware anymore, but added an ESP8266 example, based on the 1.6* release. Can you check if it runs OK before I bump the version and re-release?

ciapamusche, 2020-05-15:
hey thank you... I cannot see the example...  I go in 1.6 and 1.6.1 but example are not updated.in which release I have to see..? 

thank you

ciapamusche, 2020-05-15:
Hey thank you.. the sketch not compile.

Arduino:1.8.12 (Windows 10), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Baracca\Documents\Arduino\pedaliera_2\sketch_may14a\sketch_may14a.ino:13:0:

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h: In instantiation of 'bool appleMidi::AppleMIDISession<UdpClass, Settings, Platform>::beginTransmission(midi::MidiType) [with UdpClass = WiFiUDP; _Settings = appleMidi::DefaultSettings; _Platform = appleMidi::DefaultPlatform]':

C:\Users\Baracca\Documents\Arduino\libraries\arduino_midi_library-5.0.2\src/MIDI.hpp:180:9:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::send(midi::MidiType, midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<WiFiUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'

C:\Users\Baracca\Documents\Arduino\libraries\arduino_midi_library-5.0.2\src/MIDI.hpp:230:53:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::sendNoteOn(midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<WiFiUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'

C:\Users\Baracca\Documents\Arduino\pedaliera_2\sketch_may14a\sketch_may14a.ino:259:44:   required from here

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: error: ambiguous overload for 'operator!=' (operand types are 'IPAddress' and 'int')

   return (dataPort.remoteIP() != 0 && participants.size() > 0);

                               ^

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: candidates are:

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: operator!=(uint32_t {aka unsigned int}, int) <built-in>

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: operator!=(u32_t {aka long unsigned int}, int) <built-in>

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: operator!=(int, int) <built-in>

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: operator!=(const ip_addr_t* {aka const ip4_addr*}, const ip_addr_t* {aka const ip4_addr*}) <built-in>

C:\Users\Baracca\Documents\Arduino\libraries\Arduino-AppleMIDI-Library-master\src/AppleMIDI.h:126:31: note: operator!=(ip_addr_t* {aka ip4_addr*}, ip_addr_t* {aka ip4_addr*}) <built-in>

In file included from C:\Users\Baracca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,

                 from C:\Users\Baracca\Documents\Arduino\pedaliera_2\sketch_may14a\sketch_may14a.ino:1:

C:\Users\Baracca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266/IPAddress.h:107:14: note: bool IPAddress::operator!=(const IPAddress&) const

         bool operator!=(const IPAddress& addr) const {

              ^

C:\Users\Baracca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266/IPAddress.h:116:14: note: bool IPAddress::operator!=(uint32_t) const

         bool operator!=(uint32_t addr) const {

              ^

C:\Users\Baracca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266/IPAddress.h:119:14: note: bool IPAddress::operator!=(u32_t) const

         bool operator!=(u32_t addr) const {

              ^

Let me know for testing…..

lathoub, 2020-05-15:
check the feat/v2.1.0 branch

ciapamusche, 2020-05-15:
Hi . The sketch works and midi note is received by DAW correctly but I noticed a strange issue.

My DAW is installed on a Mac attached to the network by lan.

If I try to connect my esp It connect correctly and give an IP but from my Mac I cannot reach it. Not ping and the esp is not expose to Midi network.
Only if I connect my Mac on wifi I can see the ESP on network.

Please note all the other network devices compatible with Apple network don’t have this problem. They works correctly with the Mac connected to the network 

Any ideas?

ciapamusche, 2020-05-16:
Update. Different NodeMCU have different behaviour with same code. So. The problem maybe the board.
I can confirm with new branch NodeMcu works with less 4ms of delay.... good job thank you

lathoub, 2020-05-16:
Good to hear @ciapamusche , when you are ready please close this issue

closed 2020-05-16 by ciapamusche