lathoub/Arduino-AppleMIDI-Library #88: esp8266 build error

leecuk, 2020-06-01:
Hi, I'm sure I'm doing something silly and hopefully someone can help please.

I'm trying to build the really simple example on the front page but can't get a clean build if I pick any esp8266 board. I'm using the latest 2.1.0 branch with version 2.7.1 of the esp8266 core with Arduino 1.8.12 on a Mac running Catalina.

Many thanks
Lee

The message is as follows - 

```
Arduino: 1.8.12 (Mac OS X), Board: "SparkFun ESP8266 Thing Dev, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 512KB (FS:32KB OTA:~230KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/me/Desktop/recordlight/recordlight.ino: In function 'void setup()':
recordlight:13:52: error: invalid conversion from 'void (*)(uint32_t, const char*) {aka void (*)(unsigned int, const char*)}' to 'void (*)(const ssrc_t&, const char*) {aka void (*)(const unsigned int&, const char*)}' [-fpermissive]
   AppleMIDI.setHandleConnected(OnAppleMidiConnected);
                                                    ^
In file included from /Users/me/Desktop/recordlight/recordlight.ino:1:0:
/Users/me/Documents/Arduino/libraries/Arduino-AppleMIDI-Library-feat-v2.1.0/src/AppleMIDI.h:54:7: error:   initializing argument 1 of 'void appleMidi::AppleMIDISession<UdpClass, Settings, Platform>::setHandleConnected(void (*)(const ssrc_t&, const char*)) [with UdpClass = EthernetUDP; _Settings = appleMidi::DefaultSettings; _Platform = appleMidi::DefaultPlatform; appleMidi::ssrc_t = unsigned int]' [-fpermissive]
  void setHandleConnected(void (*fptr)(const ssrc_t&, const char*)) { _connectedCallback = fptr; }
       ^
Multiple libraries were found for "Ethernet.h"
 Used: /Users/me/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/Ethernet
 Not used: /Applications/Arduino.app/Contents/Java/libraries/Ethernet
exit status 1
invalid conversion from 'void (*)(uint32_t, const char*) {aka void (*)(unsigned int, const char*)}' to 'void (*)(const ssrc_t&, const char*) {aka void (*)(const unsigned int&, const char*)}' [-fpermissive]

```

leecuk, 2020-06-21:
Just a quick update, the v1.* branch works great, so I've used that in my project.

lathoub, 2020-10-18:
Should be fixed in 2.2.0 (update via GitHub, not via the LIbrary manager) , pls check

closed 2020-10-18 by lathoub