lathoub/Arduino-AppleMIDI-Library #90: ESP32 does not answer the connection request made by rtpmidi software

HassamK97, 2020-06-22:
Hi, I am using autoconnect library to connect esp32 to wifi. When it is connected, I start the midi session connection. It appears in rtpmidi software as green and on the network as well. But when I connect it, I receive an error that 192.168.8.108:5004 has not answered the connection request. The code is attached as follows:

`#include <WiFi.h>
#include <WebServer.h>
#include <time.h>
#include <AutoConnect.h>
#include <ESPmDNS.h>

#define APPLEMIDI_INITIATOR
#include <AppleMIDI.h>
USING_NAMESPACE_APPLEMIDI

bool wifiConnected = false;

APPLEMIDI_CREATE_DEFAULTSESSION_ESP32_INSTANCE();

int sensorPin = 34;    // select the input pin for the potentiometer
int initialVal = 0;      //Our initial potentiometer values. We need one for the first value and a second to test if there has been a change made.
int lastVal = 0;
int midiVal = 0;

static const char AUX_TIMEZONE[] PROGMEM = R"(
{
  "title": "TimeZone",
  "uri": "/timezone",
  "menu": true,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "value": "Sets the time zone to get the current local time.",
      "style": "font-family:Arial;font-weight:bold;text-align:center;margin-bottom:10px;color:DarkSlateBlue"
    },
    {
      "name": "timezone",
      "type": "ACSelect",
      "label": "Select TZ name",
      "option": [],
      "selected": 10
    },
    {
      "name": "newline",
      "type": "ACElement",
      "value": "<br>"
    },
    {
      "name": "start",
      "type": "ACSubmit",
      "value": "OK",
      "uri": "/start"
    }
  ]
}
)";

typedef struct {
  const char* zone;
  const char* ntpServer;
  int8_t      tzoff;
} Timezone_t;

static const Timezone_t TZ[] = {
  { "Europe/London", "europe.pool.ntp.org", 0 }
};

WebServer Server;

AutoConnect       Portal(Server);
AutoConnectConfig Config;       
AutoConnectAux    Timezone;
[... 183 lines cut ...]
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
static void OnAppleMidiNoteOn(byte channel, byte note, byte velocity) {
  Serial.print(F("Incoming NoteOn  from channel: "));
  Serial.print(channel);
  Serial.print(F(", note: "));
  Serial.print(note);
  Serial.print(F(", velocity: "));
  Serial.println(velocity);
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
static void OnAppleMidiNoteOff(byte channel, byte note, byte velocity) {
  Serial.print(F("Incoming NoteOff from channel: "));
  Serial.print(channel);
  Serial.print(F(", note: "));
  Serial.print(note);
  Serial.print(F(", velocity: "));
  Serial.println(velocity);
}`

lathoub, 2020-12-20:
Fixed in the latest Release (v2.2.0) - pls check

lathoub, 2020-12-30:
Fix taken up in #98

closed 2020-12-30 by lathoub