lathoub/Arduino-AppleMIDI-Library #92: ESP32 only few Midi callbacks working

Benceking24, 2020-10-05:
Hy!
I used the older version 1.5.5 version way-way back and now I'm trying to use the updated library. I managed to use an esp32 over wifi to send notes, recieve songposition and controlchanges but nothing else seems to work. I looked trough the 47's arduino_midi_library both can't find the error. Could you please help?
Testing methode: Ableton live running and midiview app used to monitor it's output. Esp32 connected directly with a usb cable for serial monitoring and trough wifi for rtp connection. The midiview app shows that midi clock information, start/stop status and everything else is sent from ableton live and when i switch over to the esp none of my callback functions are running.

[esp32_rtpMidi_callback.txt](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/5328200/esp32_rtpMidi_callback.txt)
[ETH_helper.txt](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/5328201/ETH_helper.txt)
[Compiler output.txt](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/5328203/Compiler.output.txt)

Benceking24, 2020-10-05:
I tried out the same code with the usb library on a sparkfun pro micro and every function and callback works there. Could you please take a look at it?

Benceking24, 2020-10-14:
With the setHandleError, setHandleReceivedRtp, setHandleStartReceivedMidi, setHandleReceivedMidi, setHandleEndReceivedMidi handles set and routed to serial output when i started the daw i get a constant
RECIEVE RTP (ssrc_t, RTP: vpxcc, mpayload, sequenceNr, timestamp, ssrc, int32_t) 0; 128; 97; 3060; 23208785; 4075989739; -110 message with the int32_t, timestamp and sequenceNr changing each time, but everything else stays fixed. However the callbacks i noticed not working doesen't trigger the setHandleReceivedMidi handle at all! (others like song position and incoming midi trigger it and works great!)

lathoub, 2020-10-17:
Thanks @Benceking24 for sending the issue: help me understand better: what hardware are you using? Is it a Wired ESP32 (eg wESP32) or using normal Wifi ESP32 (the ETH_helper threw me off).

> Testing methode: Ableton live running and midiview app used to monitor it's output. Esp32 connected directly with a usb cable for serial monitoring and trough wifi for rtp connection. The midiview [...]

Questions:

- so Wifi or over ETH? (The ETH code is commented out in the sketch (using a .txt extension) - but you include the ETH_helper)
- I'm confused by  'i switch over to the esp'; how do you switch? from what?

Benceking24, 2020-10-17:
Sorry for not being clear :( the main problem occurs with an esp 32 using wifi. I set up your library with the esp32 instance macro and midi begin...etc that is listed in the readme and after that only few of the callbacks works. If it is still not that clear I could send an other sletch!
The change part was only about that my computer side was doing ok and specific clock and sysex commands are being sent from it for sure. The setup was made of a arduino pro micro and your usb midi library and with those and the same computer setup all the callbacks works.

lathoub, 2020-10-17:
Posting the sketch here is a good, so we have the same starting point. (remove all non-essential code). I was confused that you included the `ETH_helper` as this is specifically for Ethernet enabled ESP32 (like the [wEsp32](https://wesp32.com/))

Benceking24, 2020-10-17:
This sketch was uploaded to an NodeMCU Esp32S dev board and the outgoing messages all worked from the ESP but only a few callbacks worked in the other direction, like the song position, command change BUT not the clock or start/stop/continue ones.
[appleMidiEsp32WifiCallback error.txt](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/5395515/appleMidiEsp32WifiCallback.error.txt)

lathoub, 2020-10-17:
Thanks @Benceking24 ([you can post code as part of MarkDown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet), no need to attach a .txt)

It looks like (at first glance) an endianness issue (Arduino Ethernet is little endian, ESP32 is big endian) - I will look into it

Benceking24, 2020-10-17:
Thank you very much! :)

lathoub, 2020-10-18:
Update: It's not an Endian issue (esp32 is _not_ big endian). I do see however now an issue with rtpMidi on Windows using Tobias Erichsen software version 1.1.14 (247) sending initial invite twice. I need to resolve that first, so i can use MidiOx to send messages.

lathoub, 2020-10-18:
Bug fixed - turned out that no callbacks were fired for realtime data - works for me (using Midi_Ox on Windows).
Please test again @Benceking24

Benceking24, 2020-10-18:
With the [39b0e417537dcde5ea5b7cafbdc99832839699f9](https://github.com/lathoub/Arduino-AppleMIDI-Library/commit/39b0e417537dcde5ea5b7cafbdc99832839699f9) commit it works perfectly for me! Thank you for resolving this issue :)

closed 2020-10-18 by Benceking24