lathoub/Arduino-AppleMIDI-Library #93: APPLEMIDI_CREATE_INSTANCE" requires 4 arguments, but only 2 given

simonegiusti, 2020-12-04:
```
"APPLEMIDI_CREATE_INSTANCE" requires 4 arguments, but only 2 given
 APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI); // see definition in AppleMidi_Defs.h
```

trying to compile on Esp8266 NodeMCU v2

I tried to add missing arguments looking for definition.

`APPLEMIDI_CREATE_INSTANCE(WiFiUDP, AppleMIDI, "ardu", 5004); // see definition in AppleMidi_Defs.h`

... a new error comes on AppleMIDI.stop();

`'class midi::MidiInterface<appleMidi::AppleMIDISession<WiFiUDP>, appleMidi::AppleMIDISettings>' has no member named 'stop'`

finally I tried to rem AppleMIDI.stop and an error on   AppleMIDI.begin("midi1button"); and ...

               ^
C:\Users\simon\Documents\Arduino\sketch_dec04c\sketch_dec04c.ino: At global scope:
C:\Users\simon\Documents\Arduino\sketch_dec04c\sketch_dec04c.ino:197:6: warning: unused parameter 'ssrc' [-Wunused-parameter]
 void OnAppleMidiConnected(uint32_t ssrc, char* name) {
      ^
C:\Users\simon\Documents\Arduino\sketch_dec04c\sketch_dec04c.ino:206:6: warning: unused parameter 'ssrc' [-Wunused-parameter]
 void OnAppleMidiDisconnected(uint32_t ssrc) {
      ^
```
C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi 
invalid conversion from 'const char*' to 'midi::Channel {aka unsigned char}' [-fpermissive]

```

lathoub, 2020-12-20:
Fixed in the latest Release (v2.2.0) - pls check

lathoub, 2020-12-30:
Fix taken up in #98

closed 2020-12-30 by lathoub

pineflowers, 2021-12-14:
I'm still having this exact issue on 3.1.2 and a similar chip. I tried 2.2.0 and also had the issue.