lathoub/Arduino-AppleMIDI-Library #94: Connectivity Issues When Using Ethernet Switch & Two Teensy 4.1s

glasscake, 2020-12-12:
Not sure what would be the best way to go about helping narrow down the issue.

Right now if I have each teensy plugged into isolated ethernet ports on my laptop (one built in one usb-ethernet) the system works flawlessly. If I connect both Teensys' to a switch then the switch to my laptop (isolated from the internet) they will have a difficult time connecting, staying connected, reconnecting, receiving commands, ext.

I am currently using a POE switch, not sure if that would mess things up but I will try a regular switch just for DD.

Here is my code. There are other functions but im not sure that is relevant to this issue.
Note each teensy is set up with a unique IP so that's not the issue. 

//Start Libaries to include
#include <Wire.h>
#include <NativeEthernet.h>
#include <AppleMIDI.h>
USING_NAMESPACE_APPLEMIDI
//End Libaries to include

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
            //169.254.142.141
byte ip[] = {169,254,142,147};

APPLEMIDI_CREATE_DEFAULTSESSION_INSTANCE();

//int VacuumSensor = A8;
int Mic = A9;

int IOEBytes[10];//the last used Input Output Expanders

elapsedMillis CurrentMillis;
unsigned long KeyDelayTime = 130; //Just slightly longer than the slowest key

unsigned long IOEOnByteTimer[80];
unsigned long IOEOffByteTimer[80];

//Key  1 2 3 4 5 6 7 8 9 10 . .. . .      //Original times
unsigned long IOEOnByteTimerModifer[80] = {120,84,81,72,90,71,72,79,92,79,98,83,117,78,78,79,74,74,94,74,81,83,66,75,78,81,93,85,92,83,69,48,117,0,73,77,61,74,84,78,68,77,86,76,77,93,117,87,78,68,82,80,0,66,60,114,68,78,0,60,70,69,79,101,83,67,65,72,66,83,61,72,104,84,64,69,70,58,86,64};
unsigned long IOEOffByteTimerModifer[80] = {10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10,10};

int IOEPosition = 0;
int IOEByteTimmerPosition = 0;
bool IOEChange[10];

void setup() 
{
  //Serial.begin(115200);
  Ethernet.begin(mac, ip);
  pinMode(Mic, INPUT);
  Wire.begin();
  Wire.setClock(400000);
  MIDI.begin();
  //setup handlers for RTP midi commands
  AppleMIDI.setHandleConnected(OnAppleMidiConnected);
  AppleMIDI.setHandleDisconnected(OnAppleMidiDisconnected);
  //set up handlers for basic midi commands
  MIDI.setHandleNoteOff(MidiNoteOff);
  MIDI.setHandleNoteOn(MidiNoteOn);
  MIDI.setHandleControlChange(MidiControlChange);
  //end setup handlers
  for (int i = 0; i <= 79; i++)
  {
    if (IOEOnByteTimerModifer[i] <= KeyDelayTime  and IOEOnByteTimerModifer[i] != 0) //only run if the digital delay is longer than the physical delay
    {
      //create the difference between the physical delay and the digital delay so all keys play at the same time
      IOEOnByteTimerModifer[i] = KeyDelayTime - IOEOnByteTimerModifer[i];
    }
    if (IOEOffByteTimerModifer[i] <= KeyDelayTime  and IOEOffByteTimerModifer[i] != 0) //only run if the digital delay is longer than the physical delay
    {
      //create the difference between the physical delay and the digital delay so all keys play at the same time
      IOEOffByteTimerModifer[i] = KeyDelayTime - IOEOffByteTimerModifer[i];
    }
  }
  SoftStartReset();
  //only used for debugging, records how long it takes for the on key command to be read/sent to the key making noise
  //RecordKeyOnDelay();
}

void SoftStartReset()
{
  //CurrentMillis = millis();
  Ethernet.begin(mac, ip);
  SetupIOExpanders();         //setup the IOE as outputs
  SendIOAllChannelOff();      //set all the channels to off
  SetTimerArrayZero();        //set all the timmers to 0, also redudent but let check notes turn notes off
  for (int i = 0; i <= 9; i++)
  {
    //set all the bytes to 0 and all changes to false
    IOEBytes[i] = 0;
    IOEChange[i] = false;
  }
}

void loop()
{
  CheckNotes(5);                                  //check a number of IOE registers at a time
  MIDI.read(1);                                //read any midi messages on the first channel in the que
}

lathoub, 2020-12-12:
It sounds like you are dropping packets (rtpMIDI uses UDP packets). What does WireShark say?
(also, when degugging, use the simple examples that come with the library and see if the bug still exist)

glasscake, 2020-12-14:
Hello,

Sorry I will lag on testing this, I only have a couple hours a week to work on this project. Please do not close the thread.

Which example is the best? the Send Receive? Also since this is on a teensy the ethernet library is not the same, not sure if that could be causing issues.

Thanks

lathoub, 2020-12-17:
Start with the [NoteOnOff ](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/examples/EthernetShield_NoteOnOffEverySec) example. If you have it working for the latest Teensy, I'd be happy to accept your pull request

lathoub, 2020-12-20:
Please upgrade to the v.2.2.0 release before checking

glasscake, 2020-12-24:
After upgrading to 2.2.0 and trying to compile the ethernet shield note on and off every second example i get the following error: (note the only thing changed is the <Ethernet> library to <NativeEthernet>

In file included from ----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:5:0:
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h: In instantiation of 'bool appleMidi::AppleMIDISession<UdpClass, Settings, Platform>::beginTransmission(midi::MidiType) [with UdpClass = EthernetUDP; _Settings = appleMidi::DefaultSettings; _Platform = appleMidi::DefaultPlatform]':
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:180:9:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::send(midi::MidiType, midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:230:9:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::sendNoteOn(midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'
----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:76:44:   required from here
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:131:31: error: no match for 'operator!=' (operand types are 'IPAddress' and 'const IPAddress')
   return (dataPort.remoteIP() != INADDR_NONE && participants.size() > 0);
                               ^
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:131:31: note: candidate: operator!=(uint32_t {aka long unsigned int}, uint32_t {aka long unsigned int}) <built-in>
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:131:31: note:   conversion of argument 2 would be ill-formed:
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:131:31: warning: passing 'const IPAddress' as 'this' argument discards qualifiers [-fpermissive]
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/Client.h:26:0,
                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src/NativeEthernet.h:56,
                 from ----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:1:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/IPAddress.h:76:2: note:   in call to 'IPAddress::operator uint32_t()'
  operator uint32_t () {
  ^
In file included from ----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI_Parser.h:3:0,
                 from ----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:22,
                 from ----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:5:
----\Documents\Arduino\libraries\AppleMIDI\src/utility/Deque.h: In instantiation of 'void Deque<T, Size>::push_back(const T&) [with T = unsigned char; unsigned int Size = 64u]':
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:124:17:   required from 'bool appleMidi::AppleMIDISession<UdpClass, Settings, Platform>::beginTransmission(midi::MidiType) [with UdpClass = EthernetUDP; _Settings = appleMidi::DefaultSettings; _Platform = appleMidi::DefaultPlatform]'
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:180:9:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::send(midi::MidiType, midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:230:9:   required from 'void midi::MidiInterface<Transport, _Settings, _Platform>::sendNoteOn(midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]'
----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:76:44:   required from here
----\Documents\Arduino\libraries\AppleMIDI\src/utility/Deque.h:120:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (++_head >= Size)
                     ^
----\Documents\Arduino\libraries\AppleMIDI\src/utility/Deque.h: In instantiation of 'void Deque<T, Size>::pop_front() [with T = unsigned char; unsigned int Size = 64u]':
----\Documents\Arduino\libraries\AppleMIDI\src/AppleMIDI.h:202:9:   required from 'byte appleMidi::AppleMIDISession<UdpClass, Settings, Platform>::read() [with UdpClass = EthernetUDP; _Settings = appleMidi::DefaultSettings; _Platform = appleMidi::DefaultPlatform; byte = unsigned char]'
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:800:44:   required from 'bool midi::MidiInterface<Transport, _Settings, _Platform>::parse() [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform]'
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:742:15:   required from 'bool midi::MidiInterface<Transport, _Settings, _Platform>::read(midi::Channel) [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; [... cut ...]
----\Documents\Arduino\libraries\MIDI_Library\src/MIDI.hpp:706:16:   required from 'bool midi::MidiInterface<Transport, _Settings, _Platform>::read() [with Transport = appleMidi::AppleMIDISession<EthernetUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform]'
----\Documents\Arduino\sketch_dec23b\sketch_dec23b.ino:64:13:   required from here
----\Documents\Arduino\libraries\AppleMIDI\src/utility/Deque.h:120:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (++_head >= Size)
                     ^
Error compiling for board Teensy 4.1.

lathoub, 2020-12-24:
wow,you're posting the entire haystack! Where is the needle?

lathoub, 2020-12-24:
updated in [branch 3.0.0rc](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/v.3.0.0rc).
(Next time pin-point the error, andavoid posting mostly irrelevant compiler output #haystacks)

glasscake, 2020-12-24:
Well the error I had was as simple as "unable to compile for Teensy 4.1"

Either way I fixed the original issue with connectivity. I was using the same MAC address for both Teensys, once I gave them individual MAC addresses the issue went away :)

Ill try the new branch for 2.2.0 and if that does not compile Ill post a new ticket.

closed 2020-12-24 by glasscake