lathoub/Arduino-AppleMIDI-Library #95: Example not working

xelarep, 2020-12-13:
Hi, tried to run the Example "ESP8266_NoteOnOffEverySec". However some trouble:

* the Example is not shown in the Arduino 1.8.13. - so I did copy & tase from the sketch here
* I use a Wemos ESP8266 D1 (ESP8266 2.7.4)
* AppleMIDI shows up as Version 2.1.0 in the library manager

I don't know where to start - I get the a long list of errors starting with 

In file included from /Users/alexander/Documents/Arduino/ESP8266_WMIDI/ESP8266_WMIDI.ino:7:0:
/Users/alexander/Documents/Arduino/libraries/AppleMIDI/src/AppleMIDI.h:297:27: error: 'EthernetUDP' was not declared in this scope
 APPLEMIDI_CREATE_INSTANCE(EthernetUDP, MIDI, "Arduino", DEFAULT_CONTROL_PORT);
                           ^
/Users/alexander/Documents/Arduino/libraries/AppleMIDI/src/AppleMIDI.h:293:43: note: in definition of macro 'APPLEMIDI_CREATE_INSTANCE'
     APPLEMIDI_NAMESPACE::AppleMIDISession<Type> Apple##Name(SessionName, Port); \
                                           ^
/Users/alexander/Documents/Arduino/ESP8266_WMIDI/ESP8266_WMIDI.ino:16:1: note: in expansion of macro 'APPLEMIDI_CREATE_DEFAULTSESSION_INSTANCE'
 APPLEMIDI_CREATE_DEFAULTSESSION_INSTANCE();
 ^

Any Help or working example for the ESP8266 appreciated

My Toolchain is basically OK (I can run Blink, FastLED or connect to WiFi)

Alexander

lathoub, 2020-12-13:
Have you selected the` WeMos D1  R1`"in the boards? `EthernetUDP` is using by the AVR boards (Uno, Leonardo, ...), not the ESP8266 boards that use <WifiUdp.h>

The example sketch should start with 

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>

//....
```

xelarep, 2020-12-13:
Yes, its the Demos R1 D1, and it's activated (I double checked with a different sketch and downloaded a FastLED sample...).

The Example Sketch is the "ESP8266_NoteOnOffEverySec", i.r. headers as you mentioned - even with the wrong example ssid/password...

lathoub, 2020-12-13:
OK, don't use the version from the library management (delete the 2.1.0 version in the library dir) - then clone the latest master branch  (or download the zipfile) and try again with the same example.

xelarep, 2020-12-14:
OK, thx. Will check this later the week and get back.

xelarep, 2020-12-15:
OK, replacing the 'AppleMIDI' library folder by the current Github-master let me build, download and run the ESP8266 Sample. Thx!

lathoub, 2020-12-17:
Thanks for checking - looks like I need to releasse a newer version to the Arduino Library manager to include ESP8266 support.

lathoub, 2020-12-17:
Thanks for reporting and checking @xelarep

closed 2020-12-17 by lathoub