lathoub/Arduino-AppleMIDI-Library #97: [ESP32"Stack smashing protect failure!"error with default bonjour name MIDI

midasgossye, 2020-12-28:


lathoub, 2020-12-28:
Thanks for reporting @midasgossye. the parser should survive a longer than expected session name. 
After some investigation, i noticed that the issue is with https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/ba3db187764bc44f8ccc18e49f8532e86e48d043/src/AppleMIDI_Parser.h#L87

and it should be:
`if (bi < DefaultSettings::MaxSessionNameLen)
`
(boundary should not be included)

Should also be corrected here
https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/ba3db187764bc44f8ccc18e49f8532e86e48d043/src/AppleMIDI_Parser.h#L277
and here
https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/ba3db187764bc44f8ccc18e49f8532e86e48d043/src/AppleMIDI_Parser.h#L339

@midasgossye Do you want to create a PR with the correction?

lathoub, 2020-12-28:
I have included the fix in the [v3.0.0rc](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/v.3.0.0rc) branch, for you to test

midasgossye, 2020-12-28:
Wow, that was fixed quickly! I just tested out the library with the same scenario from the v3.0.0rc branch. It didn't produce any errors with the long bonjour name aymore. So that fixed the issue I was having.
A new issue popped up with the v3.0.0rc branch version though: a compile error: ```'class appleMidi::AppleMIDISession<WiFiUDP>' has no member named 'setHandleError'```
Since I didn't use the HandleError event I just commented out the ```AppleMIDI.setHandleError(OnAppleMidiError);``` line and the ```OnAppleMidiError()``` routine. Maybe this is expected since V3.0.0 is still a WIP?

lathoub, 2020-12-28:
> A new issue popped up with the v3.0.0rc branch version though: a compile error: 'class appleMidi::AppleMIDISession<WiFiUDP>' has no member named 'setHandleError'

Indeed, the reason to bump the major version number. 'setHandleError' has been removed and replaced with 'setHandleException' that has an additional value parameter  (compared to the old 'setHandleError'). The examples in v3.0.0rc have been updated to reflect this change. In v3, the memory footprint can be reduced significantly (not as important for ES32 users).

I'm about to push out the Release Candidate, so any comments are appriciated.

lathoub, 2020-12-28:
(and i added a note in the ReadMe.md that long session names will be truncated)

lathoub, 2020-12-30:
solved in latest release #98

closed 2020-12-30 by lathoub