lathoub/Arduino-AppleMIDI-Library #99: Unable to change note timing in example

svanimisetti, 2021-01-10:
Thanks for sharing this project. I am trying to use the example code "ESP32_NoteOnOffEverySec.ino" to test the setup with ESP32_WROOM_32 development board. The example compiles and uploads to the board fine. I am also able to receive notes in Synethesia (via rtpMIDI by Tobias Erichsen). However,  I noticed that there is a comment specifically asking not to use delay function. I want to change the length of the note sent every seconds over rtpMIDI. Is there some issues using delay on ESP32? It appears that any delay between MIDI.sendNoteOn and MIDI.sendNoteOff does not have any affect on what's received by Synthesia - its always like a tap of the key. Am I missing something? I though the transport layer (WifiUdp) will flush all events in realtime - but not being able to control the length of the note in this example seems odd. I am trying to understand implementation from projects like Pedaluino - but its too complex for a beginner like me.

lathoub, 2021-01-10:
Not sure how to answer - not really an rtpMidi/AppliMIDI protocol question, more of a general MIDI question (I think). Consider asking in the [discussion section of the FortySevenEffects MIDI library (on which this lib is based)](https://github.com/FortySevenEffects/arduino_midi_library/discussions)

svanimisetti, 2021-06-30:
Did not realize this was still open - closed it. Apologies for keeping it open for this long.

closed 2021-06-30 by svanimisetti