lathoub/Arduino-AppleMIDI-Library #101: Ejemplo para esp8266 y rtpMidi

alomdj00, 2021-01-22:
Hola, estoy tratando de enviar notas via wifi a una conexion rtpmidi, sin embargo el jemplo que proporciona la libreria si bien compila y se conecta a la red wifi, no aparece en la interfaz de rtpMidi, mientras que una tablet con fl remote si aparece y funciona, que puede estar pasando, el modulo es un nodemcu v1, funciona con leds y hace blecturas por lo que el modulo esta bien. Podrias orientarme un poco respecto a como implementar en el codigo el rptmidi.

Muchas gracias.

lathoub, 2021-01-23:
Hi @alomdj00 the esp8266 example that comes with the library should work out of the box, you only need to provide your ssid and password.
I do not speek Spanish so i can't help. Are you using Mac of Windows (with rtpMIDI 1.1.14)?

bjoernbau, 2021-01-24:
Hi @alomdj00.
Did you try to add the device manually in rtpMidi by giving the ip address of the module? Auto detection didn't work for me either, so I also had to add the module manually.
Best regards.
Hola @alomdj00.
¿Intentaste añadir el dispositivo manualmente en rtpMidi dando la dirección ip del módulo? La detección automática tampoco funcionó por mi parte, así que también tuve que añadir el módulo manualmente.
Saludos.

alomdj00, 2021-01-25:
Hola, gracias por su respuesta, pense que se conectaba o anunciaba automaticamente, ese fue mi error sin duda, ya en este momento se puedo enviar notas midi con pulsadores.

Ahora estoy tratando de conectar un sinte para volverlo inalámbrico, asta ahora no he podido enrutar los midi wifi al puerto serial,  supongo que hay que crear la definicion sin embargo no se que me falta o si estoy bien, me pide cuatro argumentos y yo tengo esto:

APPLEMIDI_CREATE_INSTANCE(HardwareSerial, Serial, "midi",); al compilar me dice que son 4 argumentos pero no se que mas poner entre los (). Agradezco su ayuda. Un saludo

alomdj00, 2021-01-25:
Hola, una vez llegue a casa probare  el ejemplo. 
Este ejemplo sirve para midi in y out?
Mi ingles no es muy bueno, y creo que no puedo estructurar el texto correctamente  en ingles. Manejo lo basico. Uso la traduccion de chrome para entender un poco.

bjoernbau, 2021-01-26:
Hola.
Este ejemplo sólo functionar de AppleMidi a serialMidi.
Para la dirección inversa es nesecario includir la librera Midi una secunda vez. La librera [arduino_midi_library](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/DualMerger/DualMerger.ino) contiene un ejemplo para gestionar dos derecciónes de Midi.
¿Ya funciona la dirección de AppleMidi a serialMidi en tu lado?
Saludos.

alomdj00, 2021-01-26:
> Hola.
> Este ejemplo sólo functionar de AppleMidi a serialMidi.
> Para la dirección inversa es nesecario includir la librera Midi una secunda vez. La librera [...]

Hola, curiosamente a mi me funciona al reves, me sirve para enviar midi del sinte via wifi al pc, pero no logro recibir los mensajes del pc al sinte, es decir, via serial print si aparecen los bytes recibidos uno a uno, pero el sinte no los reconoce, lo probe en otro teclado con midi y tampoco reconocio los mensajes del pc, mientras que si uso un convertidor midi usb si funciona el teclado en ambas direcciones, envia y recibe sin problema. 

Tu ya has intendado esto, tienes algun codigo en el que estes trabajando y puedas compartir para ver si entre todos podemos hacer que funcione correctamente.

Un saludo.

alomdj00, 2021-01-26:
Se me olvidaba, ya hice uso de la libreria midi de fortyseveneffects y del ejemplo que propones, sin embargo en mi caso no recibi ningun dato, ocasionalmente uno que otro pero no es funcional, mucho menos recibio y mostro mensajes de tipo sysex

bjoernbau, 2021-02-01:
Hola.
¿Puede publicar las partes principales de su código?
Hasta ahora, solo he implementado la dirección AppleMidi a serialMidi, sin usar la librería midi de fortyseveneffects. Solo escribo cada byte que obtengo de AppleMidi al puerto serial.

Creo, que el principio principal de la operación bidireccional no se basa en bytes, sino en mensajes completos. Así que en lugar de obtener bytes individuales de AppleMidi, uno necesita manejar mensajes completos (nota on, nota off, etc.) y enviar estos mensajes a la librería midi de fortyseveneffects a través de los procedimientos correspondientes. Y viceversa, hay que manejar mensajes enteros de la librería midi de fortyseveneffects y enviarlos a la librería AppleMidi. Creo que esta es la única manera, ya que AppleMidi no envía cada byte como un paquete de red, sino comandos midi completos.

closed 2021-02-28 by lathoub