lathoub/Arduino-AppleMIDI-Library #107: Teensy 4.1 example not sending messages (Only receiving)

dani-carbonell, 2021-02-24:
## Context
    - lathoub/AppleMIDI @ ^3.1.0
    - Teensy 4.1 and ESP32
    - MacOS 10.13.6 and Windows 10
                  

## Problem
The example "ESP32_NoteOnOffEverySec.ino" works as expected for me. The ESP32 sends and receives MIDI normally through WIFI.

If I try the "Teensy41_NoteOnOffEverySec.ino" I am not receiving any MIDI in my computers. When I send MIDI to the board there is no problem and I see the debug callbacks for the incoming MIDI notes.

I have tried:
- Two different units of the Teensy 4.1 with its original Etherenet Kit.
- Two different OS: Windows 10 and MacOS 10.13.6 (RTPMidi by Tobias Erichsen and Audio MIDI Setup respectively)
- Two different DAWs: Ableton Live 10 and MIDI Monitor to send and receive MIDI
- Two different local Networks: (To make sure there was not strange filtering going on)

I installed Wireshark and tried to see if there was any incoming packets at all but I am not familiar with this software yet but I couldnt figure out anything.

Can somebody reproduce this error?
Thanks

lathoub, 2021-02-25:
Indeed, i see the same error on my Teensy 4.1 - but same source code works on MKR1000 with Ethernet shield.

dani-carbonell, 2021-02-25:
Does the <NativeEthernet.h> sketch work with the MKR1000?

Could that library be the source of the problem?
To my understanding that is the only difference of the Teensy 4.1 sketch. Let me know if should open this issue in "vjmuzik/NativeEthernet" repo.

lathoub, 2021-02-27:
I fixed the issue (in Master) by moving beginPacket closer to the first write; but i have to admit i have no clue why it works, do you see why?

Please test @dani-carbonell 

https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/892c03ea9956aac4f623bf36abe594646373d627/src/AppleMIDI.hpp#L638-L648

dani-carbonell, 2021-03-02:
Your commit 892c03e solved the issue for me as well. I have been for a while trying to understand why but neither do I.

closed 2021-03-05 by lathoub