lathoub/Arduino-AppleMIDI-Library #127: triggers watchdog

folkertvanheusden, 2021-08-03:
Hi,

I believe it may need a few calls to yield() here and there.
I took the basic ESP8266_NoteOnOffEverySec.ino, added mDNS and as soon as something connects to it, the watchdog gets triggered:

```
0x40203398: appleMidi::AppleMIDISession ::available() at /home/folkert/Projects/sketches/libraries/AppleMIDI/src/AppleMIDI.hpp line 30
0x402033f6: loop() at /home/folkert/Projects/sketches/libraries/MIDI_Library/src/MIDI.hpp line 712
0x401003ac: ets_post(uint8, ETSSignal, ETSParam) at /home/folkert/.arduino15/packages/esp8266/hardware/esp8266/3.0.1/cores/esp8266/core_esp8266_main.cpp line 181
0x40209ee4: loop_wrapper() at /home/folkert/.arduino15/packages/esp8266/hardware/esp8266/3.0.1/cores/esp8266/core_esp8266_main.cpp line 201
```
[esp8266_test.zip](https://github.com/lathoub/Arduino-AppleMIDI-Library/files/6921988/esp8266_test.zip)

folkertvanheusden, 2021-08-03:
Dug a bit into it and:

In **src/AppleMIDI.hpp** there's (line 27):

```
template <class UdpClass, class Settings, class Platform>
void AppleMIDISession<UdpClass, Settings, Platform>::parseControlPackets()
{
    while (controlBuffer.size() > 0)
    {
        auto retVal = _appleMIDIParser.parse(controlBuffer, amPortType::Control);
        if (retVal == parserReturn::UnexpectedData)
```

But 'parse' can also return parserReturn::NotSureGiveMeMoreData (and others) meaning that controlBuffer contains not enough data and thus may loop forever (causing the wdt to trigger).
So I suggest parserReturn::Processed to be checked against there.

It doesn't explain why it fails to communicate with https://github.com/davidmoreno/rtpmidid though.

lathoub, 2021-08-21:
additional test to avoid infinite loop (added in master branch, tested on ESP32)

https://github.com/lathoub/Arduino-AppleMIDI-Library/blob/17a1b7bc3223ad9edb2ea4c0f2edd2a3f2b7fe62/src/AppleMIDI.hpp#L32-L38

@folkertvanheusden does this resolve this issue?

lathoub, 2021-08-21:
@folkertvanheusden thank you reporting and proactive digging into the issue

folkertvanheusden, 2021-08-21:
The one from the branch I responded last on worked fine. Haven't tested the
merged version yet.

lathoub, 2021-08-21:
> The one from the branch I responded last on worked fine. Haven't tested the merged version yet.

That did not have the fix yet :-)

folkertvanheusden, 2021-08-21:
The master? Or the branch?
The branch works fine with long- and short names.

lathoub, 2021-08-21:
Only 1 branch now: [master](https://github.com/lathoub/Arduino-AppleMIDI-Library)
Master now contains a fix for the above issue (session names already solved and folded in master earlier today)

lathoub, 2021-08-22:
I did another fix: the session name of the data invitation was incorrect: it send back the remote session name, rather than the local name.

Maybe that addresses the below issue:

> It doesn't explain why it fails to communicate with https://github.com/davidmoreno/rtpmidid though.

closed 2021-08-22 by lathoub